merbanan/rtl_433 #3105: Duplicate recordings per event of Bresser-3CH device.

aleks-ko89, 2024-12-01:
Hi. I'm on my RTL finding the values of a Brasser wireless thermometer (or Brasser-like in protocol, don't really know, it's not my device), but there are 2 recordings for each measurement. 
I recorded the signal, it looks uniform to me, 24 repetitions of the signal at regular intervals.
I attach the recorded file (the file, by the way, was recorded one for two recordings)
[image]
[g011_433.92M_250k.zip](https://github.com/user-attachments/files/17969364/g011_433.92M_250k.zip)

zuckschwerdt, 2024-12-01:
Looks like a 2 second gap, that will be two sample files, maybe g010 or g012? I think the original has 15 repeats once every minute. Could be some variant with 24 repeats sent twice in succession.

aleks-ko89, 2024-12-01:
No, both records appear in the temprinal almost instantly. It is unclear why there is such a 2-second difference in the output.
If you identify the file `./rtl_433_64bit_static.exe -w OOK:- g011_433.92M_250k.cu8` then the output gives 2 records.

aleks-ko89, 2024-12-01:
The period of transmission of readings is 1 m 54 s. It may not be the original Bresser, but some other sensor with a similar coding protocol.
Sometimes readings are even decoded as `[87] Generic-Motion` in the middle
[image]
[image]

zuckschwerdt, 2024-12-01:
Ok, interesting. There is a visually continuous transmission of 24 packets but somehow that get broken into 17+7 packets.
Also noteworthy, there is first a sequence of 9 PPM packets, then the 24 PWM. So this is some other sensor which emits two different protocols for compatibility.

zuckschwerdt, 2025-07-01:
It is somewhat unusual but I don't think I'll find the time to investigate. Let's close this until someone has an idea what's going on.

closed 2025-07-01 by zuckschwerdt