merbanan/rtl_433 #3110: Cotech-367959 ID

AndyW999, 2024-12-09:
I have a NicetyMeter - the receiver is marked FT0345 - and it is recognised as protocol 153 Cotech-367959 ID.

There seems to be a problem with the Rain measurement - 

time      : 2024-12-09 08:26:56
model     : Cotech-367959 ID        : 141
Battery   : 1            Temperature: 35.7 F       Humidity  : 97 %
Rain      : 85.2 mm      Wind direction: 100       Wind      : 0.0 m/s
Gust      : 0.0 m/s      Light Intensity: 0 lux    UV Index  : 0
Integrity : CRC

I think I would have noticed!

Is there a way of outputting the Hex values to see what is going on?

Thanks

Andy.

zuckschwerdt, 2024-12-09:
Can you describe the problem you are seeing?

AndyW999, 2024-12-09:
Well it's not raining but the rain is showing -

Rain : 85.2 mm

zuckschwerdt, 2024-12-09:
I see :) Reported rain is a total value. You'll have to derive hourly or daily deltas yourself.

AndyW999, 2024-12-09:
OK - total over how long though?

zuckschwerdt, 2024-12-09:
lifetime, until you reset batteries.

AndyW999, 2024-12-09:
OK - thanks!