merbanan/rtl_433 #3112: HCS362 is not supported

yashikada, 2024-12-14:
I have a keyfob with HCS362 and TH72032 (ASK Transmitter). 
Data pin of HCS362 is directly connected to ASKDATA pin of TH72032.
The keyfob use 868 MHz ISM band.

I tried to decode using protocol 218 or 131 but is not detected, general detect not detect nothing.
I tried with Pulse Analyzer I see many "Detected OOK package" and "Detected FSK package", but none return expect data.

zuckschwerdt, 2024-12-14:
Try to grab some sample files, check if they are good (https://triq.org/pdv/) and upload here as zip.
https://triq.org/rtl_433/ANALYZE.html

yashikada, 2024-12-14:
Here [data.zip](https://github.com/user-attachments/files/18135923/data.zip) 2 capture, thank you

zuckschwerdt, 2024-12-14:
The samples look like OOK, maybe with Manchester coding.
Did you choose 868.8M to get away from interference? Usually `-f 868.3M` should work well here.

The signal is clipping (too loud). Record new samples with more distance or remove the antenna while recording.
You can see the clipping as orange and red markers here https://triq.org/spectrogram-next/ (a test version of pdv with better zoom).

yashikada, 2024-12-14:
> Did you choose 868.8M to get away from interference? Usually `-f 868.3M` should work well here.

No, with flipper zero the frequency analyzer detect 868.8M, so I used that frequency.

> The signal is clipping (too loud). Record new samples with more distance or remove the antenna while recording.

[Here](https://github.com/user-attachments/files/18136471/data.zip) a new capture using frequency 868.3M and more distance.

zuckschwerdt, 2024-12-14:
The new capture is better, the one without antenna has optimal levels (no clipping), compare in pdv to see the aliasing disappear.

Looks like 214 µs OOK with Manchester coding. There are 12 preamble pulses, then 4 widhts (864 µs) desync gap, then ~71 bits.

Use `rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=214,l=214,g=600,r=900'` for raw data or
`rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=214,l=214,g=600,r=900'` for MC decoded, but note that the MC start is not detected properly and should be e.g. `f35d1d70c41c840302` instead of the shown `000035d1d70c41c840302`

closed 2024-12-14 by zuckschwerdt

zuckschwerdt, 2024-12-14:
A datasheet is available and this coding is shown on page 9: https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/40189E.pdf

The default format seems to be PWM (mode 0), what we observed here is the alternative Manchester format (mode 1), see page 31-32 for a comparision.

We might want to add a decoder that can handle all this perhaps?

yashikada, 2024-12-14:
> We might want to add a decoder that can handle all this perhaps?

It would be great although I don't think I have the skills to implement such a change.

zuckschwerdt, 2024-12-14:
You can test the code from #3113 now. Not tested at all, position of bits and esp. the MC decoding might be wrong.

yashikada, 2024-12-14:
I tried without antenna and don't work, but with antenna I catch all trasmissions.

<details>

<summary>Output</summary>

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:19
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:20
model     : Microchip-HCS362                       id        : 0427046
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : CE2D91E0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:27
model     : Microchip-HCS362                       id        : 0427047
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : 7E80A858
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:27
model     : Microchip-HCS362                       id        : 0427047
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : 7E80A858
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:27
model     : Microchip-HCS362                       id        : 0427047
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : 7E80A858
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:27
model     : Microchip-HCS362                       id        : 0427047
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : 7E80A858
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:41:28
model     : Microchip-HCS362                       id        : 0427047
Battery   : 0            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : 7E80A858
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-12-14 21:42:13
model     : Microchip-HCS362                       id        : 0427046
Battery   : 1            Button    : 4             Learn mode: 0             Repeat    : 0             encrypted : A11A9718
```
</details>

My keyfob has 4 buttons but I see only 0 or 4. id change depending on the button pressed, I don't know if is correct.