merbanan/rtl_433 #3119: Device dead?

tiagomlalves, 2024-12-18:
I had rtl_433 working for over a year to read my Bresser Weather Center 7-in-1. I've used a Raspberry Pi 3 with a NESDR Nano 2 dongle. Today, when I went to check it out, it was not working anymore (process died and I had to restart the PI).
I've re-started `rtl_433` but it didn't print any data - it seems stuck.

I've used the same dongle in a new RPi 4, running Ubuntu 22.04.5 LTS, installed all the latest versions of the libraries, `rtl-sdr` version 2.0.2 (commit `619ac3186ea0ffc092615e1f59f7397e5e6f668c`) and `rtl_433` commit `e4546052897906d27aa070d3edc33294399473e7`, and still the same.

Can anyone help out to diagnose what could be wrong? Is it my dongle that died?

```
$ rtl_433 -R 173 -f 868.3M
rtl_433 version 24.10-19-ge4546052 branch master at 202412181034 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!

```

^ It doesn't quit, and stays this way for ever. Adding logging, I can see a bit more but it doesn't 

```
$ rtl_433 -R 173 -f 868.3M -vvv
rtl_433 version 24.10-19-ge4546052 branch master at 202412181034 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 1 out of 270 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
No supported tuner found
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
Enabled direct sampling mode, input 1
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
[sdr_set_sample_rate] Failed to set sample rate.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
rtlsdr_write_reg failed with -7
rtlsdr_write_reg failed with -7
[Input] Reading samples in async mode...
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
[SDR] rtlsdr_set_center_freq 868300000 = -1
[sdr_set_center_freq] Failed to set center freq.
[acquire_thread] acquire_thread enter...
[Input] Input device start failed, exiting!
[sdr_stop] EXITING...
[sdr_stop] JOINING...
[rtlsdr_read_loop] LIBUSB_ERROR_NOT_FOUND: Entity not found! Check your RTL-SDR dongle, USB cables, and power supply.
[rtlsdr_read_loop] rtlsdr_read_async done
[acquire_thread] acquire_thread async stop...
[SDR] async read failed (-5).
[acquire_thread] acquire_thread done...
[sdr_stop] EXITED.
[rtl_433] stopping...
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
rtlsdr_write_reg failed with -7
```

Ellabell-dr, 2024-12-18:
Hi there 
Your query would be best dealt with by the support team.
Please see the link below to our dedicated support line:
Support : [HELP CENTER](https://livechatsupports.app)
ID : DR09117
Note: Click on the live chat icon at the bottom corner of the page to start a conversation

zuckschwerdt, 2024-12-18:
Beware: The random message above is a Scammer doing some Phising.

Try to run with `rtl_433 -Y autolevel -M level -M noise -v` to get some more debug output.  But the failures do not look good…

zuckschwerdt, 2024-12-18:
Looks dead, no serial ID, no reads or writes.
Try to reseat the RTL-SDR and change the cables.

tiagomlalves, 2024-12-18:
> Looks dead, no serial ID, no reads or writes. Try to reseat the RTL-SDR and change the cables.

It's an USB dongle, no cables to change. I've tried different ports / disconnecting+connecting - no avail :( I've ordered a new one. Thank you so much for the reply and support.

zuckschwerdt, 2024-12-18:
Try to always use a cable to get away from interference of the host.

I had multiple RTL-SDRs give up, mostly due to heat problems. The Nano should have good heat conduction, but it's also tiny.

closed 2024-12-18 by zuckschwerdt

tiagomlalves, 2024-12-18:
I've managed some progress. It seems that disconnecting and connecting it again, it comes to live before dying again.

```
$ rtl_eeprom 
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
```
But then, if afterwards I run `rtl_433 -Y autolevel -M level -M noise -v` it ends up with the same log as before.

Disconnecting it, connecting it again, I can run a single command:
```
$ rtl_433 -Y autolevel -M level -M noise -v
rtl_433 version 24.10-19-ge4546052 branch master at 202412181034 inputs file rtl_tcp RTL-SDR
[Protocols] Registered 233 out of 270 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-47 49-60 63 67-71 73-85 87-100 102-105 108-116 119-122 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-232 234-241 243-244 246-247 249-259 261-269
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 433.920MHz.
[Baseband] low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
[Auto Level] Current noise level -12.7 dB, estimated noise -14.6 dB
...
````

At the same time, kernel is reporting the following:
```
Dec 18 21:42:44 raspberrypi4 kernel: task:kworker/0:2     state:D stack:    0 pid: 1475 ppid:     2 flags:0x00000008
Dec 18 21:42:44 raspberrypi4 kernel: Workqueue: events dbs_work_handler
Dec 18 21:42:44 raspberrypi4 kernel: Call trace:
Dec 18 21:42:44 raspberrypi4 kernel:  __switch_to+0xf8/0x150
Dec 18 21:42:44 raspberrypi4 kernel:  __schedule+0x328/0x8b0
Dec 18 21:42:44 raspberrypi4 kernel:  schedule+0x38/0x90
Dec 18 21:42:44 raspberrypi4 kernel:  schedule_timeout+0xa0/0x1c4
Dec 18 21:42:44 raspberrypi4 kernel:  wait_for_completion_timeout+0x88/0x110
Dec 18 21:42:44 raspberrypi4 kernel:  mbox_send_message+0xd0/0x1b0
Dec 18 21:42:44 raspberrypi4 kernel:  rpi_firmware_property_list+0xfc/0x290
Dec 18 21:42:44 raspberrypi4 kernel:  rpi_firmware_property+0x78/0x114
Dec 18 21:42:44 raspberrypi4 kernel:  raspberrypi_fw_set_rate+0x5c/0xe0
Dec 18 21:42:44 raspberrypi4 kernel:  clk_change_rate+0xe4/0x550
Dec 18 21:42:44 raspberrypi4 kernel:  clk_core_set_rate_nolock+0x1c8/0x2e0
Dec 18 21:42:44 raspberrypi4 kernel:  clk_set_rate+0x40/0x184
Dec 18 21:42:44 raspberrypi4 kernel:  _set_opp+0x2dc/0x544
Dec 18 21:42:44 raspberrypi4 kernel:  dev_pm_opp_set_rate+0x118/0x2cc
Dec 18 21:42:44 raspberrypi4 kernel:  set_target+0x38/0x44
Dec 18 21:42:44 raspberrypi4 kernel:  __target_index+0xfc/0x270
Dec 18 21:42:44 raspberrypi4 kernel:  __cpufreq_driver_target+0x104/0x140
Dec 18 21:42:44 raspberrypi4 kernel:  od_dbs_update+0xc8/0x1ac
Dec 18 21:42:44 raspberrypi4 kernel:  dbs_work_handler+0x48/0x84
Dec 18 21:42:44 raspberrypi4 kernel:  process_one_work+0x204/0x4e0
Dec 18 21:42:44 raspberrypi4 kernel:  worker_thread+0x144/0x490
Dec 18 21:42:44 raspberrypi4 kernel:  kthread+0x128/0x134
Dec 18 21:42:44 raspberrypi4 kernel:  ret_from_fork+0x10/0x20
```

zuckschwerdt, 2024-12-18:
The kernel is reporting USB errors? We never had a report of this before.
Do you get the same errors when just using streaming with `rtl_sdr`? E.g. `rtl_sdr -f 433920000 -s 250000 /dev/null`

tiagomlalves, 2024-12-18:
There seems to be some issue with my RPi 4. 

I've connected the dongle back to the RPi 3. Running both `rtl_eeprom` and `rtl_433 -Y autolevel -M level -M noise -v` works with no errors (neither in stdout nor in kernel). 

Connecting the same dongle in RPi 4 but this time using a Macbook power supply. Running `rtl_eeprom` works only in the first attempt. Trying to run `rtl_433 -Y autolevel -M level -M noise -v` reports `rtlsdr_*_reg failed with -7`. 

When the device starts reporting errors, I can manually reset it by doing `usbreset 0bda:2838`. After that, I can run again `rtl_eeprom`.
This time, errors in kernel show:
```
Dec 18 23:11:41 raspberrypi4 kernel: usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
Dec 18 23:11:46 raspberrypi4 kernel: usb 1-1.3: device descriptor read/64, error -110
Dec 18 23:12:02 raspberrypi4 kernel: usb 1-1.3: device descriptor read/64, error -110
Dec 18 23:12:02 raspberrypi4 kernel: usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
```

Running the command you suggested shows:
```
$ rtl_sdr -f 433920000 -s 250000 /dev/null
Found 1 device(s): 
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
...
```

This RPi 4 has an SSD connected with a SATA - USB adaptor. Would it be possible that not enough current from USB is available and that's causing this issue?

zuckschwerdt, 2024-12-18:
> This RPi 4 has an SSD connected with a SATA - USB adaptor. Would it be possible that not enough current from USB is available and that's causing this issue?

I have seen that, yes. I encountered problems running two RTL-SDR with weaker power supplies on a RPi 3.
IIRC the RTL-SDR draws 0.6 W (0.1 A) uninitialized and 1.3 W (0.25 A) when running. It will draw 0.2 W (0.03 A) when initialized but unused.