merbanan/rtl_433 #3120: Support for LaCrosse WS6868 weather station

jeanvaljan, 2024-12-23:
Hi,

I got a LaCrosse WS6868 weather station with a TX231RW Multi sensor and a TX232TH-LCD hygrometer and want to integrate the sensors in HA. Apparently, the sensor's protocols are some kind of variations of LaCrosse-BreezePro and LaCrosse-TH2. I grabbed some samples for each and few expected values for the codes readings of the TX232TH-LCD sensor. Unfortunately TX231RW doesn't have a display to read any values.
If anyone can help building a decoder for these sensors would be much appreciated. The attached archive includes also user manuals for both.

Regards,
Jean V
[lacrosse_ws6868.zip](https://github.com/user-attachments/files/18225675/lacrosse_ws6868.zip)

zuckschwerdt, 2024-12-23:
The samples are good. Turns out the FSK PCM is clocked at 58 µs. Using that (instead of 100µs) we get:

```
$ rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=58,l=58,r=2000' TX231RW/*.cu8
{171}1555555555695516ea0ce214050000628005557b3f0
{171}1555555555695516ea0ce214050000628005557b3f0
{171}1555555555695516ea0ce2140600006280055571360
{171}1555555555695516ea0ce2140700006280055500c68
{171}1555555555695516ea0ce2140000006280055565240
{171}1555555555695516ea0ce2140100006280055514d48
{171}1555555555695516ea0ce214020000628005551edf8
{171}1555555555695516ea0ce214030000628005556f310
{171}1555555555695516ea0ce214040000628005550acd8
{171}1555555555695516ea0ce214050000628005557b3f0
{171}1555555555695516ea0ce2140600006280055571360
{171}1555555555695516ea0ce2140700006280055500c68
{171}1555555555695516ea0ce2140000006280055565240

$ rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=58,l=58,r=2000' TX232TH-LCD/*.cu8
{220}ffffff80000000155555555a5545ba82000002057a071dc00000000
{228}ffffffff80000000155555555a5545ba82000002057a071dc00000000
{221}ffffff80000000155555555a5545ba82000002457e06e36000000000
{221}ffffff80000000155555555a5545ba82000002858206f0e000000000
{229}ffffffff80000000155555555a5545ba82000002858206f0e000000000
{221}ffffff80000000155555555a5545ba82000002c58406c06000000000
{218}ffffff80000000155555555a5545ba82000003058606cf000000000
{226}ffffffff80000000155555555a5545ba82000003058606cf000000000
{220}ffffffc00000000aaaaaaaad2aa2dd41000001a2c40364400000000
{221}ffffff80000000155555555a5545ba82000003858a06c46000000000
{229}ffffffff80000000155555555a5545ba82000003858a06c46000000000
{221}ffffff80000000155555555a5545ba82000003c58c06b8a000000000
{221}ffffffc00000000aaaaaaaad2aa2dd4100000102c7035be000000000
{228}ffffffff80000000155555555a5545ba82000002058e06b7c00000000
{221}ffffff80000000155555555a5545ba82000002458e06b72000000000
```

Putting those codes in a BitBench and shifting the bits to find a boundary we see the common preamble of `d2aa2dd4`.

So this flex decoder will recover the actual codes:
`rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=58,l=58,r=2000,preamble=d2aa2dd4'`

Please capture a few codes and table them (as text) along with the expected values. We can then find the data fields.

Aligned codes from your sample files are
```
TX231RW/*.cu8
19c4 280a 0000 c500 0aaa f67e
19c4 280a 0000 c500 0aaa f67e
19c4 280c 0000 c500 0aaa e26c
19c4 280e 0000 c500 0aaa 018d
19c4 2800 0000 c500 0aaa ca48
19c4 2802 0000 c500 0aaa 29a9
19c4 2804 0000 c500 0aaa 3dbf
19c4 2806 0000 c500 0aaa de62
19c4 2808 0000 c500 0aaa 159b
19c4 280a 0000 c500 0aaa f67e
19c4 280c 0000 c500 0aaa e26c
19c4 280e 0000 c500 0aaa 018d
19c4 2800 0000 c500 0aaa ca48

TX232TH-LCD/*.cu8
1000 0010 2bd0 38ee
1000 0010 2bd0 38ee
1000 0012 2bf0 371b
1000 0014 2c10 3787
1000 0014 2c10 3787
1000 0016 2c20 3603
1000 0018 2c30 3678
1000 0018 2c30 3678
1000 001a 2c40 3644
1000 001c 2c50 3623
1000 001c 2c50 3623
1000 001e 2c60 35c5
1000 0010 2c70 35be
1000 0010 2c70 35be
1000 0012 2c70 35b9
```

softtitude, 2024-12-24:
Hi Christian,

Thanks for your help!
Before coming back with new samples, I'd like to understand what you mean by table-ing the codes as text. Can you share an example please?

Thanks,
Jean

jeanvaljan, 2024-12-24:
Hi Christian,

Thanks for your help!
Before coming back with new samples, I'd like to understand what you mean by table-ing the codes as text. Can you share an example please?

Thanks,
Jean

ProfBoc75, 2024-12-24:
@jeanvaljan : table means a list of captured codes (from flex command) + corresponding values like temp, hum ....

A possible [bitbench](https://triq.net/bitbench#c=%5BTX232TH-LCD%2F*.cu8%5D&c=1000%200010%202bd0%2038ee%20%5B%20bat%20ok%2C%20channel%201%2C%20temp%2020.1%20%C2%B0C%2C%20hum%2056%25%20%5D&c=1000%200010%202bd0%2038ee&c=1000%200012%202bf0%20371b&c=1000%200014%202c10%203787&c=1000%200014%202c10%203787&c=1000%200016%202c20%203603&c=1000%200018%202c30%203678&c=1000%200018%202c30%203678&c=1000%20001a%202c40%203644&c=1000%20001c%202c50%203623&c=1000%20001c%202c50%203623&c=1000%20001e%202c60%2035c5&c=1000%200010%202c70%2035be&c=1000%200010%202c70%2035be&c=1000%200012%202c70%2035b9&f=ID%2024h%20BAT%20b%20TEST%20%20b%20CHANNEL%202h%20COUNTER%204h%2012d%2012d%20CRC%208h%0A&z=1&cw=4) for tx232th similar to tx141w , with an exemple of values to share with codes.

jeanvaljan, 2024-12-29:
Hi,

@ProfBoc75: the string format you proposed (ID 24h BAT ? b TEST ? b CHANNEL ? 2h COUNTER 3d ? b TEMP 12d HUM 12d CRC 8h) for TX232TH-LCD seems to work. For the temperature, the same offset and scale as for TX141 applies (500 offset, scale 10). I'm a bit concerned about the channel: it's reported as 1 but the actual channel the sensor is set to is 2; i suppose the channel numbering starts from 0. As for the ID, it's a bit strange that it is not changing when re-powered.
I'll come back with additional data for TX231RW as i have to simulate somehow wind and rain.

ProfBoc75, 2024-12-29:
Hi @jeanvaljan : yes for the channel it is common to add 1 to get the real value.

    0x0 = channel 1
    0x1 = channel 2
    ...

If the id doesn't change, this limits the installation to just 1 sensor and could potentially create false values from neighbouring sensors using the same station.