merbanan/rtl_433 #3123: Unable to monitor 2024 Chevrolet Silverado TPMS at 315 or 433

ksaye, 2024-12-26:
I have monitored many TPMS with rlt_433 and have been very happy with the results.  So far, all have been at 315 MHz, and I have monitored 2009 - 2020 Mazda, Lincoln and GMC.

I recently purchased a 2024 Chevrolet Silverado Custom and from what I read (https://accessories.chevrolet.com/product/tire-pressure-monitor-sensor-(xl8-433-mhz)-86530815); it appears that they use 433 vs. 315.  Even when switching to 315 MHz, I do not see anything even when I use a EL50448 Relearn Tool.

***Before I start deeper analysis, is this a known issue?  I did not see anything in the issues list.***

| | |
| ---- | ----- | 
| Hardware | Raspberry Pi 4 Model B Rev 1.4 |
| OS | Linux pi4 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux |
| RTL | rtl_433 version 24.10-19-ge4546052 branch master at 202412181034 inputs file rtl_tcp RTL-SDR |
| SRD | device 0: Realtek, RTL2838UHIDIR |

zuckschwerdt, 2024-12-26:
Quickest way would be to use some SDR GUI and watch the band while triggering the TPMS. When you find the signal then record a sample with rtl_433 to analyze. You could do with only rtl_433 but it's some trial and error to record something.
https://triq.org/rtl_433/ANALYZE.html

ksaye, 2024-12-27:
Thanks for the pointer @zuckschwerdt.  Unless others have already done the homework, I will start this in a few weeks and update / close the issue when complete.

e100, 2025-01-26:
If it helps, I purchased a schrader TPMS set that came with a EL-50448.
The listing shows they fit 2015-2024 Silverado, among many other GM, Holden, Tata vehicles.

I get data with:
```
rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=132,l=0,r=224'
```

```
time      : 2025-01-25 19:56:33
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 130          data      : 0000000000004c90007849093000562c0
codes     : {130}0000000000004c90007849093000562c0
```

They are 433.920Mhz, my radio is RTL2832U

I'd like to figure out how to get these decoded too but know little about SDR.

e100, 2025-01-26:
The box my sensors came in were labeled PAW-1515 searching google I found some information
433.92 MHz ± 30 KHz FSK

The EL-50448 talk to them using:
125 KHz ± 5 KHz ASK

I believe SAE J2735 DSRC MESSAGE SET DICTIONARY specifications would give us details of the data format.
But I do not know how to get my hands on those documents for free.

gdt, 2025-01-26:
You are well on your way.   I would set up monitoring to record over multiple days and see what you get in terms of codes and time.  I would use json and enable signal strenght, and perhaps put the receiver closest to one wheel.   As an example, older Ford TPMS transmits every 6h at rest, starting from when the car stops moving.   Or, you could record while driving.

Then, you can see if you always/often get 130 bits, or what the distribution is, and then what the patterns are.  Reading the sources for other tpms is a great source of clues.   surely there is an ID, and fields for temp and pressure.   probably there are flag bits for in motion, and maybe sudden change.

ksaye, 2025-01-27:
@e100 & @gdt I am starting the process today of 'measuring' the signals from my 2024 Silverado, following: https://triq.org/rtl_433/ANALYZE.html

Here is my process:
1. install Raspbian on a Pi 4
2. install rtl_433
3. plug in the SDR
4. disconnect the antenna from the SDR
5. place the pi4 w/SDR right next to the front tire stim
6. run the command 'sudo rtl_433 -S unknown' for ~ 24 hours
7. analyze the results

So far, I have the following results, but it is too early to assume with great accuracy it is the TPM sensor

https://triq.org/pdv/#AAB0130801067403B4005408280220003C01A451948455+AAB0130801067403B4005408280220003C01A451949455+AAB0150801067403B4005408280220003C01A45194A5B69755

My concern is that the results do not indicate a modulation.

Am I missing something?

Also, if I need to speed up the process, I believe if I let PSI out of the tire, it will force the signal and I can map the drop of PSI

zuckschwerdt, 2025-01-28:
That  pdv-link has no sensible data. 130 µs is not a usual timing, we expect 100 or 50 µs. Try to decode with OOK_PCM first. The MC decoder might hide some preamble or (de-)sync pattern.

Try to get one good sample to analyze and deduce timings. Use `-s 1024k`. Inspect the sample in https://triq.org/pdv/ and if the sender is exactly on 433.92 MHz then tune away 20 kHz (e.g. `-f 433.7M`).

e100, 2025-01-28:
Not sure if this is right but it seems fairly accurate.
One pressure byte * 1572 * 2 = pascals
One temp byte - 60 = degrees C

```
??????????????????????????PPTT???
0000000000004c90007849176600536d0  0x00 No pressure
0000000000004c9000784917660650700  0x06 ~3PSI 6*1572 * 2 = 18864 pa/2.75psi
0000000000004c900078491766254e8d0  0x25 ~16psi 37 * 1572 * 2 = 116328pa//16.87psi
Three other sensors:
0000000000004c90007847c1a70052550  0x00 No pressure
0000000000004c90007847c1a72658810  0x26 ~16psi 38 *1572 * 2 = 119472pa/17.32psi 

0000000000004c9000784909300050260  0x00 No pressure
0000000000004c90007849093024514b0  0x24 ~16psi  36 × 1572 × 2 = 113184pa/16.4psi

0000000000004c900078473b610059900 0x00 No pressure, in pocket 0x59, 89 - 60 = 29C/84.2F
0000000000004c900078473b610030670 0x00 No Pressure in freezer 0x30, 48 − 60 = -12C/10.4F
0000000000004c900078473b61244ca70 0x24 16,4psi, after warming a bit 0x4C, 76-60 = 16/C60.8F
0000000000004c900078473b613f50c60 0x3F ~26psi 63 × 1572 × 2 = 198072pa/28.7psi 0x50, 80 - 60 = 20C/68F
```
I think it might send shorter messages sometimes too.

The info I found online about the sensors said:
Resolution of Pressure is 1.572 KPa
Accuracy +- 0.1 bar (0-70C)  +- 0.24 bar (-20-0C 70-85C)
Temp Range -20C to 85C
Resolution 1C
Accuracy +- 3C (0-70C)

`rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=132,l=0,r=224'`  has worked well for me so far, it was one of the various suggestions that the -A option spit out.

I can confirm it will send data on pressure change and it will send when you activate the tool next to it.
I had issues receiving anything without the antenna. I do have to keep the antenna 5-6ft away from the TPMS to get a good read. 

https://triq.org/pdv/#AAB104008800E800602714828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282918292818291829182828282828282828282828292828281828282918282928281829282819281928182828291828282828282828291928191918291828291828355

https://triq.org/pdv/#AAB104008800EC0060271482828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828291829281829182918282828282828282828282829282828182828291828292828182928281928192818282829182828282828282829192819282818291829182828355

I'm planing to do what @gdt suggested and get a PI setup so I can install these on my truck and get some long term data recorded. Hopefully we can determine if they detect movement and how often they send and such.

What options would be best for recording the data?

zuckschwerdt, 2025-01-28:
Those are very good results already! Can you grab a good sample and upload here as zip? Verify there is no clipping (orange and red markers) with https://triq.org/spectrogram-next/

The timing seems to be 120 µs (for a MC half-bit), that's a timing we have seen before. The apparently longer pulses are just artifacts of the demod, if you average over pulses and gaps you should get around 120 µs.

e100, 2025-01-29:
Here is a sample
[g014_433.92M_1024k.zip](https://github.com/user-attachments/files/18580726/g014_433.92M_1024k.zip)

e100, 2025-01-29:
I've managed to figure out the checksum and what changes when it is rolling vs stationary.
????????????SSSSDDDDIIII??PPTTCC?

SSSS, status
4c90 stationary
4d93 rolling

CC, the checksum is Sum mod 256

I think DDDD might be the device type identifier and IIII and possibly the byte after it is the unique identifier.

zuckschwerdt, 2025-01-29:
That's a very clean recording, the timing is 120 µs and there is no desync, only one long string of MC data -- the OOK_MC_ZEROBIT is perfect here.

Usually we expect 7-8 chars of ID, e.g. 4 bytes, I'd say DDDDIIII is one identifier, the manufacturer might use ID prefixes though.

e100, 2025-01-31:
@zuckschwerdt 

This seems to reliably capture all 130 bits:
rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=120,l=0,r=15600'

It looks like each time data is sent it sends the same data six times

I think this is fairly close to complete:
????????????SSSSDDDDIIIIIIPPTTCC?

So far DDDD has remained identical across all 4 sensors I have
IIIIII is unique to each sensor and have not changed
Not sure if those should be separate device type/device id or just one id.

These sensors have two modes, learn mode and operational mode.
The EL50448 tool commands them into learn mode.
When entering learn mode they transmit.
If the pressure remains zero, they will not transmit again until the pressure increases.
Once they have pressure they will send data and about 15 seconds later send data again.
Then it will start transmitting every 2 minutes in operational mode.
They do not seem to have any sensors to detect motion and send every 2 minutes as long as they have pressure.

If the pressure drops by a large amount ( from what I've read on automotive sites about 8psi) they will drop back to learn mode. This is a feature so you can use pressure, rather than the EL50448, to get them in learn mode and paired to the car.
Provided they have pressure they will behave the same as outlined above and drop into operational mode.

I've read that they should transmit data any time the pressure changes by about 2.5psi, but I am not sure if they change mode when doing so. I don't have a simple way to to change the pressure so little, just a tire to put them on and off of.

The battery status changes when the voltage is 2.0-2.1v, they operate with 1.6v and stop working at 1.5v
The operation and frequency of transmission does not seem to change based on voltage.

SSSS:
Out of the box all the bits were the same on all four sensors and set as:
0100 1100 1001 0000

bits 8,15 and 16 indicate learn/operational mode
learn = 0
operational = 1

bit 11 is the battery status
0 = ok
1 = low battery

???????????? at the beginning has remained all zeros
After the checksum CC the final two bits are always zero

To make this comment complete with all decoding details
PP * 3144 = pascals
TT - 60 = degrees C

The checksum CC is sum modulo 256

I am not a C developer so not really sure where to go from here.

zuckschwerdt, 2025-01-31:
That looks in-depth and complete, thanks! From your documentation we can implement a decoder.

gdt, 2025-01-31:
Bits that might be id but are the same might or might not be.   If you have 4 units and they are the same, that could well be because they were manufactured around the same time.

As always I lean to unknown bits being labeled and included in the data output, for maximal chance of someone figuring it out.

I might be able to test a decoded that's on a branch in the repo; I know someone with a 2015ish GMC.

ksaye, 2025-01-31:
@gdt great work so far.  Sadly, that change does not work (yet) on my 2024 Silverado.  I will do more testing this weekend as the truck is currently in an area with lots of 433 MHz devices.

I plan to isolate the Silverado this weekend and run through some test to let air out of the tires and capture the results.  I do have the relearn tool but I think it is just muddying up the capture.

e100, 2025-01-31:
@ksaye The tool needs to be pretty close to the sensor for it to work and it transmits on different frequency than the 433Mhz we are interested in.

@zuckschwerdt Would it be helpful if I uploaded some recordings of the sensors in various states?

zuckschwerdt, 2025-01-31:
We only need one good sample to pin down the timings, your analysis based on codes is perfect.

ksaye, 2025-02-03:
@gdt , you are clearly much further along that I.  :)

@zuckschwerdt , happy to split this into a different issue, based on my comments below.

I am getting a much different experience than @gdt and it makes me think I have a different TPMS in my vehicle.  With his suggested command, received all kinds of 2 bit and ~85 bit signals.  I also noticed my version on a Raspberry Pi was from 2022 so I built from source -- still no change.

@gdt 's suggestion did not yield any results BUT isolating my vehicle and letting the air out did show some response below.

running ```sudo rtl_433 -S unknown``` and letting the air out of a tire did record samples where I see signals of len 144 - 146, shown below:

```
Detected FSK package    @0.375912s
Analyzing pulses...
Total count:   19,  width: 7.46 ms              ( 1864 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:    2,  width:  104 us [104;108]    (  26 S)
 [ 2] count:   11,  width:   52 us [52;52]      (  13 S)
 [ 3] count:    3,  width:  156 us [156;156]    (  39 S)
 [ 4] count:    2,  width:  208 us [208;208]    (  52 S)
Gap width distribution:
 [ 0] count:    1,  width: 2668 us [2668;2668]  ( 667 S)
 [ 1] count:    3,  width:  208 us [208;208]    (  52 S)
 [ 2] count:    1,  width:  260 us [260;260]    (  65 S)
 [ 3] count:    3,  width:  104 us [104;104]    (  26 S)
 [ 4] count:    8,  width:   52 us [52;52]      (  13 S)
 [ 5] count:    1,  width: 1352 us [1352;1352]  ( 338 S)
 [ 6] count:    1,  width:  156 us [156;156]    (  39 S)
Pulse period distribution:
 [ 0] count:    1,  width: 2668 us [2668;2668]  ( 667 S)
 [ 1] count:    5,  width:  280 us [260;316]    (  70 S)
 [ 2] count:    1,  width:  416 us [416;416]    ( 104 S)
 [ 3] count:    4,  width:  104 us [104;104]    (  26 S)
 [ 4] count:    3,  width:  208 us [208;208]    (  52 S)
 [ 5] count:    3,  width:  156 us [156;156]    (  39 S)
 [ 6] count:    1,  width: 1404 us [1404;1404]  ( 351 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:    5,  width:  104 us [104;108]    (  26 S)
 [ 2] count:   19,  width:   52 us [52;52]      (  13 S)
 [ 3] count:    4,  width:  156 us [156;156]    (  39 S)
 [ 4] count:    5,  width:  208 us [208;208]    (  52 S)
 [ 5] count:    1,  width: 2668 us [2668;2668]  ( 667 S)
 [ 6] count:    1,  width:  260 us [260;260]    (  65 S)
 [ 7] count:    1,  width: 1352 us [1352;1352]  ( 338 S)
 [ 8] count:    1,  width:   76 us [76;76]      (  19 S)
Level estimates [high, low]:   1456,     11
RSSI: -10.5 dB SNR: 21.2 dB Noise: -31.7 dB
Frequency offsets [F1, F2]:    6032,  -7744     (+23.0 kHz, -29.5 kHz)
Guessing modulation: Non Return to Zero coding (Pulse Code)
Attempting demodulation... short_width: 52, long_width: 52, reset_limit: 53248, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=52,l=52,r=53248'
[pulse_slicer_pcm] Analyzer Device
codes     : {144}0000000000001841cf0aea1a000000749ea2
```
You can see this by running rtl_433 -A against the sample below:
[g014_433.92M_250k.zip](https://github.com/user-attachments/files/18645351/g014_433.92M_250k.zip)

this prompted me to run the command ```sudo rtl_433 -X 'n=name,m=FSK_PCM,s=52,l=52,r=53248'``` all night, which I got the results below:

{"time" : "2025-02-02 19:49:16", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "0000000000001841d010660c000000a69eea8"}], "codes" : ["{145}0000000000001841d010660c000000a69eea8"]}
{"time" : "2025-02-02 19:49:17", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 144, "data" : "0000000000001841d010660c000000a69eea"}], "codes" : ["{144}0000000000001841d010660c000000a69eea"]}
{"time" : "2025-02-02 19:49:20", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 144, "data" : "0000000000001841d010660c000000a69eea"}], "codes" : ["{144}0000000000001841d010660c000000a69eea"]}
{"time" : "2025-02-02 19:59:39", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 143, "data" : "00000000000030839e15d418000001513904"}], "codes" : ["{143}00000000000030839e15d418000001513904"]}
{"time" : "2025-02-02 19:59:39", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "ffffffffffff9841cf0aea0c000000a89c838"}], "codes" : ["{145}ffffffffffff9841cf0aea0c000000a89c838"]}
{"time" : "2025-02-02 19:59:40", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "ffffffffffff9841cf0aea0c000000a89c838"}], "codes" : ["{145}ffffffffffff9841cf0aea0c000000a89c838"]}
{"time" : "2025-02-02 19:59:40", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "00ffffffffff9841cf0aea0c000000ae9ea60"}], "codes" : ["{145}00ffffffffff9841cf0aea0c000000ae9ea60"]}
{"time" : "2025-02-02 19:59:40", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "fffff80000001841cf0aea0c000000aa9ea78"}], "codes" : ["{145}fffff80000001841cf0aea0c000000aa9ea78"]}
{"time" : "2025-02-02 19:59:40", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "ffffffffffff9841cf0aea0c000000aa9ea78"}], "codes" : ["{145}ffffffffffff9841cf0aea0c000000aa9ea78"]}
{"time" : "2025-02-02 19:59:41", "model" : "tpms", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 145, "data" : "ffffffffffff9841cf0aea0c000000aa9ea70"}], "codes" : ["{145}ffffffffffff9841cf0aea0c000000aa9ea70"]}
[... 13 lines cut ...]
[... cut ...]
2025-02-02 19:59:43	145	0000000000001841ce0aea0c000000aa9ea78
2025-02-02 19:59:45	144	0000000000001841cf0aea0c000000aa9ea7
2025-02-02 19:59:46	144	0000000000001841cf0aea0c000000aa9ea7
2025-02-02 19:59:46	144	0000000000001841cf0aea0c000000aa9ea7
2025-02-02 19:59:47	145	0000000000001841cf0aea1a1fffffaa9ecf0
2025-02-02 19:59:47	145	0000000000001841cf0aea1a000000aa9ecf8
2025-02-02 19:59:47	144	0000000000001841cf0aea1a000000aa9ece

Now I guess the next step is to decode the data, unless others feel different.

ksaye, 2025-02-03:
I should have added, letting lots of air out of just one tire, using ```sudo rtl_433 -X 'n=name,m=FSK_PCM,s=52,l=52,r=53248'``` produced the following:

sudo rtl_433 -X 'n=name,m=FSK_PCM,s=52,l=52,r=53248'
rtl_433 version 24.10-48-g7d4e1d2b branch master at 202501311749 inputs file rtl_tcp RTL-SDR
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 915, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:18
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 0000000000001841cf0aea0c000000829abd8
codes     : {145}0000000000001841cf0aea0c000000829abd8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:18
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 146          data      : ffffffffffffcc20e7857506000000414d5e4
codes     : {146}ffffffffffffcc20e7857506000000414d5e4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:18
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : ffffffffffffcc20e7857506000000414d5e8
codes     : {145}ffffffffffffcc20e7857506000000414d5e8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:18
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 0000000000001841cf0aea0c000000829abc0
codes     : {145}0000000000001841cf0aea0c000000829abc0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:19
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 146          data      : ffffffffffffcc20e7857506000000414d5e4
codes     : {146}ffffffffffffcc20e7857506000000414d5e4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:19
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 0000000000001841cf0aea0c000000829abd0
codes     : {145}0000000000001841cf0aea0c000000829abd0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:19
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 0000000000001841cf0aea0c000000829abc8
codes     : {145}0000000000001841cf0aea0c000000829abc8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:21
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 0000000000001841cf0aea0c000000869ae80
codes     : {145}0000000000001841cf0aea0c000000869ae80
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:21
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 146          data      : ffffffffffffcc20e7857506000000434d740
codes     : {146}ffffffffffffcc20e7857506000000434d740
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:21
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 145          data      : ffffffffffff9841cf0aea0c000000869ae80
codes     : {145}ffffffffffff9841cf0aea0c000000869ae80
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:22
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 146          data      : ffffffffffffcc20e7857506000000454d0ac
codes     : {146}ffffffffffffcc20e7857506000000454d0ac
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:22
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 144          data      : 0000000000001841cf0aea0c0000008a9a14
[... cut ...]
len       : 146          data      : ffffffffffffcc20e785750d000000564e5fc
codes     : {146}ffffffffffffcc20e785750d000000564e5fc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-01 18:17:56
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 146          data      : ffffffffffffcc20e785750d000000564e5f8
codes     : {146}ffffffffffffcc20e785750d000000564e5f8

e100, 2025-02-05:
@zuckschwerdt 

After looking at some existing decoders and a little help from AI I'm close to having a working decoder and will submit a pull request in the coming days.
Already have it outputting good data:
```
{"time" : "2025-02-05 08:15:24", "model" : "GM TPMS", "type" : "TPMS", "id" : "784909", "status" : "0x4D93", "learn_mode" : 0, "battery_ok" : 1, "pressure_kPa" : 169.776, "pressure_PSI" : 24.624, "temperature_C" : 22.000, "mic" : "SumMod256"}
```

@ksaye 
I've looked over your data and so far not figured anything out.
If you happen to live anywhere near central Ohio I'd be happy to meet up and see if we can get it figured out.

ksaye, 2025-02-05:
@e100 , great work so far and LOVE the use of AI to help with the decoder.  :)

gdt, 2025-07-01:
Where are we on this?  Are we close to having a working decoder :-) ?

ksaye, 2025-07-01:
I was never able to get it to work for me.

You are receiving this email because you authored the thread.