merbanan/rtl_433 #3128: Add support for generic button codes

WouterGritter, 2024-12-29:
Hi there,

I recently purchased a [self-cloning 433 MHz button remote](https://www.aliexpress.com/item/1005007716075996.html), with the intention to use it with rtl_433 to perform an action when a button is pressed. This remote is very picky about which codes it can replay, and using a 433 MHz transmitter and an arduino, I was able to transmit some random button codes for the remotes to learn using the rc-switch library. Unfortunately, rtl_433 does not support these button codes out of the box, even though they seem to be very widely used.

In the [source code of rc-switch](https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L80), these protocols of encoding a button code seems fairly trivial to decode. Using a custom/flex decoder I was able to decode the button codes, but every time I analyze my newly saved 433 data I get slightly different values, and it does not read the button presses as reliably as I would have expected.

I lack the knowledge required to add proper support for these encodings to this project, but I think it would be really valuable to have support for them. Transmitting the various widely-used protocols can be easily done using an arduino and the library I linked to above.

It should be noted that protocol "2" of rc-switch is able to be picked up by rtl_433 automatically it seems, though it seems like my self-copying remote can't retransmit the data which is unfortunate.

If needed I can fiddle around with sending some random codes on my arduino and upload the data here, though it can be easily done using some [very cheap 433 MHz transmitters](https://nl.aliexpress.com/item/1005002288941680.html).

zuckschwerdt, 2024-12-29:
Since there is no structured data and no checksum a flex decoder is indeed the best choice.

Unstable and bad readings might be due to antenna overloading, try a greater distance or remove the antenna if the sender is close to the receiver.
You can use a captured .cu8 sample on the https://triq.org/spectrogram-next/ demo with scroll-zoom to inspect the waveform, clipping is shown in red.

WouterGritter, 2024-12-29:
Thanks for the interesting link. I'll continue using a flex decoder in that case.

I wasn't able to figure out the values I should use with the flex decoder when looking at the [timing and waveform/pulse patterns](https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L80) rc-switch produces. I feel like I might get better results when the flex decoder is "tuned" correctly, is this something you could give me some guidance on? I am using the first protocol.

I believe my antenna setup is working just fine. It's an omnidirectional star antenna mounted outside, and it's picking up other sensors inside a building which is about ~80-100 meters away.

zuckschwerdt, 2024-12-29:
For `{ 350, {  1, 31 }, {  1,  3 }, {  3,  1 }, false },    // protocol 1` it would be
short=350, long=1050, gap=1500, reset=12000 -- this will read all packets as one transmission and likely add the sync pulse as last bit. Otherwise use no gap and reset=1500

WouterGritter, 2025-01-02:
Thank you! I'll close the issue, and a happy new year.

closed 2025-01-02 by WouterGritter