merbanan/rtl_433 #3130: Govee H5054 falsely detected as Smoke detector GS 558

jgdf42, 2024-12-31:
I bought a two pack of these and they're both detected as the wrong device. Seen in MQTT Explorer upon button press and wet triggering;

{"time":"2024-12-31 01:25:35","model":"Smoke detector GS 558","id":25941,"unit":28,"learn":0,"code":"0caabc"}
{"time":"2024-12-31 01:27:34","model":"Smoke detector GS 558","id":25590,"unit":12,"learn":0,"code":"0c7ecc"}

```
pi@rtlsdrPI:~ $ rtl_433 -v
rtl_433 version -129-NOTFOUND branch master at 202412301727 inputs file rtl_tcp RTL-SDR
[Protocols] Registered 233 out of 270 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-47 49-60 63 67-71 73-85 87-100 102-105 108-116 119-122 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-232 234-241 243-244 246-247 249-259 261-269
[Input] The internals of input handling changed, read about and report problems on PR #1978
```

My rtl_433.conf. I'm not limiting to only certain bands, unsure if this matters - I only use this for Simplisafe v2 and these Govee.

```

[program:rtl_433]
command=/home/pi/rtl_433/build/src/rtl_433 -F "mqtt://192.168.1.30:1883,user="mqtt",pass="pass",events=Simplisafe"
user=pi
autostart=yes
autorestart=yes
startretries=100
```

Also, is there a battery operated 433mhz moisture sensor that is more reliable than Govee with this project?

ProfBoc75, 2024-12-31:
@jgdf42 : Smoke GS 558 is disabled by default to avoid false positives with generic EV1527 protocol/devices like simple rf remote control. Looks like your device is using similar protocol. You have to create your own conf file to decode your sensor.
Some EV1527 examples into [conf folder](https://github.com/merbanan/rtl_433/tree/master/conf)

zuckschwerdt, 2024-12-31:
Also [search for H5054 in the issues](https://github.com/merbanan/rtl_433/issues?q=H5054). There are multiple threads about problems with different version of the H5054 and such.

jgdf42, 2024-12-31:
> Also [search for H5054 in the issues](https://github.com/merbanan/rtl_433/issues?q=H5054). There are multiple threads about problems with different version of the H5054 and such.

Sounds like I should just return these. I was looking at the table of supported devices and couldn't find a suitable replacement, though.

zuckschwerdt, 2024-12-31:
I don't remember exactly but I think there was some effort to also support the newer versions. There might be one unsupportable or broken variant though?

closed 2025-01-01 by jgdf42