merbanan/rtl_433 #3136: Support for wireless freezer refrigerator thermometer model "WA150KM"

janiwe, 2025-01-05:
Hello everyone,

I recently got a 433MHz freezer/refrigerator thermometer from [Amazon](https://www.amazon.de/dp/B0D7Q31Z57) branded as "ORIA" with model name "WA150KM" and tried to read it with rtl_433. Unfortunately it's an unknown device still. 
The transmitter is labeled with FCC ID "2AZUG-WA144" which has some [FCC data](https://fcc.report/FCC-ID/2AZUG-WA144), but nothing regarding the encoding.
I captured samples and tabled the temperature readings on the headunit, but I'm struggling with the analysis.
I analyzed the samples with rtl_433 -A _sample.cu8_ and it came up with a rather consistent flex decoder config 
`-X 'n=name,m=OOK_PWM,s=488,l=976,r=3948,g=0,t=0,y=304'`, which I modified to 
`-X 'n=WA144,m=OOK_PWM,s=488,l=976,r=8000,g=0,t=0,y=304'` to keep the preamble in the message payload, but I'm unable to read anything out of the codes. I'm not even sure if the flex decoder config is right - still a newbie regarding this stuff unfortunately.

Capture log (HW: Nooelec RTL-SDR v5 SDR without antenna, receiver about 5cm away):
```
> rtl_433 -S unknown                                               
rtl_433 version 24.10 (2024-10-30) inputs file rtl_tcp RTL-SDR with TLS
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
*** Saving signal to file g001_433.92M_250k.cu8 (128385 samples, 262144 bytes)
*** Saving signal to file g002_433.92M_250k.cu8 (128385 samples, 262144 bytes)
*** Saving signal to file g003_433.92M_250k.cu8 (128400 samples, 262144 bytes)
[...]
```
[WA144_Samples.zip](https://github.com/user-attachments/files/18310037/WA144_Samples.zip)
[Sample_Values.csv](https://github.com/user-attachments/files/18310041/Sample_Values.csv)

Can anyone give me a hint on how to proceed here?

Thanks a lot!

Jan Niklaas

zuckschwerdt, 2025-01-05:
Samples look great, it's OOK, a warmup of ~10x 490 µs pulse / 2932 µs gap, then three packets with a 2980 µs packet gap of 24x preamble 490 / 490 µs, then 490 µs Manchester code. I.e.
`-X 'n=name,m=OOK_PCM,s=490,l=490,g=1500,r=4000'` for raw data and
`-X 'n=name,m=OOK_MC_ZEROBIT,s=490,l=490,g=1500,r=4000'` for Manchester decoded data.
But that won't work well, `-X 'n=name,m=OOK_MC_ZEROBIT,s=490,l=490,r=1500'` is somewhat better.

<details>

<summary>Use this collected PCM data</summary>

```
{227}aaaaaaaaaaaa66aa5665a655595a59966565559555665599a5a699690[g001]
{227}aaaaaaaaaaaa66aa6665a655595a5996656555955566659999a699690
{227}aaaaaaaaaaaa66aa6665a655595a5996656555955566659999a69969
{227}aaaaaaaaaaaa66aa5565a655595a59966565559555665659a66699690[g002]
{227}aaaaaaaaaaaa66aa6565a655595a599665655595556666599a6699690
{227}aaaaaaaaaaaa66aa6565a655595a599665655595556666599a669969
{227}aaaaaaaaaaaa66aa5665a655595a5955a565559555665659555699690[g003]
{227}aaaaaaaaaaaa66aa6665a655595a5955a565559555666659695699690
{227}aaaaaaaaaaaa66aa6665a655595a5955a56555955566665969569969
{227}aaaaaaaaaaaa66aa5565a655595a5965a56555955566655956aa99690[g004]
{227}aaaaaaaaaaaa66aa6565a655595a5965a5655595556659596aaa99690
{227}aaaaaaaaaaaa66aa6565a655595a5965a5655595556659596aaa9969
{227}aaaaaaaaaaaa66aa5665a655595a5999a565559555669a5955a999690[g005]
{227}aaaaaaaaaaaa66aa6665a655595a5999a56555955566aa5969a999690
{227}aaaaaaaaaaaa66aa6665a655595a5999a56555955566aa5969a99969
{227}aaaaaaaaaaaa66aa5565a655595a59555965559555669559959599690[g006]
{227}aaaaaaaaaaaa66aa6565a655595a5955596555955566a559a99599690
{227}aaaaaaaaaaaa66aa6565a655595a5955596555955566a559a9959969
{227}aaaaaaaaaaaa66aa5665a655595a59595965559555669a59965a99690[g007]
{227}aaaaaaaaaaaa66aa6665a655595a5959596555955566aa59aa5a99690
{227}aaaaaaaaaaaa66aa6665a655595a5959596555955566aa59aa5a9969
{227}aaaaaaaaaaaa66aa5565a655595a59965965559555666659956699690[g008]
{227}aaaaaaaaaaaa66aa6565a655595a59965965559555665a59a96699690
{227}aaaaaaaaaaaa66aa6565a655595a59965965559555665a59a9669969
{227}aaaaaaaaaaaa66aa5665a655595a59a59965559555669a59669a99690[g009]
{227}aaaaaaaaaaaa66aa6665a655595a59a5996555955566aa595a9a99690
{227}aaaaaaaaaaaa66aa6665a655595a59a5996555955566aa595a9a9969
{227}aaaaaaaaaaaa66aa5565a655595a59a99965559555669659655599690[g010]
{227}aaaaaaaaaaaa66aa6565a655595a59a9996555955566a659595599690
{227}aaaaaaaaaaaa66aa6565a655595a59a9996555955566a65959559969
{227}aaaaaaaaaaaa66aa5665a655595a5955696555955566a659aa5599690[g011]
{227}aaaaaaaaaaaa66aa6665a655595a59556965559555669a59965599690
{227}aaaaaaaaaaaa66aa6665a655595a59556965559555669a5996559969
{227}aaaaaaaaaaaa66aa5565a655595a59a56965559555666959a95999690[g012]
{227}aaaaaaaaaaaa66aa6565a655595a59a56965559555665659955999690
{227}aaaaaaaaaaaa66aa6565a655595a59a5696555955566565995599969
{227}aaaaaaaaaaaa66aa5665a655595a59696965559555669599aa6699690[g013]
{227}aaaaaaaaaaaa66aa6665a655595a5969696555955566a599966699690
{227}aaaaaaaaaaaa66aa6665a655595a5969696555955566a59996669969
{227}aaaaaaaaaaaa66aa5565a655595a5956696555955566a659a99699690[g014]
{227}aaaaaaaaaaaa66aa6565a655595a59566965559555669a59959699690
{227}aaaaaaaaaaaa66aa6565a655595a59566965559555669a5995969969
{227}aaaaaaaaaaaa66aa5665a655595a5995a965559555669a595aa699690[g015]
{227}aaaaaaaaaaaa66aa6665a655595a5995a96555955566aa5966a699690
{227}aaaaaaaaaaaa66aa6665a655595a5995a96555955566aa5966a69969
{227}aaaaaaaaaaaa66aa5565a655595a59a5a96555955566a959595a99690[g016]
{227}aaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a99690
{227}aaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a9969
{227}aaaaaaaaaaaa66aa5665a655595a5999a9655595556695995aa999690[g017]
{227}aaaaaaaaaaaa66aa6665a655595a5999a96555955566a59966a999690
{227}aaaaaaaaaaaa66aa6665a655595a5999a96555955566a59966a99969
{227}aaaaaaaaaaaa66aa5565a655595a59a9a96555955566a659595599690[g018]
{227}aaaaaaaaaaaa66aa6565a655595a59a9a965559555669a59655599690
{227}aaaaaaaaaaaa66aa6565a655595a59a9a965559555669a5965559969
{227}aaaaaaaaaaaa66aa5665a655595a5996a9655595556699995aa599690[g019]
{227}aaaaaaaaaaaa66aa6665a655595a5996a96555955566a99966a599690
{227}aaaaaaaaaaaa66aa6665a655595a5996a96555955566a99966a59969
{227}aaaaaaaaaaaa66aa5565a655595a5995566555955566695999a599690[g020]
{227}aaaaaaaaaaaa66aa6565a655595a59955665559555665659a5a599690
{227}aaaaaaaaaaaa66aa6565a655595a59955665559555665659a5a59969
[... 28 lines cut ...]
{227}aaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a99690
{227}aaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a9969
{227}aaaaaaaaaaaa66aa9665a655595a59a5a965559555665599aa9999690[g031]
{227}aaaaaaaaaaaa66aaa665a655595a59a5a965559555666599969999690
{227}aaaaaaaaaaaa66aaa665a655595a59a5a96555955566659996999969
{227}aaaaaaaaaaaa66aa5565a655595a5995a965559555669959596699690[g032]
{227}aaaaaaaaaaaa66aa6565a655595a5995a96555955566a959656699690
{227}aaaaaaaaaaaa66aa6565a655595a5995a96555955566a95965669969
{227}aaaaaaaaaaaa66aa9665a655595a5995a965559555666a59aaa599690[g033]
{227}aaaaaaaaaaaa66aaa665a655595a5995a96555955566559996a599690
{227}aaaaaaaaaaaa66aaa665a655595a5995a96555955566559996a59969
{227}aaaaaaaaaaaa66aa5565a655595a59966965559555665a59a96699690[g034]
{227}aaaaaaaaaaaa66aa6565a655595a59966965559555666a59956699690
{227}aaaaaaaaaaaa66aa6565a655595a59966965559555666a5995669969
{227}aaaaaaaaaaaa66aa9665a655595a599669655595556699995aa599690[g035]
{227}aaaaaaaaaaaa66aaa665a655595a5996696555955566a99966a599690
{227}aaaaaaaaaaaa66aaa665a655595a5996696555955566a99966a59969
{227}aaaaaaaaaaaa66aa5565a655595a5956696555955566a659a99699690[g036]
{227}aaaaaaaaaaaa66aa6565a655595a59566965559555669a59959699690
{227}aaaaaaaaaaaa66aa6565a655595a59566965559555669a5995969969
```

</details>

And paste it in a [BitBench](https://triq.org/bitbench/) with decoding set to Manchester.

zuckschwerdt, 2025-01-05:
There is a nibble sum, then another checksum byte and a fixe 0xa6 value at the end.
The temperature is a field in the middle, values are bit-reflected. See [this BitBench](https://triq.org/bitbench/#c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a59966565559555665599a5a699690%5Bg001%20Ch1%20N%2022%2C9%2073%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5996656555955566659999a699690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5996656555955566659999a69969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59966565559555665659a66699690%5Bg002%20Ch1%20N%2022%2C9%2073%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a599665655595556666599a6699690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a599665655595556666599a669969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5955a565559555665659555699690%5Bg003%20Ch1%20N%2023%2073%2C4%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5955a565559555666659695699690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5955a56555955566665969569969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5965a56555955566655956aa99690%5Bg004%20Ch1%20N%2023%2C2%2073%2C7%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a5965a5655595556659596aaa99690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a5965a5655595556659596aaa9969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5999a565559555669a5955a999690%5Bg005%20Ch1%20N%2023%2C5%2074%2C3%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5999a56555955566aa5969a999690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5999a56555955566aa5969a99969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59555965559555669559959599690%5Bg006%20Ch1%20N%2024%2075%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a5955596555955566a559a99599690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a5955596555955566a559a9959969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a59595965559555669a59965a99690%5Bg007%20Ch1%20N%2024%2C4%2075%2C9%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5959596555955566aa59aa5a99690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5959596555955566aa59aa5a9969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59965965559555666659956699690%5Bg008%20Ch1%20N%2024%2C9%2076%2C8%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59965965559555665a59a96699690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59965965559555665a59a9669969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a59a59965559555669a59669a99690%5Bg009%20Ch1%20N%2025%2C3%2077%2C5%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59a5996555955566aa595a9a99690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59a5996555955566aa595a9a9969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59a99965559555669659655599690%5Bg010%20Ch1%20N%2025%2C7%2078%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a9996555955566a659595599690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a9996555955566a65959559969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5955696555955566a659aa5599690%5Bg011%20Ch1%20N%2026%2078%2C8%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59556965559555669a59965599690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59556965559555669a5996559969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59a56965559555666959a95999690%5Bg012%20Ch1%20N%2026%2C3%2079%2C3%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a56965559555665659955999690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a5696555955566565995599969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a59696965559555669599aa6699690%5Bg013%20Ch1%20N%2026%2C6%2079%2C8%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5969696555955566a599966699690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5969696555955566a59996669969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5956696555955566a659a99699690%5Bg014%20Ch1%20N%2026%2C8%2080%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59566965559555669a59959699690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59566965559555669a5995969969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5995a965559555669a595aa699690%5Bg015%20Ch1%20N%2027%2C1%2080%2C7%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5995a96555955566aa5966a699690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5995a96555955566aa5966a69969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59a5a96555955566a959595a99690%5Bg016%20Ch1%20N%2027%2C3%2081%2C1%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a99690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a5a965559555669659655a9969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5999a9655595556695995aa999690%5Bg017%20Ch1%20N%2027%2C5%2081%2C5%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5999a96555955566a59966a999690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5999a96555955566a59966a99969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59a9a96555955566a659595599690%5Bg018%20Ch1%20N%2027%2C7%2081%2C8%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a9a965559555669a59655599690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a9a965559555669a5965559969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5996a9655595556699995aa599690%5Bg019%20Ch1%20N%2027%2C9%2082%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5996a96555955566a99966a599690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a5996a96555955566a99966a59969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5995566555955566695999a599690%5Bg020%20Ch1%20N%2028%2C1%2082%2C5%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59955665559555665659a5a599690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59955665559555665659a5a59969&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a59a556655595556655999a5999690%5Bg021%20Ch1%20N%2028%2C3%2082%2C9%5D&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59a55665559555666599a65999690&c=%7B227%7Daaaaaaaaaaaa66aa6665a655595a59a55665559555666599a6599969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a59a55665559555665659999999690%5Bg022%20Ch1%20N%2028%2C3%2082%2C9%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59a55665559555666659a59999690&c=%7B227%7Daa [... cut ...] %7B227%7Daaaaaaaaaaaa66aaa665a655595a5996696555955566a99966a599690&c=%7B227%7Daaaaaaaaaaaa66aaa665a655595a5996696555955566a99966a59969&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5956696555955566a659a99699690%5Bg036%20Ch1%20N%2026%2C8%2080%2C2%5D&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59566965559555669a59959699690&c=%7B227%7Daaaaaaaaaaaa66aa6565a655595a59566965559555669a5995969969&f=WARM%3A24h%20%5E8h%5E8h%20%3F%5E8h%5E8h%20TEMP%5E16h%20%3F%5E8h%5E8h%20SUM%3A8h%20CHK%3F8h%20FIXED%3A8h&z=1&d=MC&cw=4).

janiwe, 2025-01-05:
Wow, you literally decoded this in half an hour!
I'll try to understand that and capture some more samples to come up with explanations for the rest of the data.

Thanks so much!
Jan Niklaas

klohner, 2025-01-09:
@janiwe I took a stab at a flex decoder in case this helps you gather data.  There's no direct flex support for the LSB BCD values in this signal, but I put in a brief lookup table to get a few integer values.

```
# WA150KM Thermometer
# https://github.com/merbanan/rtl_433/issues/3136
decoder {
  name         = 2AZUG-WA144_Thermometer,
  modulation   = OOK_PCM,
  short        = 488,
  long         = 488,
  reset        = 1500,
  preamble     = {32}aaaa66aa,
  symbol_zero  = {2}4,
  symbol_one   = {2}8,
  get          = unknown:@0:{8}:%02x,
  get          = temp_bcd_lsb:@24:{16}:%04x,
  get          = temp_c:@32:{8}:[68:22 196:23 36:24 164:25 100:26 228:27 20:28],
  get          = sum:@56:{8}:%02x,
  get          = chk:@64:{8}:%02x,
}

```

janiwe, 2025-01-11:
Hi, I was able to collect some more samples and identified a few more fields in the data:
[Bitbench](https://triq.org/bitbench/#c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5656a565559555665999599599690%20%5B23.8%20Low%20battery%20level%2C%20Device%201%5D&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5956a565559555665599555599690%20%5B23.8%20Normal%20battery%20level%2C%20Device%201%5D&c=%7B227%7Daaaaaaaaaaaa66aa5665a6555aa55a56a56555955566a999559599690%20%5B23.8%20Device%202%5D&c=&c=%7B227%7Daaaaaaaaaaaa66aa5565a6555aa55aa9a565559555666a59569599690%20%5B23.7%20Channel%201%5D&c=%7B227%7Daaaaaaaaaaaa66aa5565a6955aa55aa9a56555955566aa59566599690%20%5B23.7%20Channel%202%5D&c=&c=%7B227%7Daaaaaaaaaaaa66aa9565a6555aa55aa9a56555955566aa59a69699690%20%5B23.7%20TX%20button%5D&c=%7B227%7Daaaaaaaaaaaa66aa5665a6555aa55aa9a565559555666999555599690%20%5B23.7%20Normal%20send%5D&c=&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5a555595569555665a599a5999690%20%5B-10.0%5D&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5a9956555695556655699a5699690%20%5B-8.5%5D&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5a69695556955566a999a99a99690%20%5B-6.6%5D&c=%7B227%7Daaaaaaaaaaaa66aa5565a655595a5a59555556955566aa599aa699690%20%5B-0.4%5D&c=%7B227%7Daaaaaaaaaaaa66aa5665a655595a5aa5955555955566aa596a6699690%20%5B1.3%5D&f=WARM%3A24h%20MSG_TYPE%3F%3A%5E8h%5E8h%20%3F4h%20CH%3A%5E4h%20DEVICE_ID%3A%5E8h%20TEMP%3A%5E16h%20NEG%3A%5E8h%20FIXED%3A%5E8h%20SUM%3A8h%20CHK%3F8h%20FIXED%3A8h&z=1&d=MC&cw=4)

Some observations:
- In normal operation, the MSG_TYPE toggles between fa20 and fa28 with every send (interval is ~34 seconds)
- Forced transmissions with the TX button have a MSG_TYPE=fa21
- DEVICE_IDs stay consistent over powercycles
- The devices transmit a "battery low" signal encoded in the byte after the temperature
- Negative temperatures have another single bit set

Best
Jan Niklaas

janiwe, 2025-01-12:
Hi, I spent some time writing a basic decoder. It does not respect the battery low signals or message types, but it seems to work and put out correct temperature readings.
See above PRs, would you have a look and consider a merge?
Cheers,
Jan Niklaas

AJolly, 2025-09-06:
What would I need to do to test this?  I have a "KeeKit Fridge Thermometer", same FCC ID, it looks identical.  

[g003_433.92M_250k.zip](https://github.com/user-attachments/files/22184644/g003_433.92M_250k.zip)

 I'm probably doing something wrong, but I can't get it to decode.

AJolly, 2025-09-15:
Took me forever to figure out how to get this compiled and running under home assistant, but it does seem to work correctly. 
The oen change I'd make is changing temperature to temperature_C so it's properly parsed from rtl_433 MQTT auto discovery.

merbanan, 2025-09-17:
You need to log all messages and their decode values until we find the bad decodes. Then we might be able to filter it out. But right now it does not seem to have a checksum. We need messages to figure that out.

AJolly, 2025-09-19:
Is that the -F logging or -S known or something else?  I'm happy to run the logging, but im not quite sure which option to use.  Is there an signal save option for a specific decoder?