merbanan/rtl_433 #3138: Support for Typhur Sync Gold Dual/Quad

kevin-david, 2025-01-05:
I started looking into how to add support for this meat thermometer. I have the dual version, so the  captures I had were from both probes running at once. These have 6 different temperature sensors (5 internal & one ambient) per probe so it ought to be fun to fully decode.

Here's what I've gathered so far: [typhur-sync-gold-samples.zip](https://github.com/user-attachments/files/18311149/typhur-sync-gold-samples.zip). I'm not positive, but based just visual observation from a few of these files at https://triq.org/pdv/, I think there is some other 915MHz-band traffic flying around too.

Those were captured with temperatures around ~210-230F for the Ambient Sensors, and ~140F-165F Internal Temperatures for the internal probe sensors (next time I take a stab at this I'll get more exact)

Some things that looked pattern-y with `rtl_433 -A -r`, especially for files in the `auto-decoded` folder. The suggested decoder was `rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=12,l=12,r=12288' *.cs16`

```
{280}d5555555555555552baa10df00507a2bb00df18dc40da80d9c8da1820800ed8546de00
{279}00000000000000002baa10df00507a2bb20df18dcb8dad0da18da1820800ef8514a83e
{279}0000000000000000035461b100a0f457cb1e711ccb1cfe1c341d0d040101f90a8f7f02
{280}fd555555555555552baa10df00507a2bb20df18dcc0daf8da40da1820800f18559d580
{280}00000000000000002baa30d880507a2be60f388e658e7f0e1a8e87020080fe850eb580
```

Maybe `preamble=0507a2b` based on the above? But wasn't obvious to me why those earlier bits were changing.

<hr/>

Product link: https://www.typhur.com/products/sync-gold-dual or https://www.typhur.com/products/sync-gold-quad

FCC Info - I _think_ this is for the probe: https://fccid.io/2BEFM-WTP3000. This is for the base: https://fccid.io/2BEFM-WT1000R

These also have Bluetooth and Wifi support (with some kind of hosted service?), so it would be possible but I figured the RF data would be fun to start with.

zuckschwerdt, 2025-01-05:
It seems there are three different senders in the samples, the slower signal at 915.0M, e.g. g021_915M_1000k.cs16, then the fast signal at 915.0M, e.g. g005_915M_1000k.cs16 and lastly the slow signal at 914.6M, e.g. g009_915M_1000k.cs16. And some others too like the first one in g015_915M_1000k.cs16.

You need to narrow it down in a quieter environment or without antenna and close to the sender.

12 µs sounds awefully fast for just a temperature probe.

zuckschwerdt, 2025-08-31:
@hakong thanks for all the detailed info! Basically you found the format to be
`SYNC_WORD:48h TYPE:8h SUB:8h TEMP1:16h TEMP2:16h TEMP3:16h TEMP4:16h TEMP5:16h T_AMB:8h BATT:8h FLAG:8h COUNTER:24h CRC:8h`, right?
What values did you get for SYNC_WORD:, TYPE, and SUB?

hakong, 2025-08-31:
These are just guesses. To confirm I have yet to pick up my 2nd probe and compare them.

- sync word/probe id: TODO: distginguish between the probes (i suspect the sync word is a form of probe ID/serial, since the app has support for multiple probes and "pairing" them. i can see my probe's serial number in the app)
- type/subtype: TODO: see if there is any difference between the two probes, and see if there is a difference between packets when the probe is charging/discharging.
- temp1-5: done
- temp ambient: TODO: figure out the encoding
- batt: just a guess. TODO: let the probes battery drain to 50% and check what values I get
- flag: just a guess. TODO: compare between 2 probes if any difference
- counter: looks like a counter, at least increments by 32 every 2 brodcasts if decoded in little endian 15 bit MSB masked. TODO: try other decodings
- CRC: just a guess, shouldn't there be a CRC at the end of brodcasts like this?

gdt, 2025-09-18:
I don't think it's ok to ask people to clean up LLM code.   Plus, most (all) LLMs are trained on stolen content and thus don't comply with Free Software licensing norms.

hakong, 2025-09-18:
Yes, I agree. Thank you for the comments. I know LLM code is at controversial and maybe rarely usable on its own. That's why I was transparent about it.
I don't want to push code on someone else and tell them its mine when it's written by an LLM. Especially so when it's in a language that I'm not proficient in. Again, that's exactly why I am transparent about it.

This was more of a draft/demo I thought was a non-zero chance it would be at least in some part usable (since it does compile and works).

```
rtl_433 version 25.02-47-g5827699d branch feat-typhur at 202509092028 inputs file rtl_tcp RTL-SDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-09-13 09:31:15
model     : Typhur-SyncGold                        id        : 2467328
in_base   : 0            unknown1_hex: 0xA0        unknown2_hex: 0x57        counter   : 12            s         : 24            on_since_approx: 2025-09-13T09:30:51Z               V         : 2.830         C         : 23.240        C         : 23.390
C         : 23.380       C         : 23.340        C         : 23.380        C         : 24.900        message   : base=0 cnt=12 up=24s core_min=23.24C core_max=23.39C
codes     : 25A600A0F4571409230922091E092209F9001B010C00E210                 mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-09-13 09:31:17
model     : Typhur-SyncGold                        id        : 2467328
in_base   : 0            unknown1_hex: 0xA0        unknown2_hex: 0x57        counter   : 13            s         : 26            on_since_approx: 2025-09-13T09:30:51Z               V         : 2.830         C         : 23.240        C         : 23.390
C         : 23.380       C         : 23.340        C         : 23.380        C         : 24.900        message   : base=0 cnt=13 up=26s core_min=23.24C core_max=23.39C
codes     : 25A600A0F4571409230922091E092209F9001B010D006413                 mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-09-13 09:31:19
model     : Typhur-SyncGold                        id        : 2467328
in_base   : 0            unknown1_hex: 0xA0        unknown2_hex: 0x57        counter   : 14            s         : 28            on_since_approx: 2025-09-13T09:30:51Z               V         : 2.830         C         : 23.240        C         : 23.390
C         : 23.380       C         : 23.340        C         : 23.380        C         : 24.900        message   : base=0 cnt=14 up=28s core_min=23.24C core_max=23.39C
codes     : 25A600A0F4571409230922091E092209F9001B010E006E13                 mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-09-13 09:31:21
model     : Typhur-SyncGold                        id        : 2467328
in_base   : 0            unknown1_hex: 0xA0        unknown2_hex: 0x57        counter   : 15            s         : 30            on_since_approx: 2025-09-13T09:30:51Z               V         : 2.830         C         : 23.240        C         : 23.390
C         : 23.380       C         : 23.340        C         : 23.380        C         : 24.900        message   : base=0 cnt=15 up=30s core_min=23.24C core_max=23.39C
codes     : 25A600A0F4571409230922091E092209F9001B010F00E810                 mic       : CRC

```

merbanan, 2025-09-18:
> I don't think it's ok to ask people to clean up LLM code. Plus, most (all) LLMs are trained on stolen content and thus don't comply with Free Software licensing norms.

I had a thought and it is hard to clearly assign copyright to LLM generated code. But derived code could possibly be licensed under the GPL. In this case the LLM code used a couple of api calls and must thus be derived and is then covered under the GPL license.

But I think we should adopt the policy of rejecting LLM contributions until there is a general consensus about LLM code among other open source projects. I think it would be safe to adopt anything that is also accepted by the Linux kernel (but we dont have to).

The PoC is impressive though but you have a very structured problem and a large set of decoder to train a LLM on. You can get quick results and then reference the code when re-implementing it. It would be interesting if a LLM could be made to decode a set of messages to parameters.