merbanan/rtl_433 #3140: Can't connect to triq.org - IP address blocked?

clockbrain, 2025-01-07:
I am unable to connect to triq.org. Browser just times out. I've been trying occasionally for several weeks in case it was a temporary issue but have never been able to connect. Just recently though I attempted to connect via a different network and I was able to access the site.

After much testing here are my conclusions.
No devices can connect via my house ISP connection. Tried various browsers, devices, operating systems, etc. All time out.
I can connect via alternative networks including hotspot mobile data, VPN, or my office network.
After posting the issue to my ISP support forum, most other users of that ISP can connect but several people cannot, they get the same time out issue.

So the problem seems localised to a subset of users of my ISP - Aussie Broadband. Is it possible some of our IP addresses are blocked or blacklisted somewhere?

Here is a traceroute showing the time outs.

tracert triq.org

Tracing route to triq.org [134.106.11.139]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.0.1
  2    11 ms    10 ms    10 ms  loop1801501040.bng.qld.aussiebb.net [180.150.104.1]
  3   113 ms   113 ms   112 ms  10.241.0.238
  4   121 ms   115 ms   115 ms  10.241.1.184
  5   117 ms     *      117 ms  be2.lsr2.8gif.nsw.aussiebb.net [180.150.0.160]
  6     *      117 ms     *     be2.lsr2.equinix-sy4.syd.aussiebb.net [159.196.252.106]
  7   116 ms   114 ms   116 ms  10.241.16.231
  8     *        *        *     Request timed out.
  9   295 ms   294 ms   295 ms  hbg-b2-link.ip.twelve99.net [62.115.132.109]
 10     *        *      375 ms  dfn-ic-373396.ip.twelve99-cust.net [213.248.100.167]
 11   281 ms   280 ms   280 ms  cr-tub3-iflag6-0.x-win.dfn.de [188.1.144.225]
 12   285 ms   293 ms   284 ms  cr-ham3-iflag3-0.x-win.dfn.de [188.1.144.109]
 13   288 ms   288 ms   288 ms  kr-uniold7-3.x-win.dfn.de [188.1.47.206]
 14     *        *        *     Request timed out.
 15     *        *        *     Request timed out.
 16     *        *        *     Request timed out.
 17     *        *        *     Request timed out.
 18     *        *        *     Request timed out.
 19     *        *        *     Request timed out.
 20     *        *        *     Request timed out.
 21     *        *        *     Request timed out.
 22     *        *        *     Request timed out.
 23     *        *        *     Request timed out.
 24     *        *        *     Request timed out.
 25     *        *        *     Request timed out.
 26     *        *        *     Request timed out.
 27     *        *        *     Request timed out.
 28     *        *        *     Request timed out.
 29     *        *        *     Request timed out.
 30     *        *        *     Request timed out.

Trace complete.

zuckschwerdt, 2025-01-07:
Might be a block listed entry. There is not block on `180.150.0.0/17` and also no block on AS4764 Aussie Broadband.
Is that your originating subnet?

clockbrain, 2025-01-08:
Yes, I'm on a static IP in the 180.150.x.x range.

zuckschwerdt, 2025-01-08:
There are blocks only on these (from a [FIREHOL](https://iplists.firehol.org/) list) but none have hit that server.
180.150.36.4
180.150.38.246
180.150.112.166
180.150.14.98
180.150.226.99
There is also an upstream firewall (some Palo-Alto shit) I don't control. Can you reach https://uol.de/ ?

zuckschwerdt, 2025-01-08:
Sorry, found an old entry: 180.0.0.0/8 ; 2014-04-11 APNIC heartbleed. Removed.

closed 2025-01-08 by zuckschwerdt

clockbrain, 2025-01-08:
Thanks very much for resolving this. I can get to triq.org now.