merbanan/rtl_433 #3141: latest changes to mqtts output seem to have broken

triplepoint, 2025-01-11:
I run the `master-debian` docker image.  Currently, that's image ID [ff37e3464d00](https://hub.docker.com/layers/hertzg/rtl_433/master-debian/images/sha256-1a326c32f26688ab7997c504d9070f2fc58122e1fa6810fc9cf4224bb850b4ce).

With this image, I get this error, with the container immediately exiting.
```
rtl_433                | rtl_433 version 24.10-34-gd9249e24 branch master at 202501071804 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
rtl_433                | MQTT: Publishing MQTT data to mqtt.myredactednetwork.lan port 8883 (TLS)
rtl_433                | MQTT: Publishing device info to MQTT topic "home/rtl_433/devices[/type][/model][/subtype][/channel][/id]".
rtl_433                | MQTT: Publishing events info to MQTT topic "home/rtl_433/events[/id]".
rtl_433                | MQTT: Publishing states info to MQTT topic "home/rtl_433/states".
rtl_433                | MQTT: mqtts (TLS) parameters are: tls_cert=(null) tls_key=(null) tls_ca_cert=* tls_cipher_suites=(null) tls_server_name=(null) tls_psk_identity=(null) tls_psk_key=(null) 
rtl_433                | MQTT: MQTT connect (mqtt.myredactednetwork.lan:8883) failed: cannot parse address
```

The only change was to pull the latest image and restart the container - the configuration remains unchaged from when it was previously working.  I think my previous image was from last week sometime; I'm afraid I don't have logs of which image it was.

The config file I'm passing rtl_433 (modified to remove passwords):
```
frequency 433.92M
convert si
report_meta time:unix:usec:utc
output kv
output mqtts://mqtt.myredactednetwork.lan:8883,user=some_user,pass=some_pass,retain=1,devices=home/rtl_433/devices[/type][/model][/subtype][/channel][/id],events=home/rtl_433/events[/id],states=home/rtl_433/states
```
and there's no environment variables being set in the docker compose file for that service.

Thanks in advance for any help you can provide.

triplepoint, 2025-01-11:
For more information, rolling back to the `24.10-debian` docker tag, without changing anything else about the configuration gives the following log premable, before successfully going on to detect radio messages:
```
rtl_433  | rtl_433 version 24.10 branch  at 202410301452 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
rtl_433  | MQTT: Publishing MQTT data to mqtt.myredactednetwork.lan port 8883 (TLS)
rtl_433  | MQTT: Publishing device info to MQTT topic "home/rtl_433/devices[/type][/model][/subtype][/channel][/id]".
rtl_433  | MQTT: Publishing events info to MQTT topic "home/rtl_433/events[/id]".
rtl_433  | MQTT: Publishing states info to MQTT topic "home/rtl_433/states".
rtl_433  | MQTT: mqtts (TLS) parameters are: tls_cert=(null) tls_key=(null) tls_ca_cert=* tls_cipher_suites=(null) tls_server_name=(null) tls_psk_identity=(null) tls_psk_key=(null) 
rtl_433  | Detached kernel driver
rtl_433  | Found Rafael Micro R820T tuner
rtl_433  | Exact sample rate is: 250000.000414 Hz
rtl_433  | [SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
rtl_433  | [R82XX] PLL not locked!
rtl_433  | Allocating 15 zero-copy buffers
```

zuckschwerdt, 2025-01-11:
Is your lan domain mDNS? That was never truly supported, see #2573
There was recent change, but only to the MQTT reconnection timing, #3134
I've never encountered "failed: cannot parse address", what is special in your address?

triplepoint, 2025-01-11:
My .lan domains are not mDNS, they're statically assigned with a dnsmasq DNS service. The mosquitto service is fronted with Traefik, providing TLS termination with a self-signed certificate. The domain itself is nothing special, all lower-ASCII characters and no dashes or underscores.

It seems telling to me that this started when I pulled the most recent image, and also that rolling back to an older build resolved the problem. Perhaps a dependency of rtl_433 was updated in the recent build and introduced an unexpected change in behavior? 

It's useful to know this wasn't intended behavior, I'll keep running the older build and see if I can figure out what changed. Thanks for your help.

JS876, 2025-01-11:
Ran into this issue today as well.

The error pops up if I use the MQTT setup below, which is what my configuration has always been:
`output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/server/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/server/events,states=rtl_433/server/states`

If I update the config and swap ${host} for the actual IP address it works:

`output mqtt://1.2.3.4:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/server/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/server/events,states=rtl_433/server/states`

-----Error Message Below-----

> Starting rtl_433 with nesdrsmart.conf...
[nesdrsmart] rtl_433 version 24.10-34-gd9249e24 branch master at 202501071804 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[nesdrsmart] MQTT: Publishing MQTT data to core-mosquitto port 1883
[nesdrsmart] MQTT: Publishing device info to MQTT topic "rtl_433/####/devices[/type][/model][/subtype][/channel][/id]".
[nesdrsmart] MQTT: Publishing events info to MQTT topic "rtl_433/####/events".
[nesdrsmart] MQTT: Publishing states info to MQTT topic "rtl_433/####/states".
[nesdrsmart] MQTT: MQTT connect (core-mosquitto:1883) failed: cannot parse address
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner successfully stopped

triplepoint, 2025-01-11:
I wonder if [this change](https://github.com/merbanan/rtl_433/commit/0737fd7b02256faa7a7b7b9998653218bf08abce) was previously resulting in an early 0 return from `mg_parse_address`, whereas fixing that bug now results in some different behavior.

zuckschwerdt, 2025-01-11:
There was one other change, which seemed like a buf in the name resolution, but that might have been a mistake. 0737fd7
Can you test to roll that change back? Or maybe check if the build at 4e98b6a works and then 0737fd7 is the one that breaks?

triplepoint, 2025-01-11:
I'm not very well set up to build this here, so bare with me, I'm going to use docker containers as my build environment.

I borrowed [this dockerfile](https://raw.githubusercontent.com/hertzg/rtl_433_docker/refs/heads/master/images/debian/build-context/Dockerfile) and placed it here as `./dockerfile.rtl_433`, un-modified.

I modified my `./docker-compose.yml` file to build an rtl-433 locally:
```yaml
services:
  rtl_433:
    build:
      context: /opt/ansible-docker-compose/rtl_433
      dockerfile: dockerfile.rtl_433
      pull: true
      args:
        - debianVersion=latest
        - rtl433GitRevision=4e98b6a        
    container_name: rtl_433
    pull_policy: always
    restart: unless-stopped
    privileged: true
    command:
      - '-c/config/rtl433.conf'
    volumes:
      - /etc/rtl_433:/config
      - /etc/self-signed-certs/live/mqtt.myredacteddomain.lan/ca/crt.pem:/etc/ssl/certs/mqtt.myredacteddomain.lan.ca.crt:ro
    devices:
     - "/dev/rtlsdr_00000001"
  rtl_433_autodiscovery:
    <redacted irrelevant autodiscovery setup>
```
(note the `rtl433GitRevision` build arg set to the "before" git ref).

I cloned down the rtl_433 repo into `./rtl_433`, and checked out the same `4e98b6a` "before" commit.

Then, I could build and launch the "before" case, and see it succeed:
```shell
$ docker compose build rtl_433
...
 ✔ Service rtl_433  Built

$ docker compose up rtl_433
[+] Running 1/0
 ✔ Container rtl_433  Created                                                                                                                                                                                                                                                        0.0s 
Attaching to rtl_433
rtl_433  | rtl_433 version 24.10-28-g4e98b6a4 branch  at 202501041534 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
rtl_433  | MQTT: Publishing MQTT data to mqtt.myredacteddomain.lan port 8883 (TLS)
rtl_433  | MQTT: Publishing device info to MQTT topic "home/rtl_433/devices[/type][/model][/subtype][/channel][/id]".
rtl_433  | MQTT: Publishing events info to MQTT topic "home/rtl_433/events[/id]".
rtl_433  | MQTT: Publishing states info to MQTT topic "home/rtl_433/states".
rtl_433  | MQTT: mqtts (TLS) parameters are: tls_cert=(null) tls_key=(null) tls_ca_cert=* tls_cipher_suites=(null) tls_server_name=(null) tls_psk_identity=(null) tls_psk_key=(null) 
rtl_433  | Detached kernel driver
rtl_433  | Found Rafael Micro R820T tuner
rtl_433  | [SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
rtl_433  | Exact sample rate is: 250000.000414 Hz
rtl_433  | [R82XX] PLL not locked!
rtl_433  | Allocating 15 zero-copy buffers
```

To prove the "after" case, I modified the docker compose `rtl433GitRevision` build arg to be "0737fd7", and checked out that same git ref in the `./rtl_433` repo clone.

```shell
$ docker compose build rtl_433
...
 ✔ Service rtl_433  Built

$ docker compose up rtl_433
[+] Running 1/1
 ✔ Container rtl_433  Recreated                                                                                                                                                                                                                                                      0.3s 
Attaching to rtl_433
rtl_433  | rtl_433 version 24.10-29-g0737fd7b branch  at 202501041534 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
rtl_433  | MQTT: Publishing MQTT data to mqtt.wopr.lan port 8883 (TLS)
rtl_433  | MQTT: Publishing device info to MQTT topic "home/rtl_433/devices[/type][/model][/subtype][/channel][/id]".
rtl_433  | MQTT: Publishing events info to MQTT topic "home/rtl_433/events[/id]".
rtl_433  | MQTT: Publishing states info to MQTT topic "home/rtl_433/states".
rtl_433  | MQTT: mqtts (TLS) parameters are: tls_cert=(null) tls_key=(null) tls_ca_cert=* tls_cipher_suites=(null) tls_server_name=(null) tls_psk_identity=(null) tls_psk_key=(null) 
rtl_433  | MQTT: MQTT connect (mqtt.wopr.lan:8883) failed: cannot parse address
rtl_433 exited with code 0
```

zuckschwerdt, 2025-01-11:
Excellent! Thanks for the triage. I'll roll-back that change.

triplepoint, 2025-01-11:
Grand, thanks @zuckschwerdt.

triplepoint, 2025-01-11:
addressed in https://github.com/merbanan/rtl_433/commit/0737fd7b02256faa7a7b7b9998653218bf08abce

closed 2025-01-11 by triplepoint