merbanan/rtl_433 #3144: Saving files not working

parkerlreed, 2025-01-13:
Arch Linux
rtl_433 24.10-2
rtl-sdr-blog-git 1:v1.3.6.r0.g240bd0e-1
RTL-SDR V4

I can't seem to find the files. Nothing in my entire home folder matching *cu8

```
[parker@parker-framework rtl_433]$ rtl_433 -S known -T 30
rtl_433 version 24.10 (2024-10-30) inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000042, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:47
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:47
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:48
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:48
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:48
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-01-12 20:51:48
model     : inFactory-TH ID        : 126
Channel   : 1            Battery   : 1             Button    : 0             Temperature: 74.80 F      Humidity  : 16 %          Integrity : CRC
[sdr_callback] Time expired, exiting!
[parker@parker-framework rtl_433]$ ls
[parker@parker-framework rtl_433]$
```

zuckschwerdt, 2025-01-13:
Can't reproduce. Try running without `-T 30` and Ctrl-C when the save is done, does that work? Generally rtl_433 won't delay any action (MQTT sending, Saving samples) when the timer is up. It's a quick exit.

parkerlreed, 2025-01-13:
The timer was subsequent testing. I initially started out with just the -S all or known

Short of strace/gdb, which I can take a look at in a bit, is there any internal logging that would show if it's trying to save the file?

zuckschwerdt, 2025-01-13:
The used function is https://github.com/merbanan/rtl_433/blob/master/src/samp_grab.c#L91 `samp_grab_write()` and there should be an unconditional output `*** Saving signal to file ...` https://github.com/merbanan/rtl_433/blob/master/src/samp_grab.c#L135
The trigger is here https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L611

closed 2025-01-13 by parkerlreed