merbanan/rtl_433 #3145: Possible Segmentation fault version 24.10-35-g31ccc52a branch master at 202501112252

sblakegit, 2025-01-13:
I have a Pi4 running PiOS Lite bookworm with Homeassistant in a docker and PiHole and unifi installed on the OS.  I had a working system but realised the oil tank I was picking up was a neighbours on R43 and I needed R234 which wasn't included on the release I had.  I checked the releases for Pi bookworm and it looks like the APT version is >2years out of date.  I have followed the instructions and downloaded the latest version from github and built using cmake.  I have tried running the system in lots of different configurations but get the same result when I try to send to mqtt.  I am using Mosquitto.  I have done a full APT update and upgrade.  I have also APT uninstalled the older version of RTL-433.

When I run the standard command with the correct IP address I get the following:

rtl_433 -F json -M utc -F mqtt://192.168.0.10,retain=0,events=rtl_433
rtl_433 version 24.10-36-gf78fe658 branch master at 202501122115 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.0.10 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
{"time" : "2025-01-13 16:19:13", "model" : "Oil-SonicSmart", "id" : 140004214, "flags" : 128, "maybetemp" : 21, "temperature_C" : 13.333, "binding_countdown" : 0, "depth_cm" : 57}
Segmentation fault

I had previously setup as a service:

sudo journalctl -f -u rtl_433 -o cat
Started rtl_433.service - RTL_433 service script.
rtl_433 version 24.10-35-g31ccc52a branch master at 202501112252 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.0.10 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433/[/model][/id]".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
/etc/systemd/system/rtl_433.service:21: Standard output type syslog is obsolete, automatically updating to journal. Please update your unit file, and consider removing the setting altogether.
/etc/systemd/system/rtl_433.service:22: Standard output type syslog is obsolete, automatically updating to journal. Please update your unit file, and consider removing the setting altogether.
{"time" : "2025-01-13 14:28:57", "model" : "Oil-SonicSmart", "id" : 140004214, "flags" : 128, "maybetemp" : 21, "temperature_C" : 13.333, "binding_countdown" : 0, "depth_cm" : 56}
rtl_433.service: Main process exited, code=killed, status=11/SEGV
rtl_433.service: Failed with result 'signal'.

I am at a very quiet location from a noise perspective and get the same outcome whether I run the default decoders or specify individually R43, R234 and R234 with the minmax modifier.

I have run out of options to try.  Could there be a bug or do I need to bite the bullet and format the card and start again?
Any pointers would be greatly received.

zuckschwerdt, 2025-01-13:
Does this only happen on received messages and only when MQTT output is active?
Can you also try the binary releases from https://github.com/merbanan/rtl_433/releases
Try 24.10 and nightly, you can just try if OpenSSL11 or OpenSSL3 work. Armhf for 32-bit otherwise Arm64.

sblakegit, 2025-01-13:
@zuckschwerdt thanks for your speedy response.  Yes only on receipt of a message and only when MQTT output is requested.  Will explore binary releases shortly.

sblakegit, 2025-01-13:
@zuckschwerdt I did Wget rtl_433-rtlsdr-openssl11-Linux-arm64-24.10.zip; unzipped to ~ and run 
./rtl_433 -F json -M utc -F mqtt://192.168.0.10,retain=0,events=rtl_433

error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory.
I have tried to apt install to no avail.  Would you point me to the correct relevant library?
Sorry for being a pain.

sblakegit, 2025-01-13:
@zuckschwerdt 

wget https://github.com/merbanan/rtl_433/releases/download/24.10/rtl_433-rtlsdr-openssl3-Linux-arm64-24.10.zip

./rtl_433 -F json -M utc -F mqtt://192.168.0.10,retain=0,events=rtl_433
rtl_433 version 24.10 branch  at 202410301452 inputs file rtl_tcp RTL-SDR with TLS
MQTT: Publishing MQTT data to 192.168.0.10 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
{"time" : "2025-01-13 21:18:30", "model" : "Oil-SonicSmart", "id" : 140004214, "flags" : 128, "maybetemp" : 21, "temperature_C" : 13.333, "binding_countdown" : 0, "depth_cm" : 56}

Seemed to work without error.  Its 30-minutes between packets so it may be an hour before I know for sure.

zuckschwerdt, 2025-01-13:
No rush, try the nightly version next day or so. Would be easier to find if something broke between those versions.

sblakegit, 2025-01-14:
./rtl_433 -F json -M utc -F log -F mqtt://192.168.0.10,retain=0,events=rtl_433[/model][/id]
rtl_433 version -128-NOTFOUND branch  at 202501112252 inputs file rtl_tcp RTL-SDR with TLS
MQTT: Publishing MQTT data to 192.168.0.10 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433[/model][/id]".
Found Rafael Micro R820T tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
{"time" : "2025-01-14 09:06:21", "model" : "Ford", "type" : "TPMS", "id" : "66fa48c6", "pressure_PSI" : 34.500, "temperature_C" : 24.000, "moving" : 1, "learn" : 0, "code" : "8a5046", "unknown" : "00", "unknown_3" : "2", "mic" : "CHECKSUM"}
{"time" : "2025-01-14 09:07:04", "model" : "Oil-SonicSmart", "id" : 140004214, "flags" : 128, "maybetemp" : 23, "temperature_C" : 10.000, "binding_countdown" : 0, "depth_cm" : 58}

Seems to work fine.  I seem to be struggling to pickup the sonic advanced transmission but its not crashing.  Does that suggest a problem my end?

zuckschwerdt, 2025-01-14:
There is no "unsafe" way to compile rtl_433. It's strange why your build would break.
Did you do a regular build (`cmake ..` or similar)? Then you got debug symbols, try to run in `gdb` an upon crash issue `bt` to print a trace.

sblakegit, 2025-01-14:
I copied the cmake commands straight from this site.  gdb and bt are a bit alien.  Can you help with a bit more of a command line?

zuckschwerdt, 2025-01-14:
Install gdb, then
```
gdb rtl_433
(now in gdb)
run -F json -M utc -F log -F "mqtt://192.168.0.10,retain=0,events=rtl_433[/model][/id]"
(wait for crash)
backtrace
```

sblakegit, 2025-01-14:
/usr/local/bin $ gdb ./rtl_433
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./rtl_433...
(No debugging symbols found in ./rtl_433)
(gdb) run -F json -M utc -F log -F "mqtt://192.168.0.10,retain=0,events=rtl_433[/model][/id]"
Starting program: /usr/local/bin/rtl_433 -F json -M utc -F log -F "mqtt://192.168.0.10,retain=0,events=rtl_433[/model][/id]"
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
rtl_433 version 24.10-36-gf78fe658 branch master at 202501122115 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.0.10 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433[/model][/id]".
[New Thread 0x7ff588a140 (LWP 4893)]
[New Thread 0x7ff588a140 (LWP 4894)]
[Thread 0x7ff588a140 (LWP 4893) exited]
[Thread 0x7ff588a140 (LWP 4894) exited]
[New Thread 0x7ff588a140 (LWP 4895)]
Found Rafael Micro R820T tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[New Thread 0x7ff507a140 (LWP 4896)]
{"time" : "2025-01-14 13:50:29", "model" : "Oil-SonicSmart", "id" : 140004214, "flags" : 128, "maybetemp" : 22, "temperature_C" : 11.667, "binding_countdown" : 0, "depth_cm" : 57}

Thread 1 "rtl_433" received signal SIGSEGV, Segmentation fault.
0x000000555556b7a4 in sdr_callback ()
(gdb) backtrace
#0  0x000000555556b7a4 in sdr_callback ()
#1  0x000000555556fd9c in sdr_handler ()
#2  0x0000005555577094 in mg_mgr_handle_ctl_sock ()
#3  0x0000005555585064 in mg_socket_if_poll ()
#4  0x000000555557cccc in mg_mgr_poll ()
#5  0x000000555556a3e4 in main ()

zuckschwerdt, 2025-01-14:
Thanks. The [`sdr_callback()`](https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L390) is processing the radio frames and will then trigger the outputs. It matches what you are seeing. It's strange that not some output call crashes but the main callback itself. It must be towards the end of the function as the output seems to work.
I'll have a closer look but it's unclear what the problematic piece of code could be and why this happens with your build only.

Murphy-Cameron, 2025-01-16:
I'm experiencing the same issue, and have traced it to changes in commit #3134 for MQTT throttling.  I'm kind of out of my element here, but I'm getting the same segmentation fault in sdr_callback() on line 415 of rtl_433.c.   I poked around with some printf statements on the changes in output_mqtt.c, and it would appear the delay timer isn't working as intended.  $ctx->reconnect_delay would start at 0, then increment to 1 on the next call, then start back over at 0.  (and continued hammering the mqtt server.  It would fail to connect, and retry ~100 times in a few seconds, then eventually seg fault.)  I can't reproduce it now, but I was also getting some backtraces that showed heap-free-after-use errors where it was trying to reference a connection that had previously been freed somewhere in mongoose.c.

I don't understand enough about how the code is structured, but it would appear the delayed connection is getting killed off by some sort of garbage collection or monitor, and eventually sdr_callback() is trying to send a message on a connection that isn't there any more.

zuckschwerdt, 2025-01-16:
Thanks! That narrows it down far enough, I'l dig around!

zuckschwerdt, 2025-01-16:
Turns out the timer socket also got a close event on shutdown which then falsely triggered a reconnect attempt.

commit d05debc6 (2025-01-16): Fix MQTT reconnect timer (closes #3145)
closed 2025-01-16 by zuckschwerdt