merbanan/rtl_433 #3149: SCE-R1 gunmetal

hmollercl, 2025-01-18:
How did you make this work? @filzek
> we have found all changes to use with SOMFY RTS TUBE
> 
> 
> 
> frame[0] = 0x95;         // is the COMMAND FRAME
> 
> frame[1] = (unsigned char)(15)<< 4;   //is fixed always
> 
> 
> 
> so, the CMD as off:
> 
> #define TUBE_SCE_R1_CMD_PROG 0x9C
> 
> #define TUBE_SCE_R1_CMD_STOP 0x95
> 
> #define TUBE_SCE_R1_CMD_UP 0x96
> 
> #define TUBE_SCE_R1_CMD_DOWN 0x98
> 
> #define TUBE_SCE_R1_CMD_UP_DOWN 0x9A
> 
> 
> 
> Repeate the FRAMES to use special Command: 
> 
> to make MY working just sent the STOP 10 times
> 
> to make the PROG to work just send it 25 times!
> 
> 
> 
> so all is done!!! 

 _Originally posted by @filzek in [#2261](https://github.com/merbanan/rtl_433/issues/2261#issuecomment-1413185617)_

filzek, 2025-01-18:
> How did you make this work? @filzek
>
>  _Originally posted by @filzek in [#2261](https://github.com/merbanan/rtl_433/issues/2261#issuecomment-1413185617)_

We have decoded the frames and signal, learn from it, and start to design the operation, so, we are able to rebuild the constructor, and so, made the code to work.

hmollercl, 2025-01-21:
thanks for you answer @filzek, could you tell me how are you using it? I don't know where to put he code you came up, I searched in the code for the frame[] variable but didn't found it.
Are you using ESP, arduino or other?
Which libraries are you using? 
The libraries I found only work with 433.42 Mhz (and CC1101), but the SCE-R1 says it work at 433.92 Mhz, how did you solve this?
Are you using  CC1101 transceiver?
I have 3 blinds that use this remote I want to automate but I haven't found more info about it, thanks a lot.

joseluisdiaz, 2025-03-16:
@hmollercl I think what @filzek is saying is that the way that you build the frame is a bit different. If you look the code for the https://github.com/Legion2/Somfy_Remote_Lib/blob/main/src/SomfyRemote.cpp#L40-L41 SomfyRemote the two first bytes are different.

gdt, 2025-07-01:
I can't tell what this issue is about.  Is it a request for a decoder, and if so that's the status and next steps?  Is it just a discussion and it's done and I should close it?

hmollercl, 2025-07-07:
sorry, haven't have the time to try what @joseluisdiaz said. I'm not sure if this work.

filzek, 2025-07-07:
@hmollercl , the code I provided is correct.

@joseluisdiaz has sent the correct info how to use it.

@gdt Greg, yes, the code shows how it works, so, need to add the convestion to tell that this use reflects the RFS Somfy Tube model of protocol, so, in the Rtl433 framework should disclouse it as SomfyRTS Tube, as its a little bit different from the original RTS protocol.

The code I have provided also show how to create the protocol for read and understand and also how to write and use with another TX framework such as the rc-switch.

Sorry for the late reply.

gdt, 2025-07-07:
Between you, could you prepare and submit a pull request that adds a decoder, or comments to an existing decoder? Really, I mean to modify the source code repo so that is has all the useful information, so that we can close the issue.