merbanan/rtl_433 #3150: Landis+Gyr GridStream support?

5ft24dave, 2025-01-18:
The latest version of the RTL_433 binary has support for the Landis+Gyr GridStream protocols  Is it possible to update the core RTL_433 to include those newer protocols?  
Thank you

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-01-18 12:35:12
model     : LandisGyr-GS Network ID: 142a          Location  : Washington    Provider  : Puget Sound Energy                      subtype   : 213           protoversion: 5           Integrity : CRC
Source Meter ID: 90f6ffd3
Source Meter WAN ID: fe90f6ffd300                  Target Meter ID: 90ef11c5 Timestamp : 1737232512    Uptime    : 49012726
[gridstream_decode] Decoded frame data
codes     : {600}2ad500475190ef11c590f6ffd372678c10800001388a02ebdff6a40b0101fe90f6ffd30090f6ffd30001032500000000000000000000000399e0000ca60300c720302d848006e9121006cc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-01-18 12:41:38
model     : LandisGyr-GS Network ID: 142a          Location  : Washington    Provider  : Puget Sound Energy                      subtype   : 213           protoversion: 5           Integrity : CRC
Source Meter ID: 90d69dc1
Target Meter ID: 90d577e0
[gridstream_decode] Decoded frame data
codes     : {168}2ad500112190d577e090d69dc196b8024612002c73
`

zuckschwerdt, 2025-01-18:
Which software do you need to be updated? Are you referring to the ESP32 port?

5ft24dave, 2025-01-19:
The rtl_433 app itself  the newest build, release 24.10 supports the GridStream  the data was captured above with the latest build from https://github.com/merbanan/rtl_433
currently in the HA addon we have rtl_433 version 23.11

zuckschwerdt, 2025-01-19:
Ok. Try to find out which HA addon with version 23.11 you have (maybe https://github.com/pbkhrv/rtl_433-hass-addons ?) and ask them for a release.

closed 2025-01-19 by zuckschwerdt