merbanan/rtl_433 #3151: New Device - WEN 340 Air Filter

gcormier, 2025-01-20:
Captures - https://github.com/gcormier/rtl_433_tests/tree/wen_dust/tests/wen_dust

3 button remote at 433MHz. Flex decoder for this one.

I'm close. I just don't know why I can't ignore the preamble

```
$ rtl_433 off.cu8 -X 'n=WEN,m=OOK_PWM,s=316,l=944,r=9156,g=968,t=253,y=0'

[Input] Test mode active. Reading samples from file: off.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.081140s
model     : WEN          count     : 1             num_rows  : 1             rows      : 
len       : 25           data      : ffffff8
codes     : {25}ffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.081140s
model     : WEN          count     : 1             num_rows  : 1             rows      : 
len       : 16           data      : 3c6e
codes     : {16}3c6e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.081140s
model     : WEN          count     : 1             num_rows  : 1             rows      : 
len       : 16           data      : 3c6e
codes     : {16}3c6e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.081140s
model     : WEN          count     : 1             num_rows  : 1             rows      : 
len       : 16           data      : 3c6e
codes     : {16}3c6e
```

Trying to add preamble

```
$ rtl_433 off.cu8 -X 'n=WEN,m=OOK_PWM,s=316,l=944,r=9156,g=968,t=253,y=0,preamble=ffffff8'
[Input] Test mode active. Reading samples from file: off.cu8

$ 
```

gcormier, 2025-01-20:
[image]

gcormier, 2025-01-20:
Ah, I see. That first set of sync pulses is not the same packet. This works

`-X 'n=WEN3410,m=OOK_PWM,s=316,l=944,r=9156,g=968,t=253,y=0,preamble=3c,get=@0:{8}:CMD:[110:OFF 61:ON ce:SPEED]'`

Should I find a way to ignore those sync pulses in this decoder, or ignore them?

zuckschwerdt, 2025-01-20:
If there is no sync word or checksum then checking a preamble/warmup might be the only way to improve robustness.
You should be able to pull all packets into a single bitbuffer using `-X 'n=name,m=OOK_PWM,s=312,l=944,r=11000,g=1100' /Users/zany/Desktop/off.cu8` (1100 is safely longer than the usual 950 µs gap and 11000 is longer than the packet gap of 9950 µs). Use `,match={25}ffffff8` to require the full preamble (`ffffff8` is not good as it is equal to `match={28}ffffff8`)

gcormier, 2025-01-20:
Hmm ok, that works, but I'm not sure how to complete it now.

`rtl_433 off.cu8 -X 'n=WEN3410,m=OOK_PWM,s=316,l=944,r=11000,g=1100,match={25}ffffff8,preamble=3c'` returns

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.081140s
model     : WEN3410      count     : 3             num_rows  : 4             rows      : 
len       : 25           data      : ffffff8, 
len       : 8            data      : 6e, 
len       : 8            data      : 6e, 
len       : 8            data      : 6e
codes     : {25}ffffff8, {8}6e, {8}6e, {8}6e
```

But `rtl_433 off.cu8 -X 'n=WEN3410,m=OOK_PWM,s=316,l=944,r=11000,g=1100,match={25}ffffff8,preamble=3c,get=@26:{8}:CMD:[110:OFF 97:ON 200:SPEED]'` does not seem to work (out is same as above).

I tried to look for an example with multiple rows of data in https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf but didn't have any luck. 

Also I wonder if `repeats` would be useful? But needs to ignore the 1st preamble.

gcormier, 2025-01-20:
Bingo, one step further. I guess the get starts after the match and preamble, so @0 instead of @26 

`rtl_433 off.cu8 -X 'n=WEN3410,m=OOK_PWM,s=316,l=944,r=11000,g=1100,match={25}ffffff8,preamble=3c,get=@0:{8}:CMD:[110:OFF 97:ON 200:SPEED],unique`

Adding `repeats>=3` will break it though.

gdt, 2025-07-01:
Is there anything in this issue that isn't in the PR?  If we can close the issue without losing useful information, please do.
and maybe this will poke a reviewer to look at the PR!

closed 2025-07-01 by gcormier