merbanan/rtl_433 #3165: Audi A6 2020 TPMS supported?

toto99303, 2025-01-27:
Hello people,

I am trying to read my Audi A6 2020 TPMS 433Mhz sensors, but it seems like they are not supported.
When I collect verbose I can see this:

`[tpms_bmwg3_decode] Preamble not found`

Not sure if they are really bmw gen3 or 4 or 5...

They are produced by Huf and here's the datasheet:

https://fccid.io/ANATEL/02294-15-03616/6718_MANUAL/5C6D018E-58B4-4EF0-B74A-9A072F589A12/PDF

Can anyone tell me by looking at the FCC pdf if this sensor is in the supported list?
I also collected some *.cu8 files, but I have no clue how to proceed. (there might be some other stuff from my neighbours in these, so ignore the unnecessary if possible...)

Here:

https://drive.google.com/file/d/1AdDk6rIpYU7yiYWXAHQApLnaBh2ppYzN/view?usp=sharing

Thanks in advance!

toto99303, 2025-01-28:
Thanks for your reply, I did this on a raspberry pi, I think there might be an issue when using the RTL-SDR on an ARM based computer.
I will compile again and will try to re-capture the TPMS using normal PC.

zuckschwerdt, 2025-01-28:
RTL-SDR and rtl_433 is regularly used on Pi's. Just make sure to put distance between the Pi and the SDR, use a USB cable.

toto99303, 2025-01-30:
> RTL-SDR and rtl_433 is regularly used on Pi's. Just make sure to put distance between the Pi and the SDR, use a USB cable.

Thanks for the advice! Why is that? We are sniffing at 433Mhz, and the pi only does 2.4/5Ghz?

zuckschwerdt, 2025-01-30:
Electromagnetic Interference will impact the SDR if too close to a Pi as neither the Pi or SDR are shielded.

toto99303, 2025-02-01:
Hello, I used USB extender cable and grabbed some fresh captures. Please have a look if there's any valuable information now:

[https://drive.google.com/file/d/1Xi7c_KJyOz8_buch3_UhVIH9ocLiRFkU/view?usp=sharing](url)

zuckschwerdt, 2025-02-01:
Please preview and sort your samples. https://triq.org/pdv/
The samples are all empty (except g057 and that does not look like TPMS, more like a doorbell-style sender).

toto99303, 2025-02-01:
I tried revieweing them, but I don't know how a normal TPMS signal should look like. Can someone attach a sample so I can see what to look for? I'm also not sure If the TPMS sensor was really awake. I tried waking it by dropping the pressure to 1 bar. Next time I might try while the car is moving.
I also have some spare sensors around, but I can't wake them up. I tried shaking them, but seems like they really need to be inside and under pressure to be awake... Any recommendations are appreciated :)

zuckschwerdt, 2025-02-01:
You can see some TPMS signals in the tests repo: https://github.com/merbanan/rtl_433_tests/tree/master/tests

ProfBoc75, 2025-02-01:
@toto99303 

Some think like that:
On the top, the picture with the colored signal,
On the bottom, an attempt of decoding with pulse information.

Exemples:
OOK signal: one line (one central frequency)
[image]

FSK signal: 2 lines (2 frequencies)
[image]

If you have only noise into the picture and nothing decoded, the signal is not good, try another one.

Your file, is empty, only noise

[image]