merbanan/rtl_433 #3167: broadcast flag in output_udp.c

AkaBoing, 2025-01-28:
hi Christian and ofc all others,
Christian, you know, im using rtl_433 in syslog **broadcast** mode under Debian
therefore i changed and have to change it each time new code is incoming by GIT

"output_udp.c"

    120  //int broadcast = 1;
    121  //int ret = setsockopt(client->sock, SOL_SOCKET, SO_BROADCAST, &broadcast, sizeof(broadcast));
to 
   120  int broadcast = 1;
   121  int ret = setsockopt(client->sock, SOL_SOCKET, SO_BROADCAST, &broadcast, sizeof(broadcast));

with this small change it works perfect !

a minor fix which works for 2 years now. - NO side effect.
pls think about it..

Boing

zuckschwerdt, 2025-01-29:
Yes, `SO_BROADCAST` is just the permission request to use broadcast addresses. There might have been problems with this at some point, I'm not sure anymore.

Does anybody know if `SO_BROADCAST` is safe to always request for a socket?

AkaBoing, 2025-01-29:
thank you for taking on the matter.

there is one known side effect.
if one member in the local net has (by mistake or whatever) set up an alias for a IP address,
this member gets the UDP broadcast 2 times, but who cares ?

gdt, 2025-01-29:
The first question is standardization/portability.

I checked NetBSD sources, and the definition of `SO_BROADCAST` in `/usr/include/sys/socket.h` dates from version 1.1, 21 March 1993, which would have been the import of 386bsd-0.1, and matching 4.4BSD Lite and Lite-2.  `setsockopt(2)` says

  - The option SO_BROADCAST requests permission to send broadcast datagrams on the socket.  Broadcast was a privileged operation in earlier versions of the system.

I would say that the code should be written to only use `SO_BROADCAST` after testing with `ifdef`. And, to only set it if the address is broadcast.   It's hard to be sure that turning it on is 100% safe for normal usage, and I don't think we should enable it for the 99.9999% of people that aren't sending broadcast.   (Maybe we  don't have that many users and I should say N-1.)

I see using broadcast as irregular.   It has been good practice since the early 90s at least to be multicast.    Still, if someone configures broadcast, I don't think rtl_433 should stop them.

I'm happy to review a PR with adequate guards:
  - it will continue to build on any system meeting POSIX
  - `SO_BROADCAST` is only enabled for broadcast addresses

gdt, 2025-01-29:
Also, grepping NetBSD kernel, it looks like one can set this, but I don't see where it ever is checked, which matches the man page.

@AkaBoing Can you read the sources for the kernel and see what's going on and why?   It would be good to understand if this is a choice by Linu to require it to be set, vs matching the BSD behavior (which I think predates Linux).   And then to have a comment explaining this, in the changed code.

AkaBoing, 2025-01-29:
sure , i viewed the  sources for this flag, read more on kernel 6.1+ docs and can see one problem: on SUN/Solarix  this is not valid
without a second parameter: the lxnet flag.
Im not firm with Solarix, so it might be a real issue..

im using  -F syslog:192.168.2.**255**:1234
so **255** is a clear broadcast to all member in the 192.168.2/24 subnet

a "ifdef" for this "**255**"  looks good to to me.

AkaBoing, 2025-01-29:
Christian, you are right, as always.
im the one of 1234 users who needs this flag..
the rest uses tcp with tons of overhead. lol
simply say "YES" or "NO" and i can live with it..
no drama.
in newest kernel 6.1.10+  the limit is btw: 2048 bytes over udp broadcast.
i can send a Dolby Digital 5.1 stream with 1536 bytes in one message over UDP broadcast, i tested it a week ago.
Even a DD+chunk with 1866 bytes works

gdt, 2025-01-29:
Maybe we should only set it if `#ifdef linux` then.   It seems better than assuming that only sunos varies from the adopted BSD pattern.

The in-use broadcast address can vary.   Back in the old days it was all zeros, not all ones.   And indeed it has to be interpreted relative to netmask.   So it would require too much code to check.

So I think this can be set regardless of broadcast.   If it's limited to linux, then that seems safer as code reading seems to indicate it won't do anything other than avoid those two checks.

And @AkaBoing you should try out multicast.  That's really the right thing, as only machines intending to listen will get packets arriving at their stacks.   Basically nobody should be using broadcast for anything.

zuckschwerdt, 2025-01-29:
Agreed. And broadcast is of very limited use today. You likely want something at layer 3 to use routing (docker, vpn, …) like multicast or some pub/sub.

I'll `#ifdef __linux__` enable the `SO_BROADCAST` flag.

gdt, 2025-01-29:
> I'll `#ifdef __linux__` enable the `SO_BROADCAST` flag.

Sounds good.

IP broadcast and multicast is layer 3!   My point is that broadcast overdelivers, and that was one of the main reasons why we have multicast.  The other is that it's limited to single L2 object (switched Ethernet, these days).    But your point, that using broadcast should be replaced by something like MQTT, I completely agree with.  I use syslog udp to 127.0.0.1 only, to get to a python script that sends  mqtt.

PR #3171 (2025-01-29): Add broadcast flag for syslog output
Enables the SO_BROADCAST for tested platforms, so far Linux and MacOS.

Closes #3167
closed 2025-01-29 by zuckschwerdt