merbanan/rtl_433 #3179: Add support for Ecowitt WH43 AQI sensor

andrewjmcginnis, 2025-02-02:
[WH43samples.zip](https://github.com/user-attachments/files/18633111/WH43samples.zip)

I was pretty sure the only difference between the WH41 and WH43 is the 41 has a PV panel on top, but while some of the existing decode of the WH41 lines up with what I get for the 43, there's chunks that I don't know what they are. 

The 43 I received from ecowitt does not have a honeywell sensor in it as they list on the product page.  It is a Sensirion SPS30 in it.  I purchased a WH46 that arrived defective, so in taking it apart to diagnose, I also disassembled the 43 to see what the 46 might be doing wrong. That's where I discovered the same SPS30 sensor in both units. I wonder if the 41 still uses a honeywell sensor, of if they've moved to it to the SPS30 like this 43, and the 45/46.

I have a bunch of samples to check against attached.

triq.org/pdv using FSK_PCM, 56/56, 10000 gap, gets to something like this:
`Bits: {336} 55 55 55 55 55 51 6E A2 18 06 85 1A 01 44 00 F0 F6 D1 00 F0 / 55 55 55 55 55 51 6E A2 18 06 85 1A 01 44 00 F0 F6 D1 00 E0 /`

I have a flex decoder of `-X "n=Fineoffset-WH43,m=FSK_PCM,s=56,l=56,t=0,r=10000,preamble=aa2dd4"` that gets me a data output of 

```
time      : 2025-02-02 11:56:18
model     : Fineoffset-WH43                        count     : 1             num_rows  : 1             rows      : 
len       : 92           data      : 4300d0a3400a800a6cf6205
codes     : {92}4300d0a3400a800a6cf6205
Modulation: FSK          Freq1     : 433.9 MHz     Freq2     : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 32.0 dB       Noise     : -32.1 dB
```

43 is the model, 00d0a3 is the unit ID of my sensor.

I'm trying to adapt the layout of the fineoffset_wh45.c decoder-

```
/**
Fine Offset Electronics WH45 air quality sensor,

- also Ecowitt WH45, Ecowitt WH0295
- also Froggit DP250
- also Ambient Weather AQIN

Preamble is aaaa aaaa, sync word is 2dd4.

Packet layout:

     0  1  2  3  4  5  6  7  8  9 10 11 12 13 14
    YY II II II 0T TT HH Bp pp BP PP CC CC XX AA
    45 00 36 60 02 7e 36 40 23 00 29 02 29 07 4f

- Y: 8 bit fixed sensor type 0x45
- I: 24 bit device ID
- T: 11 bit temperature, offset 40, scale 10
- H: 8 bit humidity
- B: 1 bit MSB of battery bars out of 5 (a value of 6 indicates external power via USB)
- p: 14 bit PM2.5 reading in ug/m3 * 10
- B: 2 bits LSBs of battery bars out of 5
- P: 14 bit PM10 reading in ug/m3 * 10
- C: 16 bit CO2 reading in ppm
- X: 8 bit CRC
- A: 8 bit checksum
```
I've gotten to my decode line to
`TYPE:8h ID:24h 1b Bat_MSB:1d PMTWO:14d Bat_LSB:2d PMTEN:14d CRC:8h CHECKSUM: 8h BITSIM:16h`

Running my first sample g001_433.92M_250k.cu8, I got:
```
43 00 d0 a3 40 64 80 5a 6e a2 20 58
TYPE:43 ID:00d0a3 0 Bat_MSB:1 PMTWO:00100 Bat_LSB:2 PMTEN:00090 CRC:6e CHECKSUM: a2 BITSIM:2058
```
Really only the first two are correct (type and ID).  It works out to a PM2.5 of 100, and PM10 of 90.  Without a monitor display, I can't really tell if that's correct.  The WH46 was meant to help decipher since they should both have the same or very similar readings for PM2.5.  Knowing now that they have the same sensor in use between them, I imagine the output should be substantially the same, just without the temp, humidity, and CO2 from the SCD30 sensor the WH46 has, nor the PM1 and PM4 that are wired up on the 46 that are *not* wired up on the 43.

I haven't cracked the battery bits yet.  I tried it plugged in to USB power with no batteries; adding batteries > while still on USB; and removing USB power to run only batteries.  I can get the bat_MSB to read '1' or '4' depending on the decode, but not '6' (what is supposed to be 'on USB/DC power').  Messing with it changes the apparent pm2.5 and further datapoints...

Any help would be very appreciated.

ChrisM510, 2025-02-03:
+1 from me on this.  I just bought a WH41 and a WH46D for monitoring indoor and outdoor AQI.  Help greatly appreciated!

ChrisM510, 2025-02-03:
I don't see anything, but I am backing up my WeeWX station right now before I update to the latest RTL_433 (it occurred to me after my post that I hadn't updated in a few months).  I will report back after my update.  :-)

andrewjmcginnis, 2025-02-08:
I think I was going down the wrong path in assuming this was more like a WH45/46, than a WH41.  While the sensor in use is the same as the 45/46, the packet is laid out like the WH41, with a few extra bytes for the unit ID.

So where the WH41 packet looks like this
```
Data layout:
             41 c7 41 ae 01 c2 f9 b3 00000, Ecowitt 41
    aa 2d d4 42 cc 41 9a 41 ae c1 99 9
             FF DD ?P PP ?A AA CC BB

- F: 8 bit Family Code?
- D: 8 bit device id (corresponds to sticker on device in hex)
- ?: 1 bit?
- b: 1 bit MSB of battery bars out of 5
- P: 14 bit PM2.5 reading in ug/m3
- b: 2 bits LSBs of battery bars out of 5
- A: 14 bit PM10.0 reading in ug/m3
- C: 8 bit CRC checksum of the previous 6 bytes
- B: 8 bit Bitsum (sum without carry, XOR) of the previous 7 bytes
```

The WH43 adds two bytes to the ID length (for a total of 24bits). Changing the byte and bit positions to account for this, the WH41 parsing applies to the 43. Rather than 8 bytes extracted and parsed, there are 10 bytes.

FAM:8d ID: 24h 1b Bat_MSB:1d PMTWO:14d Bat_LSB:2d PMTEN:14d CRC:8h BITSIM:8h

// Decode data for WH43
uint8_t family    = b[0];  
uint8_t id        = b[1];  
uint8_t unknown1  = (b[4] & 0x80) ? 1 : 0;  
int pm25          = (b[4] & 0x3f) << 8 | b[5];  
int pm100         = (b[6] & 0x3f) << 8 | b[7];  
int battery_bars  = (b[4] & 0x40) >> 4 | (b[6] & 0xC0) >> 6; // out of 5  
float battery_ok  = battery_bars * 0.2f; // Convert out of 5 bars to 0 (0 bars) to 1 (5 bars)

ChrisM510, 2025-02-08:
I have the 41 and 46D working now in my WeeWX server.  I needed to update my rtl_433 and my weewx-sdr plugin (thanks to Andy Little for adding support for the 46D to weewx-sdr).  The only thing not really working on the 46D is the 4um PM readings, but that's really not a big deal.  You can see my station by going to http://weather.clmnetwork.com:8080/ and scrolling to the bottom 2 graphs.

gdt, 2025-02-08:
@andrewjmcginnis That's great progress!    Sounds like you should  extend the WH41 decoding to work with this too, and submit a PR.  Is that how you see it?

@ChrisM510 Are you using the just-posted code, or something else?

ChrisM510, 2025-02-08:
I merely updated my rtl_433 to the latest release posted, and then Andy Little added support for the 46D to weewx-sdr and I used his version of sdr.py, and it all just started working. No modifications were necessary to the current release of rtl_433.

andrewjmcginnis, 2025-02-08:
> [@andrewjmcginnis](https://github.com/andrewjmcginnis) That's great progress! Sounds like you should extend the WH41 decoding to work with this too, and submit a PR. Is that how you see it?

I don't want to commit to it just yet, in case I'm wrong.  I'm still waiting on my WH46 to arrive. Since that model is already supported, I can cross check the pm readings to make sure those are correct. The crc and checksum works at b[9] and b[10], but the packet is much longer (+/- 153 bits) than the WH41 (+/- 129 bits)- despite having the same listed outputs.  If there's additional readings in this, I want to make sure their captured.

But (and I don't know how you'd want to concisely perform the if/then/else), here's a crack at it:
```
static int fineoffset_WH0290_callback(r_device *decoder, bitbuffer_t *bitbuffer)
{
    data_t *data;
    uint8_t const preamble[] = {0xAA, 0x2D, 0xD4};
    uint8_t b[8];
    unsigned bit_offset;

    bit_offset = bitbuffer_search(bitbuffer, 0, 0, preamble, sizeof(preamble) * 8) + sizeof(preamble) * 8;
    if (bit_offset + sizeof(b) * 8 > bitbuffer->bits_per_row[0]) {  // Did not find a big enough package
        decoder_logf_bitbuffer(decoder, 1, __func__, bitbuffer, "short package. Row length: %u. Header index: %u", bitbuffer->bits_per_row[0], bit_offset);
        return DECODE_ABORT_LENGTH;
    }
    bitbuffer_extract_bytes(bitbuffer, 0, bit_offset, b, sizeof(b) * 8);

// Verify checksum, same as other FO Stations: Reverse 1Wire CRC (poly 0x131)
uint8_t crc = crc8(b, 6, 0x31, 0x00); // WH41: Compute CRC over first 6 bytes
uint8_t checksum = 0;
for (unsigned n = 0; n < 7; ++n) {
    checksum += b[n]; // WH41: Sum first 7 bytes
}

// Check WH41 CRC/Checksum first
if (crc == b[6] && checksum == b[7]) {
    // WH41 packet detected
    uint8_t family    = b[0];
    uint8_t id        = b[1];
    uint8_t unknown1  = (b[2] & 0x80) ? 1 : 0;
    int pm25          = (b[2] & 0x3f) << 8 | b[3];
    int pm100         = (b[4] & 0x3f) << 8 | b[5];
    int battery_bars  = (b[2] & 0x40) >> 4 | (b[4] & 0xC0) >> 6;
    float battery_ok  = battery_bars * 0.2f;
    // Proceed with WH41 decoding...
    decoder_logf(decoder, 2, __func__, "Decoded as WH41");
    return DECODE_SUCCESS;
}

// Retry using WH43's longer packet structure
crc = crc8(b, 8, 0x31, 0x00); // WH43: Compute CRC over first 8 bytes
checksum = 0;
for (unsigned n = 0; n < 9; ++n) {
    checksum += b[n]; // WH43: Sum first 9 bytes
}

// Check WH43 CRC/Checksum
if (crc == b[8] && checksum == b[9]) {
    // WH43 packet detected
    uint8_t family     = b[0];  
    uint32_t id        = (b[1] << 16) | (b[2] << 8) | b[3];  // 24-bit ID
    uint8_t unknown1   = (b[4] & 0x80) ? 1 : 0;
    int pm25           = (b[4] & 0x3f) << 8 | b[5];
    int pm100          = (b[6] & 0x3f) << 8 | b[7];
    int battery_bars   = (b[4] & 0x40) >> 4 | (b[6] & 0xC0) >> 6;
    float battery_ok   = battery_bars * 0.2f;
    // Proceed with WH43 decoding...
    decoder_logf(decoder, 2, __func__, "Decoded as WH43");
    return DECODE_SUCCESS;
}

// If neither CRC check passes, return failure
decoder_logf(decoder, 1, __func__, "Checksum error: %02x %02x", crc, checksum);
return DECODE_FAIL_MIC;

if (b[0] == 0x41)
    type = MODEL_WH41;
else
    type = MODEL_WH43;

    /* clang-format off */
    data = data_make(
            "model",            "",             DATA_STRING, type == MODEL_41 ? "Fineoffset-WH0290" :"Fienoffset-WH43",
            "id",               "ID",           DATA_INT,    id,
            "battery_ok",       "Battery Level",  DATA_FORMAT, "%.1f", DATA_DOUBLE, battery_ok,
            "pm2_5_ug_m3",      "2.5um Fine Particulate Matter",  DATA_FORMAT, "%d ug/m3", DATA_INT, pm25/10,
            "estimated_pm10_0_ug_m3",     "Estimate of 10um Coarse Particulate Matter",  DATA_FORMAT, "%d ug/m3", DATA_INT, pm100/10,
            "family",           "FAMILY",       DATA_INT,    family,
            "unknown1",         "UNKNOWN1",     DATA_INT,    unknown1,
            "mic",              "Integrity",    DATA_STRING, "CRC",
            NULL);
    /* clang-format on */

    decoder_output_data(decoder, data);
    return 1;
}
```

andrewjmcginnis, 2025-02-11:
@gdt 
That last comment's code was... uncompilable.  This is working. The code would begin in `fineoffset.c` at line 382 as it's currently constituted in the repo. 
-I changed some of the existing output for `data_make()`, to include the `ext_power` source Yes/No, since that is used in the WH45 and WH46 decoders, and is equally present in the WH41/43 data. 
-I changed battery_ok to use the 45/46 format of `MIN(battery_bars * 0.2f, 1.0f)` because it was otherwise allowing for a `Battery Level` reading of 1.2 when plugged into USB power.
-`Family` is present in the current WH0290 decoder `data_make()`, so I left it here as well. The WH45/46 decoders do not show it, so it may be removed/commented out.
-I changed the "um" to "µm" because the micro character is easy enough to add as the proper prefix.
-I left "Estimate of 10µm Coarse Particulate Matter" as-described, because I do not know that the WH41 has changed it's sensor from the Honeywell to the Sensirion SPS30. The WH45/46 decoders don't call the PM10 of the SPS30 an estimation in the way that the Honeywell is here, with the comment

> The sensor specification statement states that PM10 values are estimated
> from distribution profiles of PM0.5, PM1.0, and PM2.5 measurements, but
> the datasheet does a specify a degree of accuracy for the values unlike
> the Honeywell sensor.

Working code for WH41/WH43 AQI devices (with differing bit length payloads):
```
#define MODEL_WH41 41
#define MODEL_WH43 43
static int fineoffset_WH0290_callback(r_device *decoder, bitbuffer_t *bitbuffer)
{
    data_t *data;
    uint8_t const preamble[] = {0xAA, 0x2D, 0xD4};
    uint8_t b[10];
    unsigned bit_offset;
    int type;

    /* Declare these variables in the outer scope so they are available everywhere in the function */
    uint8_t family = 0;
    uint32_t id = 0;       // Use uint32_t for a 24-bit id.
    int pm25 = 0;
    int pm100 = 0;
    int battery_bars = 0;
    float battery_ok = 0.0f;
    int ext_power = 0;

    bit_offset = bitbuffer_search(bitbuffer, 0, 0, preamble, sizeof(preamble) * 8) + sizeof(preamble) * 8;
    if (bit_offset + sizeof(b) * 8 > bitbuffer->bits_per_row[0]) {  // Did not find a big enough package
        decoder_logf_bitbuffer(decoder, 1, __func__, bitbuffer, "short package. Row length: %u. Header index: %u", bitbuffer->bits_per_row[0], bit_offset);
        return DECODE_ABORT_LENGTH;
    }
    bitbuffer_extract_bytes(bitbuffer, 0, bit_offset, b, sizeof(b) * 8);

    /* Determine packet type based on first byte */
    if (b[0] == 0x41)
        type = MODEL_WH41;
    else if (b[0] == 0x43)
        type = MODEL_WH43;
    else {
        // Unknown type
        decoder_logf(decoder, 1, __func__, "Unknown device family: %02x", b[0]);
        return DECODE_FAIL_MIC;
    }

    /* Try WH41 packet (8-bit ID) */
    {
        uint8_t crc = crc8(b, 6, 0x31, 0x00);  // Compute CRC over first 6 bytes.
        uint8_t checksum = 0;
        for (unsigned n = 0; n < 7; ++n)
            checksum += b[n];

        if (crc == b[6] && checksum == b[7]) {
            // WH41 packet detected
            family   = b[0];
            id       = b[1];
            pm25     = ((b[2] & 0x3F) << 8) | b[3];
            pm100    = ((b[4] & 0x3F) << 8) | b[5];
            battery_bars = ((b[2] & 0x40) >> 4) | ((b[4] & 0xC0) >> 6);
            ext_power = battery_bars == 6 ? 1 : 0;
            battery_ok  = MIN(battery_bars * 0.2f, 1.0f);
            decoder_logf(decoder, 2, __func__, "Decoded as WH41");
            goto packet_found;
        }
    }

    /* Try WH43 packet (24-bit ID) */
    {
        uint8_t crc = crc8(b, 8, 0x31, 0x00);  // Compute CRC over first 8 bytes.
        uint8_t checksum = 0;
        for (unsigned n = 0; n < 9; ++n)
            checksum += b[n];

        if (crc == b[8] && checksum == b[9]) {
            // WH43 packet detected
            family   = b[0];
            id       = (b[1] << 16) | (b[2] << 8) | b[3];  // 24-bit ID
            pm25     = ((b[4] & 0x3F) << 8) | b[5];
            pm100    = ((b[6] & 0x3F) << 8) | b[7];
            battery_bars = ((b[4] & 0x40) >> 4) | ((b[6] & 0xC0) >> 6);
            ext_power = battery_bars == 6 ? 1 : 0;
            battery_ok  = MIN(battery_bars * 0.2f, 1.0f);
            decoder_logf(decoder, 2, __func__, "Decoded as WH43");
            goto packet_found;
        }
    }

    /* If neither CRC check passes, return failure */
    decoder_logf(decoder, 1, __func__, "Checksum error on packet");
    return DECODE_FAIL_MIC;

    /* clang-format off */
packet_found:
    /* Build data output (adjust field names, macros, and values as needed) */
    data = data_make(
        "model",            "",             DATA_COND, type == MODEL_WH41, DATA_STRING, "Fineoffset-WH0290",
        "model",            "",             DATA_COND, type == MODEL_WH43, DATA_STRING, "Fineoffset-WH43",
        "id",               "ID",           DATA_FORMAT, "%06x", DATA_INT, id,
        "battery_ok",       "Battery Level",  DATA_FORMAT, "%.1f", DATA_DOUBLE, battery_ok,
        "ext_power",        "External Power", DATA_STRING, ext_power == 1 ? "Yes" : "No",
        "pm2_5_ug_m3",      "2.5µm Fine Particulate Matter",  DATA_FORMAT, "%d µg/m3", DATA_INT, pm25/10,
        "estimated_pm10_0_ug_m3", "Estimate of 10µm Coarse Particulate Matter", DATA_FORMAT, "%d µg/m3", DATA_INT, (pm100/10),
        "family",           "Family",       DATA_FORMAT, "%02x", DATA_INT, family,
        "mic",              "Integrity",    DATA_STRING, "CRC",
        NULL);

    decoder_output_data(decoder, data);
    return 1;
}
```

I've gone through and made it an -R selectable decoder in my `include/rtl_433_devices.h`, as well as the `static char const *const output_fields_WH0290[]` and `r_device const fineoffset_WH0290`. It was just for my testing purposes since the WH0290/WH41/PM25 code wasn't called out explicitly like some of the others in `fineoffset.c`. I'm not including that here.

gdt, 2025-02-11:
Glad you are making progress!   The path to getting this in is normally a pull request.  It's ok to make one and mark it in the comments as not 100% baked yet.

I personally would want to see separate commits for each logical change; you have a bunch of separable fixes/improvements.  But this project doesn't tend to do that so you shouldn't really listen to me.

If you are not able to deal with git, I'm sure there is another way, but as soon as you've arrived at wanting others to read, build, and run code, then it IMHO belongs in a PR not pasted in.

andrewjmcginnis, 2025-02-11:
yeah, not previously familiar in editing anything on github. I just forked and changed the files in my own repo a few minutes ago. I'll make a PR in a little bit, or tomorrow morning. 

As far as individual commits, I kinda wrapped the original because the logic to determine 41 vs 43 made me pull the variables outside of the if statements to be useable in the data_make() function. The old code is still there, but I ended up rearranging it all. Gotta see what I can do for a more granular commit.

gdt, 2025-02-11:
you can git clone and work in git locally like one normally would and push.

refactoring to support both I didn't mean to suggest be separated.   It just sounded like you made some fixes that weren't really part of that, like um.   (I have a WH41 but not a 43 -- although my 41 is old and starting to be a little troubled.)

andrewjmcginnis, 2025-02-16:
> I have the 41 and 46D working now in my WeeWX server. I needed to update my rtl_433 and my weewx-sdr plugin (thanks to Andy Little for adding support for the 46D to weewx-sdr). The only thing not [...]

Hey side question- how are you pulling two aqi sensors into weewx? multiple weewx instances, extended schema?

ChrisM510, 2025-02-16:
Just one weewx instance... I am logging a WH41 (outside/external) and a WH46D (inside/internal). Indid extend the schema for both sensors (not using the default DB columns). The only issue I am having is that if I run weewx as a service, it doesn't log the WH46D (yet the WH41 logs fine... go figure). If I run weewx from the command line, everything works fine.

jwjenkin, 2025-06-19:
How's this coming along? I've got my WH43 itching, and I've so far failed at being able to write encoders :)