merbanan/rtl_433 #3181: rtl_433 stopped working

theMoe, 2025-02-03:
Hi,

I use rtl_433 on a raspberry pi 3 with raspbian OS (11). Version 20.11-1 of rtl_433 (apt install rtl-433) is used. With standard update/install command, the pi doesnt find a newer version like 22 or 24 of rtl_433.

I see this log since a couple of days and nothing happens:
```
rtl_433 version -128-NOTFOUND branch  at 202408270014 inputs file rtl_tcp RTL-SDR with TLS
MQTT: Publishing MQTT data to localhost port 1883
MQTT: Publishing device info to MQTT topic "sensors/rtl_433/P[protocol]/C[channel]".
Detached kernel driver
Found Fitipower FC0012 tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
```

Any idea how I could fix it?

Thanks!

Moe

theMoe, 2025-02-03:
I added autolevel and noise to my command

```
/usr/bin/rtl_433 -F log -Y autolevel -M utc -C si -M protocol -M level -M noise -F "mqtt://localhost:1883,retain=0,devices=sensors/rtl_433/P[protocol]/C[channel]" -R 200 -R 19 -R 141 -R 146 -R 183 -R 184
```

Log looks like:
```
rtl_433 version -128-NOTFOUND branch  at 202408270014 inputs file rtl_tcp RTL-SDR with TLS
MQTT: Publishing MQTT data to localhost port 1883
MQTT: Publishing device info to MQTT topic "sensors/rtl_433/P[protocol]/C[channel]".
Detached kernel driver
Found Fitipower FC0012 tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Auto Level: Current signal level 2.6 dB, estimated noise -13.0 dB
Auto Level: Current signal level 2.6 dB, estimated noise -6.0 dB
Auto Level: Current signal level 2.6 dB, estimated noise -2.1 dB
Auto Level: Current noise level 2.6 dB, estimated noise 0.9 dB
Auto Level: Current noise level 2.6 dB, estimated noise 2.4 dB
Auto Level: Current noise level 2.6 dB, estimated noise 2.6 dB
Auto Level: Current noise level 2.6 dB, estimated noise 2.6 dB
Auto Level: Current noise level 2.5 dB, estimated noise 2.6 dB
```

Does it help to get an idea of the problem?

zuckschwerdt, 2025-02-03:
That looks really really bad. Levels should never exceed -0 dB and you'd want a noise floor of -15 dB at least, usually you'd see -20 dB or even quieter.

Something is truely broken or jammed. Grab a sample or short recording and upload here as zip. https://triq.org/rtl_433/ANALYZE.html

theMoe, 2025-02-07:
After I completely disconnected the Pi from the power supply and unplugged the antenna, everything is working again. The reboot command was somehow not enough.

Another question, I use the following command in a service. Is it possible to send data to two MQTT brokers in one command?

```
rtl_433 -F json -M utc -C si -M protocol -M level -F "mqtt://localhost:1883,retain=0,devices=sensors/rtl_433/P[protocol]/C[channel]" -R 200 -R 19 -R 141 -R 146 -R 183 -R 184
```

zuckschwerdt, 2025-02-07:
Yes, it is possible to send data to two MQTT brokers in one command.

closed 2025-02-07 by zuckschwerdt

theMoe, 2025-02-07:
How would the command look like with two MQTT brokes?

Just two -F "mqtt://..."?

Where to place MQTT broker credentials?

zuckschwerdt, 2025-02-07:
You can use as many `-F` as you like.
`E.g. -F "mqtt://localhost:1883,user=USERNAME,pass=PASSWORD,retain=0,devices=rtl_433[/id]"`

theMoe, 2025-02-07:
So, the PI ran about 8 hours and stopped. After several reboots, I always get this error:

`usb_claim_interface error -6`

Any ideas how to fix it?

zuckschwerdt, 2025-02-07:
Reset the SDR by unplugging. Also try a better power supply or a powered hub.