merbanan/rtl_433 #3188: model : Emax-W6 weather station total rainfall

Guslogy, 2025-02-05:
Good day,
Is it possible to get the rainfall data for the model : Emax-W6 weather station not as the total rainfall in mm but rather the current rainfall in mm?
I would like to visualise my data in Home Assist in a graph showing the daily rainfall.

zuckschwerdt, 2025-02-05:
No, you have to derive hourly/daily/monthly… values yourself. We can deliver all the different timeframes and options.

Guslogy, 2025-02-05:
> No, you have to derive hourly/daily/monthly… values yourself. We can deliver all the different timeframes and options.

If you say you can deliver the different timeframes and options is that to use in Home Assistant?

zuckschwerdt, 2025-02-05:
With options I mean e.g. do you need the values for the last 60min or 24h period or the last full hour or day?

Solution would be to use a script like https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_relay.py
or to inject the values when a weather device is detected in https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

PR welcome.

gdt, 2025-07-01:
answered question

closed 2025-07-01 by gdt