merbanan/rtl_433 #3192: Calibo CloudFan

filmackay, 2025-02-09:
When I press buttons on the RF remote (433.92 MHz) I get a pair of RF messages (this button press is "fan speed 2, of 6"):

https://triq.org/pdv/#AAB03304040138039414582714819081909081909081908181819090818181819081909090818181818190818181818181819081818255+AAB032040101380394145827148190819090819090819081818190908181818190819090908181818181908181818181818190818355
https://triq.org/pdv/#AAB03304040134039414642714819081909081909081908181819090818181819081909090819081818190818181908181819081818255+AAB032040101340394146427148190819090819090819081818190908181818190819090908190818181908181819081818190818355

Any ideas how to create a decoder for this - I am assuming this does not have security/integrity checks in there that I can't work with.

inonoob, 2025-02-09:
Hey 

Have you tired rtl_433 -A "insert file name.of the capture" 

And show what kind of decoder is proposed for the command 

rtl_433 -X 

From there you can catch more messages and then the fun start to decode :)

inonoob, 2025-02-09:
You have already two codes to start with to be able to decode..from your file it seems to be repeated. But more Infos and more data are needed

A4 B9 E8 BB BB 80

A4 B9 E8 FB FB 80

zuckschwerdt, 2025-02-09:
@inonoob hint is good. With -A you should get and -X decoder something like this:
`-X 'n=calibo,m=OOK_PWM,s=300,l=900,g=1500,r=6000'`

gdt, 2025-07-01:
Status?   What's the path to closing this issue while capturing the wisdom in the sources?