merbanan/rtl_433 #3200: Steelmate TPMS decoder issues

e100, 2025-02-15:
I have some new after market TPMS that sometimes are detected by the Steelmate decoder -R 59 and I'm not sure if I should tweak Steelmate decoder or make a new one because they seem so similar yet different.

What I found is that sometimes my sensors send 73 bits, sometimes only 72 bits where the decoder only decodes 72 bits.
So I modified the code to decode 72 or 73 bits:
```
--- a/src/devices/steelmate.c
+++ b/src/devices/steelmate.c
@@ -42,13 +42,23 @@ static int steelmate_callback(r_device *decoder, bitbuffer_t *bitbuffer)
         uint8_t *b = bitbuffer->bb[row];
 
         //Length must be 72 bits to be considered a valid packet
-        if (bitbuffer->bits_per_row[row] != 72)
+        if ((bitbuffer->bits_per_row[row] != 72) && (bitbuffer->bits_per_row[row] != 73))
             continue; // DECODE_ABORT_LENGTH
 
```

Ignoring the 73rd bit, if it exists, it now decodes the signals from my TPMS, the checksum passes too.  However, the values returned from rtl_433 do not match what is displayed on the display panel included with my TPMS sensors. The values are close but not the same and I would expect them to be the same if rtl_433 was decoding the signal in the same manner as the display does. While assumptions are bad I would expect the display bundled with the TPMS sensors would properly decode the data so making rtl_433 decode to match the display seems to make sense to me.

The attached zip has a recording where the sensor sent the same data, one is 73 bits, the other is 72 bits. 

```
Display   rtl_433
30 PSI    34 PSI
12C/70F   71F
27 PSI    30.5 PSI
24 PSI    27 PSI
18 PSI    20 PSI
10 PSI    11.5 PSI
7 PSI     8 PSI
```

What I noticed is the higher the pressures the more rtl_433 diverged from the display.
My assumption was that the value was not transmitted in PSI but rather kPa. I found that multiplying the value by 3.15 to get Kpa, then divide by 6.895 to get PSI the values closely matched the TPMS display between 0 and 30 psi. I do not have a way to test beyond 30 PSI yet but that needs to be done to reach any final conclusions.

If we plot the original conversion to PSI to calculating it as kPA then converting to PSI we see the same thing as my real data above, the higher in pressure we go the more the two calculations diverge.

[image]

I also see some oddities in the decoded voltage and temperature from rtl_433 but I've not investigated either of those yet.

Was the original steelmate decoder correct or was it just close?
I'm not sure how I should proceed here, steelmate is either flawed and I should fix it or my TPMS are some close variant that should have its own decoder created.

My unit is branded R-Lake but it looks just like one of the models that SteelMate sells.

[cu8_samples.zip](https://github.com/user-attachments/files/18812397/cu8_samples.zip)

ProfBoc75, 2025-02-17:
@e100 :

About the temp, we need more samples to confirm, I guess it's a typo, i.e 21C/70F (not possible 12C/70F ) , as it could by some round up or down values.

About the pressure, the conversion looks not accurate, and not in PSI (scale 0.5) but in kPa scale 3

Try this to get both values

```
--- a/src/devices/steelmate.c
+++ b/src/devices/steelmate.c
@@ -73,6 +73,7 @@ static int steelmate_callback(r_device *decoder, bitbuffer_t *bitbuffer)

         int sensor_id      = (id1 << 8) | id2;
         float pressure_psi = p1 * 0.5f;
+        float pressure_kpa = p1 * 3.0f;
         int battery_mV     = tmpbattery_mV * 2;

         char sensor_idhex[7];
@@ -83,7 +84,8 @@ static int steelmate_callback(r_device *decoder, bitbuffer_t *bitbuffer)
                 "type",             "",             DATA_STRING, "TPMS",
                 "model",            "",             DATA_STRING, "Steelmate",
                 "id",               "",             DATA_STRING, sensor_idhex,
-                "pressure_PSI",     "",             DATA_DOUBLE, pressure_psi,
+                "pressure_PSI",     "",             DATA_FORMAT, "%.1f PSI", DATA_DOUBLE, pressure_psi,
+                "pressure_kPa",     "",             DATA_FORMAT, "%.1f kPa", DATA_DOUBLE, pressure_kpa,
                 "temperature_F",    "",             DATA_DOUBLE, (float)tempFahrenheit,
                 "battery_mV",       "",             DATA_INT,    battery_mV,
                 "mic",              "Integrity",    DATA_STRING, "CHECKSUM",
@@ -104,6 +106,7 @@ static char const *const output_fields[] = {
         "model",
         "id",
         "pressure_PSI",
+        "pressure_kPa",
         "temperature_F",
         "battery_mV",
         "mic",
```

Then try these 3 commands:

    rtl_433
    rtl_433 -C si
    rtl_433 -C customary

Add your options also.

1. The first will give you the 2 values, one in PSI based on the raw x 0.5 conversion, and one in kPa based on the raw x 3 conversion (my assumption)
2. The second line will give you both pressure values in kPa: the first in PSI is converted in kPA by rtl_433 from the above values
3. The third line will give you both pressure values in PSI: the second in kPa is converted in PSI by rtl_433.

This is only to let you compare, and up to you to add more values, more conversions, until you find the figures.

e100, 2025-02-20:
My TPMS is branded R-Lake model TP-S15, Steelmate has a model TP-S15 that looks identical other than the branding and both brands have the same specifications listed.
This makes me confident that using the steelmate decoder is correct.

The display unit can display in Bar or PSI, I choose to use Bar since I believe the raw data is likely to be some factor of Bar/Kpa not PSI.
My kit has external sensors limited to 6.0Bar, the internal sensors are rated to 7.9Bar.
Multiplying the pressure byte by 3.14 returns Kpa that is accurate across the whole 0-6.0Bar range that my sensors are rated for.  It would also work with the 7.9Bar sensors as 255 * 3.14 would be 800kPa = 8.0Bar

I still need to look deeper into the voltage and temperature and will do so before submitting a pull request.

Where I need some guidance is how to reconcile what I am observing with what the original developer observed.
I see two possible solutions:
1. The original code was just 'close enough' not accurate, changing the code to use kPa and multiply by 3.14 is the correct fix.
2. The original code was accurate for their sensors but inaccurate for mine. Do we make two decoders and how can we distinguish between the two since they seem to use the same protocol?

Data collected from my pressure gauge, the TPMS display and the byte value from the decoded data.
Gauge and Display in Bar
```
gauge   display raw_value
5.85    5.8     185
5.65    5.6     179
5.25    5.2     166
5.06    5.0     160
4.87    4.9     155
4.75    4.7     150
4.56    4.5     144
4.44    4.4     139
4.34    4.3     136
4.15    4.1     130
3.99    3.9     125
3.96    3.9     124
3.90    3.9     124
3.83    3.8     120
3.58    3.5     112
3.47    3.4     108
3.24    3.2     103
2.98    2.9     94
2.87    2.8     89
2.68    2.6     84
2.55    2.5     79
2.39    2.4     75
2.12    2.1     66
1.80    1.8     56
1.60    1.5     49
1.44    1.4     45
1.30    1.3     40
1.13    1.1     35
0.94    0.9     29
0.74    0.7     22
0.45    0.4     13
0.28    0.2     7
0.00    0.1     2
0.00    0.0     0
```

zuckschwerdt, 2025-02-20:
[File history](https://github.com/merbanan/rtl_433/commits/master/src/devices/steelmate.c) shows that #396 was the PR for this decoder (and there is also #395). But there is little info how the values were derived. Maybe @johnjore has some additional information? He wrote "I'm guessing temp, pressure and voltage as they "appear" to corrolate wth observed behavour. I dont have any documents that states the chipset outputs in fahrenheit, double PSI's or half mV." -- would be interesting what range he observed. From @e100's experiments I lean towards trusting those new factors. (Though having π as conversion factor is irrational ;)

johnjore, 2025-02-20:
Hm... I would create a new decoder unless you can get more data from existing users that R59 is outputting wrong data.

Using the value of pi, or floating point in general, on a battery operated sensor sounds a bit odd to me, also flipping between 72 and 73 packet lenghts is a bit strange.

Values I have were a result of tossing them in a freezer and room temp and multiple tire pressures, using the display that came with the sensor.

From memory, the Steelmate sensor I had matched what I found in a datasheet for a programmable Schrader? chip. Can't say for sure the Steelmate used that chip, or that the datasheet I found was the correct one, but the overlap in several areas and matching my display was enough for me at the time.

e100, 2025-02-21:
I suspect the extra bit might be signaling an alarm state.
These only seem to transmit when pressure changes or they detect rolling and most of my testing is actively changing the pressure. I'll need to go driving with them installed to see what data they send.

I doubt the sensor itself is doing any floating point. Each step of the sensor output likely represents some value of pascals, the math is done in the display.
Since we do not know precisely what the conversion factor is we can only estimate it based off of samples.
3.125, 3.136, 3.14 and 3.15 are all accurate enough where 3.136 has the least mean error from all of my samples.
If the value is Bar rather than kPa 0.03125 might make sense since its a simple fraction 1/32

The manual included with my sensors states the precision as:
At 0-50C
+- 0.1Bar
(0 ~ 4 Bar)
+-0.2 Bar
(4 ~ 8 Bar)

Up to about 35 PSI the original conversion factor vs using 3.14 for kPa end up with a difference that is close to the margin of error for the sensors.

70 * 0.5 = 35 PSI +- 1.45 ( 33.55 - 36.45)
70 * 3.14/6.895 = 31.87 PSI +- 1.45 ( 30.42 - 33.32)

Consider that 3.14/6.895 = 0.455, quit close 0.5.
We are talking about a difference of 0.045 in the conversion factor.

In PSI mode the display shows only whole numbers so it is quite possible that @johnjore and my sensors could both cause the display to show 33PSI while in reality we have slightly different pressures. 

While I have quite limited experience with pressure sensors all of them I have encountered used pascals for their native units, not PSI. Based on that I think the original conversion is likely wrong but close enough at lower pressures that it resulted in values that matched the display.

Graph with my samples, original conversion, and using 3.14 or 3.125:

[image]

ProfBoc75, 2025-02-21:
Hi @e100 : I found the same with 3.125 for Bar or 1/32 for kPa , and I like this ratio because 32 is a binary value, 2^5, easy for such simple device, just have to shift the bits to the right 5 times. raw value >> 5 

185 / 32 = 5.78125 Bar , your display is 5.8 Bar, looks good.
179 / 32 = 5.59375 Bar, your display is 5.6 Bar, same looks good too.

So for me, the raw value is probably Bar values scale 32. 

Another point, if for 255 / 32 = 7.96875 Bar, this is matching also the limits of 7.9 Bar of this sensor TP-S15

zuckschwerdt, 2025-02-21:
There is little difference to the originally guessed formula and the new regression analysis is solid.
Unless there are strong objections I'd like to see the Steelmate decoder updated to these findings.
Please PR a change including a summary of your findings and a mention of this issue.

ProfBoc75, 2025-02-22:
I discovered a non standard round up between Display and Bar values at the second decimal.

Round mechanism is usually xx.x0000 to xx.x4999 = xx.x , and xx.x5000 to xx.x9999 = xx.x+0.1

Here, with Bar values (bb.b = raw / 32), bb.b0000 to bb.b3999 = bb.b, then bb.b4000 to bb.b9999 = bb.b+0.1

Probably for security reason and may explain also the current wrong conversion ratio.

And it's not a little offset of +0.1 for all values, as you can see, the 5 bar value is matching the exact whole result of 160/32 = 5, in case of offset , the display would be 5.1 bar. Same for zero, not displayed 0.1 but 0, so no offset here.

display | raw_value | Bar (raw / 32) | Bar (Rounded 1 decimal) | Gap Display - Rounded
-- | -- | -- | -- | --
5.8 | 185 | 5.7813 | $${\color{lightgreen}5.8}$$ | $${\color{lightgreen}0.0}$$
5.6 | 179 | 5.5938 | $${\color{lightgreen}5.6}$$ | $${\color{lightgreen}0.0}$$
5.2 | 166 | 5.1875 | $${\color{lightgreen}5.2}$$ | $${\color{lightgreen}0.0}$$
5 | 160 | 5.0000 | $${\color{lightgreen}5}$$ | $${\color{lightgreen}0.0}$$
4.9 | 155 | 4.8438 | $${\color{red}4.8}$$ | $${\color{red}0.1}$$
4.7 | 150 | 4.6875 | $${\color{lightgreen}4.7}$$ | $${\color{lightgreen}0.0}$$
4.5 | 144 | 4.5000 | $${\color{lightgreen}4.5}$$ | $${\color{lightgreen}0.0}$$
4.4 | 139 | 4.3438 | $${\color{red}4.3}$$ | $${\color{red}0.1}$$
4.3 | 136 | 4.2500 | $${\color{lightgreen}4.3}$$ | $${\color{lightgreen}0.0}$$
4.1 | 130 | 4.0625 | $${\color{lightgreen}4.1}$$ | $${\color{lightgreen}0.0}$$
3.9 | 125 | 3.9063 | $${\color{lightgreen}3.9}$$ | $${\color{lightgreen}0.0}$$
3.9 | 124 | 3.8750 | $${\color{lightgreen}3.9}$$ | $${\color{lightgreen}0.0}$$
3.9 | 124 | 3.8750 | $${\color{lightgreen}3.9}$$ | $${\color{lightgreen}0.0}$$
3.8 | 120 | 3.7500 | $${\color{lightgreen}3.8}$$ | $${\color{lightgreen}0.0}$$
3.5 | 112 | 3.5000 | $${\color{lightgreen}3.5}$$ | $${\color{lightgreen}0.0}$$
3.4 | 108 | 3.3750 | $${\color{lightgreen}3.4}$$ | $${\color{lightgreen}0.0}$$
3.2 | 103 | 3.2188 | $${\color{lightgreen}3.2}$$ | $${\color{lightgreen}0.0}$$
2.9 | 94 | 2.9375 | $${\color{lightgreen}2.9}$$ | $${\color{lightgreen}0.0}$$
2.8 | 89 | 2.7813 | $${\color{lightgreen}2.8}$$ | $${\color{lightgreen}0.0}$$
2.6 | 84 | 2.6250 | $${\color{lightgreen}2.6}$$ | $${\color{lightgreen}0.0}$$
2.5 | 79 | 2.4688 | $${\color{lightgreen}2.5}$$ | $${\color{lightgreen}0.0}$$
2.4 | 75 | 2.3438 | $${\color{red}2.3}$$ | $${\color{red}0.1}$$
2.1 | 66 | 2.0625 | $${\color{lightgreen}2.1}$$ | $${\color{lightgreen}0.0}$$
1.8 | 56 | 1.7500 | $${\color{lightgreen}1.8}$$ | $${\color{lightgreen}0.0}$$
1.5 | 49 | 1.5313 | $${\color{lightgreen}1.5}$$ | $${\color{lightgreen}0.0}$$
1.4 | 45 | 1.4063 | $${\color{lightgreen}1.4}$$ | $${\color{lightgreen}0.0}$$
1.3 | 40 | 1.2500 | $${\color{lightgreen}1.3}$$ | $${\color{lightgreen}0.0}$$
1.1 | 35 | 1.0938 | $${\color{lightgreen}1.1}$$ | $${\color{lightgreen}0.0}$$
0.9 | 29 | 0.9063 | $${\color{lightgreen}0.9}$$ | $${\color{lightgreen}0.0}$$
0.7 | 22 | 0.6875 | $${\color{lightgreen}0.7}$$ | $${\color{lightgreen}0.0}$$
0.4 | 13 | 0.4063 | $${\color{lightgreen}0.4}$$ | $${\color{lightgreen}0.0}$$
0.2 | 7 | 0.2188 | $${\color{lightgreen}0.2}$$ | $${\color{lightgreen}0.0}$$
0.1 | 2 | 0.0625 | $${\color{lightgreen}0.1}$$ | $${\color{lightgreen}0.0}$$
0 | 0 | 0.0000 | $${\color{lightgreen}0}$$ | $${\color{lightgreen}0.0}$$

ProfBoc75, 2025-02-22:
@e100 : About the 73 bits of data, try with `-Y filter=5` option, this will remove some noise. Take care about the antenna and keep the sdr dongle as far as possible from rf interference from the PC / PI. Use a USB cable between dongle and computer, it will help.

filter option may impact other sensors, not to be used in normal situation.

I just create a PR with our findings, please review it and let me know.

e100, 2025-02-23:
@ProfBoc75 
Your changes look good so far other than I would have also included:
```
-        if (bitbuffer->bits_per_row[row] != 72)
+        if ((bitbuffer->bits_per_row[row] != 72) && (bitbuffer->bits_per_row[row] != 73))
```

Would it be appropriate to rename steelmate.c to tpms_steelmate.c so it matches the naming of other tpms decoders?

My SDR is connected to a USB hub that has about 3 ft of cable between hub and computer.
In one of my earlier posts I included a zip file with a 72 and 73bit recording maybe someone who knows this stuff better than I could determine if it is noise or not.

I am also looking at the temperature and voltage.

The temperature in the original decoder was assumed to be Fahrenheit, I believe it is actually raw_value - 50 = Deg C
I looked and see that 9 other TPMS use  raw - 50 = deg C so that seems rather common.

Looking at the samples in rtl_433_test repo I think it is quite plausible that the raw values observed happened to be very close to matching the real temp in Fahrenheit.

Looking in rtl_433_tests/tests/steelmate-TPMS/01/
gfile001.json 75F vs 75-50 = 25C = 77F
gfile005.json 81F vs 81-50 = 31C = 87.8F
gfile006.json 74F vs 74-50 = 24C = 75.2F
gfile006.json 76F vs 76-50 = 26C = 78.8F
gfile015.json 77F vs 77-50 = 27C = 80.6F

My TPMS display only shows temperature in Celsius and raw - 50 = Deg C is exactly what I observe:
```
raw value   display C
41          -9
46          -4
47          -3
55          5
56          6
58          8
61          11
65          15
67          17
71          21
76          26
```

I think the battery_mV might not be correct too but I've also not determined much about it.
One observation, if the pressure decreases the raw value for battery_mV becomes 0xFF temporarily.
So I think that 0xFF might represent an alarm, not a voltage.

Here is the data I collected, changing the voltage did not cause transmission so I had to do something like change the pressure to trigger data sending:
```
volts   action                      value
2.1v    power loss/restore          0x5C 092 01011100
2.1v    decrease pressure            nothing
2.1v    increase pressure           nothing
2.2v    decrease pressure            0xA5 165 10100101
2.2v    increase pressure           0xA5 165 10100101
2.1v    decrease pressure           nothing
2.15v   no change other than voltage 0xFF 255 11111111
-> followed by                      0xC2 194 11000010
2.12v   increase pressure           0xC2 194 11000010
2.12v   decrease pressure            0xFF 255 11111111
-> followed by                      0xC2 194 11000010
2.1v    increase pressure           nothing
2.1v    decrease pressure            nothing
2.11v   decrease pressure           0xFF 255 11111111
2.11v   increase pressure           0xC2 194 11000010
2.2v    increase pressure           0xA6 166 10100110
2.3v    increase pressure           0x9C 156 10011100
2.35v   increase pressure           0x97 151 10010111
2.4v    increase pressure           0x93 147 10010011
2.45v   increase pressure           0x8E 142 10001110
2.5v    increase pressure           0x89 137 10001001
2.55v   increase pressure           0x90 144 10010000
2.60v   increase pressure           0x81 129 10000001
2.65v   increase pressure           0x7C 124 01111100
2.7v    increase pressure           0x78 120 01111000
2.75v   increase pressure           0x7D 125 01111101
2.75v   decrease pressure            0xFF 255 11111111
-> followed by                      0x73 115 01110011
2.8v    increase pressure           0x6E 110 01101110
2.85v   increase pressure           0x6A 106 01101010
2.9v    increase pressure           0x65 101 01100101
2.95v   increase pressure           0x60 096 01100000
3.0v    increase pressure           0x5C 092 01011100
3.05v   increase pressure           0x57 087 01010111
3.1v    increase pressure           0x53 083 01010011
3.1v    power loss/recovery         0x60 096 01100000
3.05v   power loss/recovery         0x65 101 01100101
3.0v    power loss/recovery         0x68 104 01101000
2.95v   power loss/recovery         0x6D 109 01101101
2.9v    power loss/recovery         0x71 113 01110001
2.85v   power loss/recovery         0x76 118 01110110
2.8v    power loss/recovery         0x79 121 01111001
2.7v    power loss/recovery         0x82 130 10000010
2.6v    power loss/recovery         0x8C 140 10001100
2.5v    power loss/recovery         0x93 147 10010011
2.4v    power loss/recovery         0x9B 155 10011011
2.3v    power loss/recovery         0xA5 165 10100101
2.2v    power loss/recovery         0x5C 092 01011100
```

2.11v seems to be the lowest my sensors will operate at.

I did take my sensors on a drive, once rotation is detected they transmitted every three minutes.

e100, 2025-02-23:
> Here, with Bar values (bb.b = raw / 32), bb.b0000 to bb.b3999 = bb.b, then bb.b4000 to bb.b9999 = bb.b+0.1

@ProfBoc75 

Perhaps that rounding was used because it resulted in values that more closely matched the real pressure values than when using standard rounding.

In my testing 0.03136 and 0.0314 were more accurate based on my samples.

Of the three times where rounding did not match the display for you, using 0.03136 would have resulted in proper rounding matching the display:
```
display raw value
2.4     75  2.352
4.4     139 4.35904
4.9     155 4.8608
```

I did not check if it would have broke any of the other values tho

buxtronix, 2025-05-02:
I have been playing with this recently also. I have an "Ebat" branded TPMS which is a rebranded Steelmate (the FCC ID points to a Steelmate and it's roughly the same protocol). However this decoder did not pick it up. I played around with URH and got the following data reads from it:

```
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa6aaa6a669a66aaaa69a65555a595 [Pause: 978 samples]
aaaa5555 [Pause: 992 samples]
aaaaaaaa6aaa6a669a66aaaa69a65555a595 [Pause: 978 samples]
aaaa5555 [Pause: 33834 samples]
```

which decodes to:

```
000000000000000000000000000000000000001015250094ffce [Pause: 978 samples]
00ff [Pause: 992 samples]
00001015250094ffce [Pause: 978 samples]
00ff
```

So there is a packet of 208 bits as well as one with 72 bits. So I modified the decoder for this, and now it decodes well. I'll work on a PR for this update.

I also found discrepancies with the pressure readings so I built a transmitter with an ESP32 and CC1101 to directly manipulate and send data to the display to figure out what was what. I found the following:

- I sent every pressure value from 0x0 to 0xFF and indeed it seems to display exactly Bar*32 with correct rounding. A few rounding discrepancies are on the PSI setting. The fewest are when I try `PSI = <value>*0.4549`, although this is really a display issue, not a sensor encoding issue.

- My display is a compact one and only displays PSI/Bar. No voltage or temperature, but it does have alerts when they're out of range. So I haven't been able to fuzz the volt or temp values but @e100 seems to have useful values (It's peculiar that the value increases as the voltage decreases) Still, I found alert triggers.

These alerts are:

 - If voltage value is > 185 and < 254 then the display gives a low voltage alert. It's probably the display deciding the alert threshold.
 - If voltage value is 0xfe, the display gives a "slow" beep alert. Sent by the sensor for a slow leak.
 - If voltage value is 0xff, the display gives a "fast" beep alert. Sent by the sensor for a fast leak or 0 pressure
 - If the temperature value is >130 I get a temperature alert.
 - There are also low/high pressure thresholds but they are decided by the display.

I'm happy to put these updates into a PR also.

buxtronix, 2025-05-02:
I played around with the voltage values from @e100 and found a formula that seems to follow the displayed trend quite closely (rounding is off but the line seems to fit somewhat). I discarded very low values at ~2.1v as the sensor probably starts to glitch given the weird values that appear.

[image]

`volts = 3.0 - (raw-100)/90`

Obviously this is not completely accurate, but it's already much closer than what the current decoder does.

ProfBoc75, 2025-05-02:
@buxtronix : There is already a PR and a branch for this sensor, #3209 , I was ready to merge but I was too busy last weeks to move forward. You can use this branch based on your findings.

buxtronix, 2025-07-03:
Right now all the fixes are in the `fix-steelmate_TPMS` branch and not yet merged to master. Though this branch seems to be working well based on the devices I have.

ProfBoc75, 2025-07-03:
@buxtronix , @gdt , Merged.

gdt, 2025-07-03:
Great, so are we set to close the issue, or do we need to capture some info that isn't in the code?

ProfBoc75, 2025-07-04:
> Great, so are we set to close the issue, or do we need to capture some info that isn't in the code?

The point was wrong pressure formula now corrected, then battery level and temp have been also refined. I guess this issue can be closed yes.

closed 2025-07-04 by gdt