merbanan/rtl_433 #3206: Danfoss - RET B RF Support

liambresnahan, 2025-02-20:
Hi All, 

I have a Danfoss RET B RF that I was hoping rtl_433 supported but it doesn't appear to. I see the Danfoss CFR support, and would hope that the RET B works in a similar way. 

What would it take to get this additional thermostat supported please?

zuckschwerdt, 2025-02-20:
Try to get a sample of a transmission and upload here as zip. Then we can work out the timings.
https://triq.org/rtl_433/ANALYZE.html

gdt, 2025-07-04:
It was a question, and promptly responded to with hints for how to proceed, and then nothing for months, and hence closed as inactive, because I don't see leaving questions like that open as benefiting the project -- and we have too many open issues.

If you're going to work towards a decoder see the link above and capture and analyze away!

liambresnahan, 2025-07-04:
Thanks for clarifying. I was just trying to understand how it was closed as it is marked as completed but it seems to have been stale (per your comment). I didn't know if I was missing an implementation outside of this ticket.

gdt, 2025-07-04:
I am unclear on github closure reasons and will try to be more careful.    Was just stale.

As for whether there is something else, I'm not aware, but that doesn't mean much.   I always suggest checking out the latest sources from git and building and running them.   That's what you'll need to do to test your changes anyway ;-)

closed 2025-07-04 by gdt