merbanan/rtl_433 #3210: Hidden option not documented

llamatrails, 2025-02-26:
re: rtl_433 version 25.02 branch  at 202502191252

rtl_433 provides an http server that can be accessed from any web browser.
It isn't documented in the command line usage output, or in the rtl_433.conf file(s).
It seems to work very well, as it is easier to read than just having the output scroll on the terminal.

Could it please be documented by updating the rtl_433.conf file and the usage output?

Below are different command line executions showing it starting up with and without a setting in the rtl_433.conf file.

        **Release file used**
$ ls -l rtl_433*
-rwxr-xr-x 1 rick rick 777568 Feb 19 06:55 rtl_433
-rw-r--r-- 1 rick rick 432590 Feb 19 06:55 rtl_433-soapysdr-openssl11-Linux-armhf-25.02.zip
$ unzip -l rtl_433-soapysdr-openssl11-Linux-armhf-25.02.zip
Archive:  rtl_433-soapysdr-openssl11-Linux-armhf-25.02.zip
  Length      Date    Time    Name
   777568  2025-02-19 06:55   rtl_433
   777568                     1 file

        **no "option http" in rtl_433.conf**
$ rtl_433
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Reading conf from "rtl_433.conf".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

        **no "option http" in rtl_433.conf, using command line**
$ rtl_433 -F http
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Reading conf from "rtl_433.conf".
HTTP server: Starting HTTP server at 0.0.0.0 port 8433
HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

        **"option http" in rtl_433.conf, using command line**
$ rtl_433
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Reading conf from "rtl_433.conf".
HTTP server: Starting HTTP server at 0.0.0.0 port 8433
HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

        **option not shown in /usr/local/etc/rtl_433/rtl_433.example.conf**
$ grep http /usr/local/etc/rtl_433/rtl_433.example.conf
# https://github.com/merbanan/rtl_433_tests/tree/master/tests/elro/db270/01
# https://github.com/merbanan/rtl_433_tests/tree/master/tests/euroster/3000tx/01
# https://github.com/merbanan/rtl_433_tests/tree/master/tests/Byron-BY101 and Byron-BY34

        **usage not showing option for "http"**
$ rtl_433 -h
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Reading conf from "rtl_433.conf".
Generic RF data receiver and decoder for ISM band devices using RTL-SDR and SoapySDR.
Full documentation is available at https://triq.org/
Usage:
  A "rtl_433.conf" file is searched in "./", XDG_CONFIG_HOME e.g. "$HOME/.config/rtl_433/",
  SYSCONFDIR e.g. "/usr/local/etc/rtl_433/", then command line args will be parsed in order.
                = General options =
  [-V] Output the version string and exit
  [-v] Increase verbosity (can be used multiple times).
       -v : verbose notice, -vv : verbose info, -vvv : debug, -vvvv : trace.
  [-c <path>] Read config options from a file
                = Tuner options =
  [-d <RTL-SDR USB device index> | :<RTL-SDR USB device serial> | <SoapySDR device query> | rtl_tcp | help]
  [-g <gain> | help] (default: auto)
  [-t <settings>] apply a list of keyword=value settings to the SDR device
       e.g. for SoapySDR -t "antenna=A,bandwidth=4.5M,rfnotch_ctrl=false"
       for RTL-SDR use "direct_samp[=1]", "offset_tune[=1]", "digital_agc[=1]", "biastee[=1]"
  [-f <frequency>] Receive frequency(s) (default: 433920000 Hz)
  [-H <seconds>] Hop interval for polling of multiple frequencies (default: 600 seconds)
  [-p <ppm_error>] Correct rtl-sdr tuner frequency offset error (default: 0)
  [-s <sample rate>] Set sample rate (default: 250000 Hz)
  [-D restart | pause | quit | manual] Input device run mode options.
                = Demodulator options =
  [-R <device> | help] Enable only the specified device decoding protocol (can be used multiple times)
       Specify a negative number to disable a device decoding protocol (can be used multiple times)
[-X <spec> | help] Add a general purpose decoder (prepend -R 0 to disable all decoders)
  [-Y auto | classic | minmax] FSK pulse detector mode.
  [-Y level=<dB level>] Manual detection level used to determine pulses (-1.0 to -30.0) (0=auto).
  [-Y minlevel=<dB level>] Manual minimum detection level used to determine pulses (-1.0 to -99.0).
  [-Y minsnr=<dB level>] Minimum SNR to determine pulses (1.0 to 99.0).
  [-Y autolevel] Set minlevel automatically based on average estimated noise.
  [-Y squelch] Skip frames below estimated noise level to reduce cpu load.
  [-Y ampest | magest] Choose amplitude or magnitude level estimator.
                = Analyze/Debug options =
  [-A] Pulse Analyzer. Enable pulse analysis and decode attempt.
       Disable all decoders with -R 0 if you want analyzer output only.
  [-y <code>] Verify decoding of demodulated test data (e.g. "{25}fb2dd58") with enabled devices
                = File I/O options =
  [-S none | all | unknown | known] Signal auto save. Creates one file per signal.
       Note: Saves raw I/Q samples (uint8 pcm, 2 channel). Preferred mode for generating test files.
  [-r <filename> | help] Read data from input file instead of a receiver
  [-w <filename> | help] Save data stream to output file (a '-' dumps samples to stdout)
  [-W <filename> | help] Save data stream to output file, overwrite existing file
                = Data output options =
  [-F log | kv | json | csv | mqtt | influx | syslog | trigger | null | help] Produce decoded output in given format.
       Append output to file with :<filename> (e.g. -F csv:log.csv), defaults to stdout.
       Specify host/port for syslog with e.g. -F syslog:127.0.0.1:1514
  [-M time[:<options>] | protocol | level | noise[:<secs>] | stats | bits | help] Add various meta data to each output.
  [-K FILE | PATH | <tag> | <key>=<tag>] Add an expanded token or fixed tag to every output line.
  [-C native | si | customary] Convert units in decoded output.
  [-n <value>] Specify number of samples to take (each sample is an I/Q pair)
  [-T <seconds>] Specify number of seconds to run, also 12:34 or 1h23m45s
  [-E hop | quit] Hop/Quit after outputting successful event(s)
  [-h] Output this usage help and exit
       Use -d, -g, -R, -X, -F, -M, -r, -w, or -W without argument for more help

eabase, 2025-03-02:
Which options are missing? 
`-l` and `-F` ?

llamatrails, 2025-03-02:
That "http" exists as an output option, in the "help" output and in the rtl_433.example.conf file.

[-F log | kv | json | csv | mqtt | influx | syslog | trigger | null | help]

should show "http" with info on what it does, something like the following

[-F http | log | kv | json | csv | mqtt | influx | syslog | trigger | null | help] 
Starts a web server: i.e. -F http
HTTP server: Starting HTTP server at 0.0.0.0 port 8433
HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .

eabase, 2025-03-02:
@zuckschwerdt 

I'm thinking to help update some `*.md` files. 
Do you take PR's? (There are a lot of them never closed..)

zuckschwerdt, 2025-03-02:
There is a noticeable lack of free time for PR's right now, sorry. I hope to merge some new-decoder PR's soon.

I might clean up and document the http feature also. It's only an incomplete experiment but maybe just good enough for now.

commit b91e971a (2025-03-30): docs: Add rtl_tcp and http output docs (closes #3210)
closed 2025-03-30 by zuckschwerdt