merbanan/rtl_433 #3215: Correcting Build instructions for Windows in `BUILDING.md`

eabase, 2025-03-02:
I'd like to see some improvement in the documentation for Building on Windows.

In particular, the section for `Visual Studio 2017` need to be updated. 
As it is presented today, it is useless since it is not reproducible.
* https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md

Instead I can provide detailed build/install instructions for using `Visual Studio 2022` + `SoapySDR` + `RTL-SDR`, via **CMake**.

---

### A Quick Build Summary for Windows-11 

Assumptions:

* Have installed native Windows versions of:
  - `LIBUSB` - also included below
  - `RTL-SDR` - also included below
  - `SoapySDR` - optional?
  - `Visual Studio 2022`
  - `CMake`
  - `Windows-11`
* You are able to start a **"VS Dev Shell"** in Admin mode, in `Windows Terminal`.
* You know how to set and order the `User` and `Machine` PATH in Windows

Here are the steps. 
<sub>Please ensure to replace the `\path\to\` part with your own PATH.</sub>

```bash
#----------------------------------------------------------------------
# Quick Build - Summary 
#----------------------------------------------------------------------

#------------------------------------------------
# 1. Install Requirements
#------------------------------------------------

# (a) For OpenSSL
winget install -e --id 'FireDaemon.OpenSSL'

# (b) For vcpkg [pkgconfig, pthreads]
# Which is needed to install pkgconfig and pthreads...
git clone https://github.com/microsoft/vcpkg.git
cd vcpkg; .\bootstrap-vcpkg.bat -disableMetrics

# (b.1) Configure the VCPKG_ROOT environment variable.
# Make these permanent, once confirmed to work.
$env:VCPKG_ROOT = "C:\path\to\vcpkg"
$env:PATH = "$env:VCPKG_ROOT;$env:PATH"
$env:VCPKG_DISABLE_METRICS=1

# (b.2) Add dependencies and project files
vcpkg version
vcpkg new --application
vcpkg add port fmt

vcpkg list
vcpkg search pkgconf
vcpkg search pthread

# (b.3) Install 2 most suitable packages
vcpkg install pkgconf:x64-windows
vcpkg install pthreads:x64-windows-static

# (c) libusb
# 	Download latest release from here:
#	https://github.com/libusb/libusb/releases
# 	Extract to a suitable directory. For example: "C:\mylib\libusb"

# (d) librtlsdr
# 	It, seem that the best way to proceed, is to create your own library from this.
#	(1) Create a suitable library directory. For example: "C:\mylib\rtl-sdr"
#
# 	(2) Download latest release from here:
#		https://github.com/rtlsdrblog/rtl-sdr-blog/releases
#		Put extracted files into `./lib/*`
#
#	(3) Copy "include" folder from repo, into your own.
#		git clone https://github.com/rtlsdrblog/rtl-sdr-blog 
#		cd rtl-sdr-blog; 
#		cp .\include C:\mylib\rtl-sdr
#
#	(4) You library structure should be: 
#		tree -C -L 2 -P "*.dll|*.h|*.lib"
#		.
#		├── include
#		│   ├── reg_field.h
#		│   ├── rtl-sdr.h
#		│   ├── rtl-sdr_export.h
#		│   ├── rtlsdr_i2c.h
#		│   ├── tuner_e4k.h
#		│   ├── tuner_fc0012.h
#		│   ├── tuner_fc0013.h
#		│   ├── tuner_fc2580.h
#		│   └── tuner_r82xx.h
#		└── lib
#			├── convenience_static.lib
#			├── libgetopt_static.lib
#			├── msvcr100.dll
#			├── pthreadVC2.dll
#			├── rtlsdr.dll
#			├── rtlsdr.lib
#			└── rtlsdr_static.lib
#

#------------------------------------------------
# 2. Add Locations
#------------------------------------------------

# (a) OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR)

$OPENSSL_ROOT_DIR='C:\Program Files\OpenSSL'
-DOPENSSL_ROOT_DIR="$OPENSSL_ROOT_DIR"

# (b.1) vcpkg.exe
# Already set in (b.1) above.

# (b.2) pkconfig.exe

$PKG_CONFIG_EXECUTABLE="C:\path\to\vcpkg\installed\x64-windows\tools\pkgconf\pkgconf.exe"
-DPKG_CONFIG_EXECUTABLE="$PKG_CONFIG_EXECUTABLE"

# (b.3) pthreads
# NOTE: This is OPTIONAL for rtl_433

$THREADS_PTHREADS_INCLUDE_DIR='C:\path\to\vcpkg\installed\x64-windows-static\include'
$THREADS_PTHREADS_WIN32_LIBRARY='C:\path\to\vcpkg\installed\x64-windows-static\lib\pthreadVC3.lib'
# -DTHREADS_PTHREADS_INCLUDE_DIR="$THREADS_PTHREADS_INCLUDE_DIR"
# -DTHREADS_PTHREADS_WIN32_LIBRARY="$THREADS_PTHREADS_WIN32_LIBRARY"
-DTHREADS_PTHREADS_WIN32_LIBRARY="$THREADS_PTHREADS_WIN32_LIBRARY" -DTHREADS_PTHREADS_INCLUDE_DIR="$THREADS_PTHREADS_INCLUDE_DIR" 

$THREADS_USE_PTHREADS_WIN32=true
-DTHREADS_USE_PTHREADS_WIN32=true

# (c) Could NOT find LibUSB (missing: LibUSB_LIBRARY LibUSB_INCLUDE_DIR)

$LIBUSB_INCLUDE_DIR="C:\mylib\libusb\include"
$LIBUSB_LIBRARY="C:\mylib\libusb\VS2022\MS64\static\libusb-1.0.lib"		# has only *.lib
-DLibUSB_INCLUDE_DIR="$LIBUSB_INCLUDE_DIR" -DLibUSB_LIBRARY="$LIBUSB_LIBRARY" 

# (d) Could NOT find LibRTLSDR (missing: LibRTLSDR_LIBRARY LibRTLSDR_INCLUDE_DIR)

$RPATH='C:\mylib\rtl-sdr'
$RTLSDR_INCLUDE_DIR="$RPATH\include"
$RTLSDR_LIBRARY="$RPATH\lib\rtlsdr.lib"
-DLibRTLSDR_INCLUDE_DIR="$RTLSDR_INCLUDE_DIR" -DLibRTLSDR_LIBRARY="$RTLSDR_LIBRARY"

#------------------------------------------------
# 3. Add Other Build Options
#------------------------------------------------
-DFORCE_COLORED_BUILD:BOOL=ON
-DENABLE_RTLSDR=ON -DENABLE_SOAPYSDR=ON -DENABLE_OPENSSL=ON -DBUILD_DOCUMENTATION=OFF -DCMAKE_BUILD_TYPE=Release

#------------------------------------------------
# 4. Construct final build config
#------------------------------------------------
# NOTE: 
# 	- To manually add permanent environment variables, run: 
# 		SystemPropertiesAdvanced
# 	- 
#------------------------------------------------

# Enable logs of session, using Tee:
$XXLOG='log_install_rtl_433.txt'
# Then add:  | Tee-object -Append $XXLOG

# Open VS DEV Admin Powershell (from Windows Terminal)
git clone --depth=1 https://github.com/merbanan/rtl_433

# From above parts:

$OPENSSL_ROOT_DIR='C:\Program Files\OpenSSL'
$PKG_CONFIG_EXECUTABLE="C:\path\to\vcpkg\installed\x64-windows\tools\pkgconf\pkgconf.exe"
$LIBUSB_INCLUDE_DIR="C:\mylib\libusb\include"
$LIBUSB_LIBRARY="C:\mylib\libusb\VS2022\MS64\static\libusb-1.0.lib"
$RPATH='C:\mylib\rtl-sdr'
$RTLSDR_INCLUDE_DIR="$RPATH\include"
$RTLSDR_LIBRARY="$RPATH\lib\rtlsdr.lib"

# configure
cmake -S .. -B . -Ax64 -Wno-dev -Wno-deprecated --fresh --install-prefix 'C:\Program Files\rtl_433' -DCMAKE_BUILD_TYPE=Release `
-DOPENSSL_ROOT_DIR="$OPENSSL_ROOT_DIR" `
-DPKG_CONFIG_EXECUTABLE="$PKG_CONFIG_EXECUTABLE" `
-DLibUSB_INCLUDE_DIR="$LIBUSB_INCLUDE_DIR" -DLibUSB_LIBRARY="$LIBUSB_LIBRARY" `
-DLibRTLSDR_INCLUDE_DIR="$RTLSDR_INCLUDE_DIR" -DLibRTLSDR_LIBRARY="$RTLSDR_LIBRARY" `
-DENABLE_RTLSDR=ON -DENABLE_SOAPYSDR=ON -DENABLE_OPENSSL=ON -DBUILD_DOCUMENTATION=OFF --loglevel=VERBOSE -G "Visual Studio 17 2022"

# build 
cmake --build . --config Release --parallel 8

# install
cmake --build . --config Release --target install

# Test if *.exe works
cd C:\Program Files\rtl_433\bin
.\rtl_433.exe -V

#------------------------------------------------
# 5. Testing with: RTL-SDR
#------------------------------------------------
# Refresh (Desktop) - This should reload registry
# Start a new Powershell Admin shell.
# cd C:\Program Files\rtl_433\bin

rtl_433.exe -V
rtl_433.exe -h
rtl_433.exe -R help		# List available device protocols

# Now you need to find a signal, you know what it is, that you can test this on. 

```

> [!IMPORTANT]
> `VS 2022` actually comes with it's own `vcpkg.exe` version embedded in a long path. 
> and is proably already set in your environment `$env:VCPKG_ROOT`. However, using 
> the VS provided version from elsewhere, will require you to create a `vcpkg.json` 
> config file, in the current directory. This is annoying AF (or at least not clear why.)
> So instead downlaod and install in a separate directory, where all your VC packages will live.
> Then set: `$env:VCPKG_ROOT="C:\path\to\vcpkg"`.

Hope this helps!

eabase, 2025-03-02:
It should be noted that the build step above, will spit out a few compiler and linker `warnings`, that should be fixed. 

```bash
# cmake --build . --config Release --parallel 8

data.c(155,9): 		warning C4068: unknown pragma 'GCC'
sdr.c(960,9): 		warning C4068: unknown pragma 'GCC' 
mongoose.c(4992,12): 	warning C4996: 'PEM_read_bio_DHparams': Since OpenSSL 3.0
output_udp.c(62,13): 	warning C4005: '_POSIX_HOST_NAME_MAX': macro redefinition

# warning LNK4098
libusb-1.0.lib(core.obj) : MSIL .netmodule or module compiled with /GL found; 
restarting link with /LTCG; add /LTCG to the link command line to improve linker performance
LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; 
use /NODEFAULTLIB:library [C:\path\to\rtl_433\build\src\rtl_433.vcxproj]
```

eabase, 2025-03-02:
To build `SoapySDR` check out:
* https://github.com/pothosware/SoapyRTLSDR/issues/79

eabase, 2025-03-02:
There is another issue how one of the CMake functions is collecting git tags:
`-- Using Git version tag: -128-NOTFOUND on master at 202503021839`

eabase, 2025-03-02:
Tried to enable documentation generation. Results in:
- Several warnings about `obsolete` items. - Need to run: `doxygen -u Doxyfile`
- A warning about invalid enum for `PAPER_TYPE`:
  `CUSTOMBUILD : warning : argument 'a4wide' for option PAPER_TYPE is not a valid enum value`
- An error about too many nodes:

```bash
error : Included by graph for 'decoder.h' not generated, too many nodes (241), 
threshold is 200. Consider increasing DOT_GRAPH_MAX_NODES.
```

Replaced:
- `./Doxyfile.in:2437:DOT_GRAPH_MAX_NODES = 200`, changed to `400`, and ran `doxygen -u Doxyfile.in`.
- `./Doxyfile.in:2055:PAPER_TYPE  = a4wide`,  changed to `a4`. 

> [!WARNING]
> Enabling running documentation on too many parallel CPU threads will max out your CPU', 
> because doxygen's `Dot` will generate 781 graphs... 

With these fixes the Doc generation works without issues.

zuckschwerdt, 2025-07-04:
This is great information and a nice guide. It's many steps and details though, we can't verify or maintain this.
It would be a nice Wiki page though. Anybody on Windows feel free to check transfer this.