merbanan/rtl_433 #3216: Fixing Compiler & Linker warnings on Windows Builds

eabase, 2025-03-02:
It should be noted that the build step in #3215 , will spit out a few compiler and linker `warnings`, that should be fixed. 

```bash
# cmake --build . --config Release --parallel 8

data.c(155,9): 		warning C4068: unknown pragma 'GCC'
sdr.c(960,9): 		warning C4068: unknown pragma 'GCC' 
mongoose.c(4992,12): 	warning C4996: 'PEM_read_bio_DHparams': Since OpenSSL 3.0
output_udp.c(62,13): 	warning C4005: '_POSIX_HOST_NAME_MAX': macro redefinition

# warning LNK4098
libusb-1.0.lib(core.obj) : MSIL .netmodule or module compiled with /GL found; 
restarting link with /LTCG; add /LTCG to the link command line to improve linker performance
LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; 
use /NODEFAULTLIB:library [C:\path\to\rtl_433\build\src\rtl_433.vcxproj]
```

IDK, the code base so can't help here.

zuckschwerdt, 2025-03-02:
The only warning we can address is the macro redefinition, it's fixed now.

closed 2025-03-02 by zuckschwerdt

eabase, 2025-03-02:
Just want to add some possible fixes:

[1] https://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Pragmas.html
[2] https://docs.openssl.org/master/man3/PEM_read_bio_PrivateKey/#description
[3] https://learn.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-warning-lnk4098?view=msvc-170

Things to fix:
- [ ] [C4068](https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-1-c4068?view=msvc-170) - Unknown pragma 'GCC' (See: [1])
- [ ] [C4996](https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-3-c4996?view=msvc-170) - Deprecated cryptographic algorithm: DH (See: [2]) 
- [x] [C4005](https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-1-c4005?view=msvc-170) - Duplicate Macro definition
- [ ] [LNK4098](https://learn.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-warning-lnk4098?view=msvc-170) -  Linking with incompatible libraries

For **`C4068`**:
```cpp
#pragma GCC diagnostic push 
#pragma GCC diagnostic ignored "-Wunknown-pragmas"
#pragma GCC diagnostic pop
```

So we can probably compile with: `-Wno-unknown-pragmas` ?

For **`C4996`**:
- To turn off the warning for a specific line of code:
  `#pragma warning(suppress : 4996)`
- Turn off the warning within a file:
  `#pragma warning(disable : 4996)`
- Turn off the warning in command-line builds:
  `/wd4996`

So we can add: 
`add_compile_options("/wd4996")` to main `CMakeLists.txt` (line ~117)

From [3]:  
- For example, if you received this warning and you want to create an executable file that uses the non-debug, DLL version of the run-time libraries, you could use the following options with the linker:  
  `/NODEFAULTLIB:libcmt.lib /NODEFAULTLIB:libcmtd.lib /NODEFAULTLIB:msvcrtd.lib`
- You can use the linker's `/VERBOSE:LIB` switch to find out which libraries the linker searches.

commit d422a7e6 (2025-03-02): minor: Fix redefinition warning (closes #3216)