merbanan/rtl_433 #3221: minor: new OregonV3 meter

AkaBoing, 2025-03-11:
hi all,
the Oregon Scientific V3 is back with new models
new brands from 2023/2024 are UNNI, Newentor, maybe some more.. 

[image]

[image]

they have a different osv3_sensor_id (second nibble only)
0xf024, 0xf224, 0xf424, 0xf824, 0xfa24
i made a small patch/change to "oregon_scientific.c"

654 	int sensor_id=	(msg[0] << 8) | msg[1];
new:
+655	if ((sensor_id & 0xf0ff) == 0xf024) {
+656		sensor_id = ID_THGR810 ;
+657	} 

now all of them gets decoded by rtl_433

{"time":"2025-03-11 15:03:39","model":"Oregon-THGR810","id":82,"channel":1,"battery_ok":1,"temperature_C":8.3,"humidity":61,"data":"f8240254380016a44c","mod":"ASK","freq":432.93206,"rssi":-0.10952,"snr":37.26347,"noise":-37.373}
{"time":"2025-03-11 15:03:39","model":"Oregon-THGR810","id":82,"channel":1,"battery_ok":1,"temperature_C":8.3,"humidity":61,"data":"fa240254380016a64a","mod":"ASK","freq":432.93206,"rssi":-0.10952,"snr":37.26347,"noise":-37.373}
{"time":"2025-03-11 15:04:10","model":"Oregon-THGR810","id":82,"channel":1,"battery_ok":1,"temperature_C":8.3,"humidity":61,"data":"f0240254380016ac34","mod":"ASK","freq":432.93206,"rssi":-3.49197,"snr":31.66252,"noise":-35.1545}
{"time":"2025-03-11 15:04:10","model":"Oregon-THGR810","id":82,"channel":1,"battery_ok":1,"temperature_C":8.3,"humidity":61,"data":"f2240254380016ae32","mod":"ASK","freq":432.93206,"rssi":-3.49197,"snr":31.66252,"noise":-35.1545}

have fun..

zuckschwerdt, 2025-03-19:
We still need to add these new model numbers. I just don't have the time right now.

ProfBoc75, 2025-03-20:
Hi all, may I propose you test this:

Change the default id to 0xf024 at line 30

```diff
- #define ID_THGR810  0xf824 // This might be ID_THGR81, but what's true is lost in (git) history
+ #define ID_THGR810  0xf024 // This might be ID_THGR81, but what's true is lost in (git) history, #3221 Add new Oregon Scientific v3 models, historical id is 0xf824, new ids are 0xf024, 0xf224, 0xf424, 0xfa24 for this rebrand sensor.
```

Then more simple update at the line 659 :

```diff
-     if ( sensor_id == ID_THGR810 || sensor_id == ID_THGR810a) {
+     if ((sensor_id & 0xf0ff) == ID_THGR810 || sensor_id == ID_THGR810a) { // 3 from 4 nibbles are checked see #3221
```

If ok, I can PR an update, let me know.

AkaBoing, 2025-03-21:
thx ProfBoc75,

 short Check:  positiv result.
A Long Check logging over night is in Work.
But lgtm.

AkaBoing, 2025-03-21:
overnight logging with your code changes:
Newentor
0xf024, 0xf224, 0xf824, 0xfa24
unni
0xf024, 0xf224, 0xf824, 0xfa24
Liorque
0xf024, 0xf224, 0xf824, 0xfa24

the SID 0xf424 was not found.

so OK for me.

time      : 2025-03-21 09:09:39                    Protocol  : 12
model     : Oregon-THGR810                         House Code: 82
Channel   : 1            Battery   : 1             Celsius   : 27.10 C       Humidity  : 16 %          data      : f8240254172061a344
time      : 2025-03-21 09:10:10
Protocol  : 12
model     : Oregon-THGR810
House Code: 82
Channel   : 1
Battery   : 1
Celsius   : 27.10 C
Humidity  : 16 %
data      : f0240254172061ab3c

time      : 2025-03-21 09:10:41
Protocol  : 12
model     : Oregon-THGR810
House Code: 82
Channel   : 1
Battery   : 1
Celsius   : 27.10 C
Humidity  : 16 %
data      : fa240254172061a542
time      : 2025-03-21 09:11:12
Protocol  : 12
model     : Oregon-THGR810
House Code: 82
Channel   : 1
Battery   : 1
Celsius   : 27.10 C
Humidity  : 16 %
data      : f2240254172061ad3a

time      : 2025-03-21 09:11:43
Protocol  : 12
model     : Oregon-THGR810
House Code: 82
Channel   : 1
Battery   : 1
Celsius   : 27.10 C
Humidity  : 16 %
data      : f8240254172061a344

the "data"  output debug shows the SID (first 2 bytes)

ProfBoc75, 2025-03-21:
PR merged, enjoy !

I let you close this issue if all ok for you.

roger-, 2025-06-02:
I have a Hodik sensor that works with this protocol but has some issues. Created issue #3289 if you can take a look.