merbanan/rtl_433 #3225: CSV output prints only headers

HaxtonFale, 2025-03-17:
I'm trying to capture the signals from my weather sensor outside so I tried running the following invocation:
```
$ rtl_433 -v -F csv:weather/sdr.csv -R 19
```
The problem is that it produced the following file:
```
$ cat weather/sdr.csv
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity
```
This feels less than ideal.

I'm using RTL-SDR V4 with Debian 12 (bookworm), rtl_433 version 22.11 (2022-11-19) from the repo.

HaxtonFale, 2025-03-17:
I mentioned the version at the bottom, but it is indeed out of date. I downloaded the most recent release and tried using it.

If I don't specify any parameters, it seems to... not pick anything up, for some reason. It's very close to a digital clock that reads the same sensor, so I'd expect it to be fine even if I don't use the largest antenna I can for the radio. Are there any libraries I might want to upgrade as well?

AkaBoing, 2025-03-17:
hi HaxtonFale,
on bookworm try gqrx (gui version) -(should be in repo)
check if your  RTL-SDR V4 was found and any signal at all at 433.9M

Next try -with -R1 and -R19, > **rtl_433 -f 432.85M -s1024k -R1 -R19 M -level** 
without -F !
-R1 Rubicson and -R19 nexus use the same decoder, so one or both should find the meter.
and give an output.
the -F json variant gives me:
{"time":"2025-03-17 10:53:17","model":"Rubicson-Temperature","id":134,"channel":1,"battery_ok":1,"temperature_C":14.6,"mic":"CRC","mod":"ASK","freq":432.89728,"rssi":-0.544018,"snr":34.61048,"noise":-35.1545}

now test with -F csv

have fun

HaxtonFale, 2025-03-17:
> Each new start will produce a header and append that. The csv file looks like expected. Try to run with `-Y autolevel -M level -M noise` to get an idea what the reception levels are.

Seems like the program settles on noise being at ca. -31 dB if I use all the options you listed. Other than that, if I ask it to be a bit extra verbose (`-vv`) it (maybe) reads _some_ data but most of them fail:
```
[markisol_decode] bits_per_row[0] = 1
[risco_agility_decode] Preamble not found
```
And so on.

> hi HaxtonFale, on bookworm try gqrx (gui version) -(should be in repo) check if your RTL-SDR V4 was found and any signal at all at 433.9M
> 
> Next try -with -R2 and -R19, > **rtl_433 -f 432.85M -s1024k -R2 -R19 -M level** without -F ! -R2 Rubicson and -R19 nexus use the same decoder, so one or both should find the meter. and give an output. the -F json variant gives me: {"time":"2025-03-17 10:53:17","model":"Rubicson-Temperature","id":134,"channel":1,"battery_ok":1,"temperature_C":14.6,"mic":"CRC","mod":"ASK","freq":432.89728,"rssi":-0.544018,"snr":34.61048,"noise":-35.1545}
> 
> now test with -F csv
> 
> have fun

I'll check out gqrx shortly. As for the rest of your reply: see above for picking up any signals; I was able to see the signals on my Windows PC via SDR# (and decode with the rtl_433 plugin that's how I identified the protocol as Nexus) so I'm pretty confident the hardware _can_ see it, and I just need to figure out how to configure it properly or work around obsolete dependencies (like I had to fix in URH).

HaxtonFale, 2025-03-17:
Running gqrx shows me a sea of noise.

AkaBoing, 2025-03-17:
sorry, i didnt know , u already checked it with sdr# on Windows.
the sea of noise (red noise) is normal, but there should be some signal from time to time, even some heavy pilot, jammers or TPMS signals.
for the depencies:
check the "BUILDING.md"

on Debian: 
sudo apt-get install libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential cmake pkg-config
is all u need.

CMake:
git clone https://github.com/merbanan/rtl_433.git

cd rtl_433/
cmake -B build
cmake --build build --target install

or, if your CMake is V 3.13 or older:
( i startet this way years ago with DEB 10 )
cd rtl_433
mkdir build (if not present)
cd build
make
sudo make install.
done, now run rtl_433.

**start rtl_433 -f433.85M -s1024k -M level -g +20 -S unknown**.
-g +20 gives a fix gain of +24dB (see the start messages of rtl_433)

view the cus on https://triq.org/iqs/ to get an overview.

[markisol_decode] bits_per_row[0] = 1
[risco_agility_decode] Preamble not found
or
Possible Oregon Scientific v2.1 message, but sync nibble wasn't found. Raw
codes     : {24}555555
inFactory, nor-tec, FreeTec NC-3982-913 temperature humidity sensor
codes     : {40}f780641362

all these "errors" are normal -vv debug outputs, u will get a lot of them, no problem, but in between
there should be a message like this:

time      : 2025-03-13 22:41:08
model     : Oregon-v1
SID       : 3
Channel   : 2          
Battery   : 1
Temperature: 21.7 C
Integrity : CHECKSUM

or

time      : 2025-03-17 23:41:11
model     : Rubicson-Temperature       
Channel   : 1
Battery   : 1
Temperature: 6.1 C
Integrity : CRC

im shure, the cracks here will help u more..

GN8