merbanan/rtl_433 #3232: Adding custom decoder (Verisure Alarm)

mikkelsiggaard, 2025-03-19:
I'm trying to add some Verisure door sensors to Home Assistant using this conf: https://github.com/merbanan/rtl_433/blob/master/conf/verisure_alarm.conf

@mdeweerd I hope you can verify that the conf-file I'm using is appropriate for this device:
[image]

I also hope you'll be able to elaborate on this quote from the conf-file:
> The payload is most likely encrypted, but you can still use the id and state to figure out if the contact switch is opened or closed.
When I look at the logs there don't seem to be a pattern that I'm able to figure out. The state is always "0" and the id differs a lot. 
I have disabled all but one door sensor, to make sure I only get one device in the logs. 

I have created a file called "rtl_433.conf" under the "/config/rtl_433" folder in Home Assistant. The content of that file is this: 
`output mqtt://192.168.0.190:1883,user=mqtt,pass=3028Tiger3028

output kv
frequency 868M
protocol 40
convert si
verbose 7
sample_rate   250k
pulse_detect classic

decoder {
    name=Verisure Alarm,
    modulation=FSK_PCM,
    short=208,
    long=208,
    reset=4025,
    gap=2500,
    tolerance=10,
    get=@24:{24}:id,
    get=@52:{12}:state,
}

`

<details><summary>Log from rtl_433 Home Assistant Add-on</summary>
<p>

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-03-19 20:03:29
model     : Verisure Alarm
count     : 1
num_rows  : 1
rows      : 
len       : 29
data      : face4548
id        : 4718592
state     : 0
codes     : {29}face4548
[pulse_slicer_pcm] General purpose decoder 'Verisure Alarm'
codes     : {29}face4548
[Verisure Alarm] 
codes     : {25}05392a0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-03-19 20:03:47
model     : Verisure Alarm
count     : 1
num_rows  : 1
rows      : 
len       : 25
data      : 05392a0
id        : 0
state     : 0
codes     : {25}05392a0
[pulse_slicer_pcm] General purpose decoder 'Verisure Alarm'
codes     : {25}05392a0
[Verisure Alarm] 
codes     : {27}029c952
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-03-19 20:04:05
model     : Verisure Alarm
count     : 1
num_rows  : 1
rows      : 
len       : 27
data      : 029c952
id        : 2097152
state     : 0
codes     : {27}029c952
[pulse_slicer_pcm] General purpose decoder 'Verisure Alarm'
codes     : {27}029c952
[Verisure Alarm] 
codes     : {28}029c8a9
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-03-19 20:04:24
model     : Verisure Alarm
count     : 1
num_rows  : 1
rows      : 
len       : 28
data      : 029c8a9
id        : 9437184
state     : 0
codes     : {28}029c8a9
[pulse_slicer_pcm] General purpose decoder 'Verisure Alarm'
codes     : {28}029c8a9
[Verisure Alarm] 
codes     : {26}05392a4

</p>
</details> 

Am I missing something?

merbanan, 2025-03-21:
The device does not look like the one I used to create the config for. You need to treat this as a new device.

mikkelsiggaard, 2025-03-23:
> The device does not look like the one I used to create the config for. You need to treat this as a new device.

Ah okay, I'll try to do that. Any recommended sources on how to do this?

gdt, 2025-07-04:
also you are seeing codes of varying lengths, which to me is a clue that the demod to bits is not right.