merbanan/rtl_433 #3239: Need Help for write my first device

s0k33t, 2025-03-27:
Good morning,
I have a wood stove that I can listen to on the 868.4 MHz frequency. I managed to extract the information from the device using a flex decoder.

rtl_433 -f 868400000 -R0 -g 100 -X 'n=Piazzetta,m=FSK_MC_ZEROBIT,s=10,r=100,preamble={10}fd4,match={26}00563412,invert,get=@0:{32}:id:%x,get=@48:{8}:Hour:%x,get=@40:{8}:Minutes:%x,get=@104:{8}:Fan1:%2x,get=@112:{8}:Fan2:%2x,get=@256:{8}:Temp,get=@224:{16}:Check:%x,get=@68:{4}:Power:[1:On 0:Off]'

time      : 2025-03-27 18:44:09
model     : Piazzetta    count     : 1             num_rows  : 1             rows      :
len       : 305          data      : 0056341201441804010101010101010000000019034d0000000000000564000016010000165d8               id        : 563412
Hour      : 18           Minutes   : 44            Fan1      :  1            Fan2      :  1            Temp      : 22            Check     : 564           Power     : On
codes     : {305}0056341201441804010101010101010000000019034d0000000000000564000016010000165d8
*** Saving signal to file g050_868.4M_1000k.cu8 (56018 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-03-27 18:46:15
model     : Piazzetta    count     : 1             num_rows  : 1             rows      :
len       : 305          data      : 0056341201461804010101010101010000000019034d00000000000005640000160100001a2e8               id        : 563412
Hour      : 18           Minutes   : 46            Fan1      :  1            Fan2      :  1            Temp      : 22            Check     : 564           Power     : On
codes     : {305}0056341201461804010101010101010000000019034d00000000000005640000160100001a2e8
*** Saving signal to file g051_868.4M_1000k.cu8 (55995 samples, 131072 bytes)

I'm trying to write the decoder without luck. I can't get the same length. I've tried with 305 and 315(305 + preamble).
I don't known how to parse to find the preamble and realign if it's necessary.
I also don't know if I'm encoding the device constants correctly... Here's how it is.

r_device const piazzetta = {
        .name        = "Piazzetta",
        .modulation  = FSK_PULSE_MANCHESTER_ZEROBIT,
        .short_width = 10,  // short gap 
//        .long_width  = 10,  // long gap 
//       .gap_limit   = 10,  // some distance above long
        .reset_limit = 100, // a bit longer than packet gap
        .decode_fn   = &piazzeta_decode,
        .disabled    = 0, // disabled and hidden, use 0 if there is a MIC, 1 otherwise
        .fields      = output_fields,
};

Thanks in advance for your help

zuckschwerdt, 2025-03-27:
Your approach looks good. MC is not always stable in the output. Upload a sample as zip here so we can check if there is a (de)sync before the MC. In that case you might need to read PCM data and then use the MC decode in your code.

s0k33t, 2025-03-28:
I hope the sample is good

[g050_868.4M_1000k.zip](https://github.com/user-attachments/files/19503877/g050_868.4M_1000k.zip)

zuckschwerdt, 2025-03-28:
The sample has too much gain (sender and receiver are too close together) but otherwise looks good. You can check with https://triq.org/pdv/
The signal has a warmup pulse and then FSK MC, like you already figured out. It should work well with the decoder definition you gave. What bitbuffers contents do you receive? Finding a preamble is a common operation in decoders, look in other decoder for `bitbuffer_search`.

s0k33t, 2025-03-28:
How can I compare the bitbuffer of my decoder and the flex?

zuckschwerdt, 2025-03-28:
That's the same but shifted 10 bits. See [this BitBench](https://triq.org/bitbench/#c=%7B314%7Dfd40158d048049050140404040404040400000000640d0c000000000000159000013c04000290b00&c=%7B10%7D0%20%7B304%7D00563412012414050101010101010100000000190343000000000000056400004f010000a42c&f=8h&z=1&cw=4).

s0k33t, 2025-03-28:
Okay. I think I understand. Find the position of the preamble, add up its length, and shift it.

I'm going to try.

Thanks.

closed 2025-07-04 by gdt