merbanan/rtl_433 #3243: Support for Auriol Weather Station With Multisensor HG11911

shrx, 2025-03-29:
This is a weather station sold by Lidl. The external unit is providing temperature, humidity, rainfall, wind speed, wind direction (potentially also battery info).

I recorded some IQ samples and analyzed them in https://triq.org/pdv/

[image]

[cu8_samples.zip](https://github.com/user-attachments/files/19517162/cu8_samples.zip)

From there it looks like these are the parameters I should be using (I could be wrong though, since it automatically detected OOK demodulation but that produced nothing usable): 

`rtl_433 -f 433.887M -R 0 -X "n=auriol,m=FSK_PWM,s=88,l=176,g=1400,r=20000,bits>=80,repeats=3"`

With this command I obtained the following readouts, and mapped them to the info displayed on the weather station's panel:

readout | wind speed | wind direction | rain daily | temperature | humidity
-- | -- | -- | -- | -- | --
7ffffb7feb63e7cffbf6a | 0 | 271 | 0.4 | 25.8 | 51
7ffffb7feb77e7cffbf48 | 0 | 271 | 0.4 | 25.7 | 52
7ffffb7feb77f3e7fdf8a | 0 | 271 | 0.4 | 25.6 | 53
7ffffb7feb6fe7cffbf04 | 0 | 271 | 0.4 | 25.5 | 53
7ffffb7feb4fe7cffbf10 | 0 | 271 | 0.4 | 25.5 | 53
7ffffb7feb53e7cffbf38 | 0 | 271 | 0.4 | 25.3 | 54
7ffffb7feb63e7cffbf650 | 0 | 271 | 0.4 | 25.2 | 54
7ffffb7feb73e7cffbf26 | 0 | 271 | 0.4 | 25.1 | 55
7ffffb7feb73e7cffbf58 | 0 | 271 | 0.4 | 25.1 | 55
7ffffb7feb47cf9ff7e88 | 0 | 271 | 0.4 | 25.0 | 55
7ffffb7feb4e4fe7fdf8a | 0 | 357 | 0.4 | 24.6 | 55
7ffffb7feb4e4ee7fdf88 | 0 | 357 | 0.4 | 24.6 | 54
7ffffb7feb2e4dcffbf12 | 0 | 196 | 0.4 | 24.5 | 57
7ffffb7feb0d53f3fefd38 | 9.8 | 166 | 0.4 | 24.3 | 57
7ffffb7feb0db373fefcac | 11.2 | 313 | 0.4 | 24.3 | 58
7ffffb7feb0db3fcffbf0 | 12.0 | 284 | 0.4 | 24.3 | 58
7ffffb7feb2c2fe7fdfb3 | 1.2 | 203 | 2.1 | 24.5 | 57
7ffffb7feb2febf3fefd4 | 0 | 73 | 2.1 | 24.5 | 57
7ffffb7feb2febe7fdf9f0 | 0 | 73 | 2.3 | 24.5 | 57
7ffffb7feb2f47cffbf3a | 0 | 24 | 2.5 | 24.5 | 57

Can anyone help decoding the readout and mapping it to the sensor values?

ProfBoc75, 2025-03-29:
@shrx : really good starting point, but your device is already known as Emax-EM3551H, protocol 215 (emax.c decoder)
Your 2 samples are properly decoded

    rtl_433 *.cu8 -M protocol

[image]

What is your rtl_433 version ? You may have to just update rtl_433, the last distribution version is recent from February 2025 and already decoding this sensor.

shrx, 2025-03-29:
Oh that is great news. I installed rtl_433 with apt, it looks like it's a very old version 21.12 from 2021.

shrx, 2025-03-29:
Confirmed working with latest binary from github version 25.02 with `rtl_433 -M 215`.

closed 2025-03-29 by shrx

ProfBoc75, 2025-03-29:
@shrx : PR merged, thanks for your contribution !