merbanan/rtl_433 #3245: CMake 4.0 requires the minimum CMake version be bumped to at least 3.5

kode54, 2025-03-30:
Arch Linux has pushed CMake 4.0 to the stable extra repository, which obsoletes compatibility with CMake older than 3.5:

```
CMake Error at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```

zuckschwerdt, 2025-03-30:
Thanks! We need to review dropping support for old CMake versions (pre 3). For now we know and declare compatibility with at least version 3.10.

closed 2025-03-30 by zuckschwerdt

gdt, 2025-03-30:
ok to require 3.10, but we should be careful.   It seems unstable for a packaging system to declare 4.0 stable this week.

commit 42ac6414 (2025-03-30): build: Update for compat with newer CMake (closes #3245)