merbanan/rtl_433 #3248: No pulses or signal found in rtl_433 while it is clearly visible in sdrpp waterfall

Quentis, 2025-04-04:
I would like to catch and decode 433.92MHz PWM OOK signal from my garden gate opener.
It uses the Microchip KeeLoq PWM protocol according to [HCS300](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/21137G.pdf).
In short the protocol is the following:
- 23 x TE (50%) duty cycle preamble (12 pulse, 0->1->0)
- 10 x TE header (no pulse, idle)
- 65 bit PWM data
    - 2 x TE On & 1 x TE Off => 0
    - 1 x TE On & 2 x TE Off => 1

SDR device: **RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"**

Disclaimer: this is my first SDR project and I did my best to cover the available documentation but I might have made silly mistakes.

My approach was the following:
- I used the `sdrpp` program to visualize the spectrum at 433.92MHz and figure out the correct sample rate and gain settings
    - Noise level doesn't increase significantly with 30dB gain setting
    - The center frequency of gate opener was 433919155 Hz --> I don't know if this small difference matters.
- I used rtl_433 with gazillion parameter combinations to catch any pulses/frames, to name a few:
    - `rtl_433 -A`
    - `rtl_433 -f 433919155 -vvv -Y level=-35 -M level -M noise -s 2048k -g 30 -A -R 0`
        - I used 30dB gain because it looked promising
        - I set detection level to -35dB because the program reported -40dB noise
        - I tried a lot of combination with and without level settings but nothing was detected
    -  `rtl_433 -f 433919155 -vvv -Y level=-30 -M level -M noise -s 2048k -g 30 -T 5 -w file_433.919155M_2048k_g30.cu8`
        - I wanted to record the samples and visualize the spectrum from the `rtl_433` perspective
        - I used https://triq.org/iqs/ to visualize it but I had nothing except for maybe the local oscillator at center frequency
        - The result was almost like noise and I couldn't spot any signals however those were clearly visible in `sdrpp`
- I used `Audacity` to check the baseband signal
    - I created a baseband wav recording in `sdrpp` and I loaded it in Audacity
    - The KeeLoq signal was kind of recognizable but the pulses had a slow sinusoidal envelope --> I was able to increase the period of this sinusoidal envelop by tuning the frequency to 433919155 (I used the default originally)
        - Nevertheless, the KeeLoq protocol and timings were clearly visible
- I tried to decode the signal with general purpose decoder assuming something was wrong with https://triq.org/iqs/ visualizer
  (I was quite desperate at this point)
    -  rtl_433 -vvv -f 433.919155M -Y level=-30 -s 2048k -g 30 -t -X "n=KeeLoq,m=OOK_PWM,s=422,l=844,r=10000,t=100" -R 0 -A
    - Timing parameters were passed based on Audacity measurement and checking the datasheet
    - Note: it is configurable in the IC what should be the timing and I calculated the time based on sample count in Audacity (where I had to set the sampling rate beforehand to measure correctly)

I did my best to figure this problem out in my own without luck.
Could you help me figure out what could be the problem please?

I have attached a zip file with pictures and cu8 recordings:
[rtlsdr_v4_433919155hz_32p8db_keeloq.zip](https://github.com/user-attachments/files/19598650/rtlsdr_v4_433919155hz_32p8db_keeloq.zip)

**System and version:**
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR
OS Name    : Microsoft Windows 11 Pro
OS Version : 10.0.26100 N/A Build 26100

ProfBoc75, 2025-04-04:
@Quentis you can also specify less options 

    rtl_433 -Y autolevel -M level -M noise -vvv

Default frequency is 433.92M, and for ook it's better to not specify the exact frequency to remove dc value. Reduce sample rate to 1024k or keep default 250k as the pulses are long with 422 / 844 us

Quentis, 2025-04-06:
@ProfBoc75 Thanks a lot for the tip, it works now properly.
I made a double mistake:
- I used `rtl_433-win-x64-25.02` instead of `rtl_433-win-msvc-x64-25.02`
- I haven't updated the `rtlsdr.dll` library
I thought that if `sdrpp` works then there shouldn't be any driver like issues.

closed 2025-04-06 by Quentis