merbanan/rtl_433 #3250: 433mhz buttons continue sending up to several seconds after releasing them - sw config issue or hardware issue?

booherbg, 2025-04-07:
Hi! I have rtl433 working great but am a novice. I have little 433mhz buttons, and they seem to keep sending data if I hold down on them for up to several seconds after releasing the button. For example if I hold for 2 seconds, it may spam messages for 7 seconds total.

I can't tell if this is a hardware "feature" or if this is something in the software config that's queuing the messages and dumping them as fast as it can. 

In reality I have several buttons and they interfere with each other when pressed at the same time so I am curious if this is a software issue (where I can ask rtl433 to ignore duplicate messages over a period of time) or if this is actually a hardware issue (where the device appears to be asleep but is still sending messages for many seconds).

(in addition my buttons seem to randomly send garbage even when not pressed which seems weird to me - is this a glitch on the rtl433 side? these are dumb buttons that just have a battery and radio chip).

Would love any help :) I wasn't sure if there is a discord channel or another place that's better.

Button Config - admittedly this is generated with help from an LLM but seemingly works great
```
# Custom configuration for 433MHz buttons
decoder {
    name=MyButton,
    modulation=OOK_PWM,
    short=364,
    long=1084,
    reset=10616,
    gap=1032,
    tolerance=288,
    bits>=24,
    get=@0:{24}:code,
    unique
}
```

MQTT Script for forwarding messages, while filtering out my local weather data (I couldn't get rtl433 to publish otherwise, but I'll try again now that this script is working):

``` bash
#!/bin/bash

# Configuration
MQTT_HOST="localhost"
MQTT_PORT="1883"
MQTT_TOPIC="rtl_433/events"
CONFIG_FILE="button_config.conf"

echo "Starting RTL_433 to MQTT bridge..."
echo "- MQTT Topic: $MQTT_TOPIC"
echo "- Configuration: $CONFIG_FILE"

# Run rtl_433 and pipe to MQTT
rtl_433 -c $CONFIG_FILE -F json | while read -r line; do
    # Always show what was detected
    echo "Detected: $line"
    
    # Check if the line contains JSON data
    if [[ $line == *"{"* ]]; then
        # Filter for buttons if desired
        if [[ $line == *"\"model\" : \"MyButton\""* ]]; then
            # Filter out noise code 16777215
            if [[ $line == *"\"code\" : 16777215"* ]]; then
                echo "Filtered out noise signal with code 16777215"
            else
                echo "Button press detected: $(date)"
                echo "$line" | mosquitto_pub -h $MQTT_HOST -p $MQTT_PORT -t "$MQTT_TOPIC" -l
                echo "Published to MQTT"
            fi
        else
            echo "Non-button device detected (not publishing to MQTT)"
        fi
        echo "----------------------------"
    fi
done
```

booherbg, 2025-04-08:
Update: Turns out this is almost certainly a delay in rtl433's processing into MQTT (likely because of my script). Will try to make this faster by using native rtl433 and see if de-duping will help.

zuckschwerdt, 2025-04-08:
Simple buttons (mostly EV1527 like) will repeat the data packets as long as the button is held -- but that is a very long transmission in RF terms. There should be no noticeable delays in rtl_433, just the time to fully receive the RF data frame, at the default 250k sample rate that's 500 ms, at 1024k it would be 125 ms.

Look into the relays and filters at https://github.com/merbanan/rtl_433/tree/master/examples if you want custom MQTT publishing.