merbanan/rtl_433 #3252: Acurite 592TXR RSSI (Protocol 40)

LeeThompson, 2025-04-11:
I'm using RTL_433 to drive a database of sensor data from rooms and appliances and I've noticed the RSSI values don't seem to be where they should.

Having said that, I am not an expert by any means with this stuff and I could simply not be understanding this but I have researched a bit and the RSSI numbers don't seem to be correct to me so I felt I should at least bring this to the attention of people that know what they're doing :)

The sensor are all Acurite 592TXR.

4 sensors are on the floor below, one just in the kitchen but another 2 in the refrigerator (one in the door and the other nearer to the center); and one in a chest freezer.

All of the sensors are the same model.   The ones in the refrigerator and freezer are using Energizer Lithium batteries.   The remainder use high quality Li-On rechargeables.

Very rarely is there a RSSI worse than -6.    The SNR levels seem kinda okay, I haven't really researched the NOISE stuff yet and have no opinion on it.

Here some some sample readings.

I log all of the data into Mosquitto (temporary use) and SQL and so if there is a problem and you guys need data, I have a lot of it.

Anyway here is a snapshot.

| Sensor Purpose | RSSI | SNR | NOISE | Notes |
| ---------------- | ----- | ----- | ------ | ------ |
| Air Conditioner | -1.70621 | 20.84077 | -22.547 | (Computer Room) Same Room as Receiver for RTL_433 |
| Bedroom | -6.2887 | 17.5759 | -23.8646 | (Bedroom) Same Floor as Receiver on opposite side of the house. |
| Freezer | -0.227737 | 22.93781 | -23.1656 | (Garage) Chest freezer, below room as Receiver |
| Kitchen | -0.238632 | 23.92901 | -24.1676 | (Kitchen) Floor below Receiver and roughly a room farther away |
| Refrigerator (Core) | -0.210342 | 23.3136 | -23.5239 | (Kitchen) Floor below Receiver and roughly a room farther away, inside a standard sized refrigerator |
| Refrigerator (Door) | -13.6206 | 9.72232 | -23.3429 | (Kitchen) Same as Refrigerator (Core) but sensor is on one of the door shelves |

To me what stands out are:
* The RSSI and SNR between the two refrigerator sensors, if anything, are reversed.  The signal should be stronger inside the door since it has less stuff to travel through.

zuckschwerdt, 2025-04-11:
SNR is just the difference between noise and RSSI. Noise is the estimated reception level when there is no signal, we don't measure the noise when there is a signal, so these values are at slightly different points in time. The automatic gain (in the SDR hardware) will constantly adjust the gain stages to optimize reception. You will only see low RSSI when the gain is maxed out or when a strong signal has just before reduced the gain. And when you see a RSSI close to -0 dB or above then the gain is too high. Noise is also not linear with gain.

In short, all values given are actually dBr relative to the current gain settings. (These settings are internal and can't be known or read for e.g. the RTLSDR).

If you want true differential readings between different senders then try setting a fixed gain with e.g. `-g 10`, optimize for strong levels to come in at -1 dB (this might loose the weak senders), or use a higher gain to just read the weakest senders (this might loose too strong senders). Use `-Y autolevel -M level -M noise` to better read weak signals and also show running stats about noise (even if there is no signal currently).

You could even control a running rtl_433 via the HTTP API and cycle through different gains to cover a wider range -- we should maybe add the current selected gain to the output lines too? Could be interesting.

LeeThompson, 2025-04-11:
@zuckschwerdt Thanks for your response, that makes more sense now.

closed 2025-07-04 by gdt