merbanan/rtl_433 #3257: Bad formatting of STDOUT when using pipeline v25.02-13

xeroski, 2025-04-20:
Hi!

I use "grep" filtering with STDOUT from rtl_433 (pipeline). Everything was OK until I updated rtl_433 today. Sometimes telegrams are OK and sometimes they split into multiple lines (Wireless M-Bus). Each parameter is on a separate line.
I have this problem on two machines: Kali Linux and Raspberry OS Bookworm
Can someone confirm this?
Simple check:

`$ rtl_433 -f 868.95M -R 104 | grep ''`

xeroski, 2025-04-20:
[image]

AkaBoing, 2025-04-20:
confirmed, 
not only M-bus, but nearly all devices show this behavior, dunno why.. but there is a expert here..
i take JSON output, but even then it occurs like that.
but not a real issue for me.

zuckschwerdt, 2025-04-21:
Wide lines are only enabled if the TTY can report column size (ioctl TIOCGWINSZ), that's not the case for a pipe.

You really should use JSON (actually [jsonl](https://jsonlines.org/)), filter that and pretty print. E.g. `rtl_433 -F json | grep --line-buffered 'LaCrosse' | jq --unbuffered`

closed 2025-04-21 by zuckschwerdt