merbanan/rtl_433 #3258: Ecowitt WH51 868Mhz decode only within few meters

RodEzIO, 2025-04-25:
Hello, I am running RTL 433 on Windows with a 868 MHz magnet antenna on a metal base and a DVB-T USB stick. The sensor is an Ecowitt WH51 at 868 MHz. Attached is a picture of the received message.

It seems that I have some reception issues:

1.     It simply doesn't work without the -Y classic option.
2.     If I move the sensor more than 10 meters away, it stops detecting packets.
3.     I expected that this frequency band should have a range of at least 100 meters, isn’t that right?

Do you have any suggestions for the configuration or setup to improve reception? Do I need to buy a Nooelec, and would it make a difference compared to a standard DVB-T USB stick?

Thanks!

[image]

zuckschwerdt, 2025-04-25:
Hitting a frequency dead on may reduce reception, try `-f 868.3M`.
`minmax` is the newer demod, but for 433.92M and sometimes 868M `classic` performs better.
More than 10 m is expected, but 100m is a best case with no walls in the way.
Maybe capture samples and inspect them visually to see what's wrong. https://triq.org/rtl_433/ANALYZE.html

ProfBoc75, 2025-04-25:
@RodEzIO : if you look to a better rtl usb stick, look at R828D or R860 tuners.
They are both replacing older references R820T / R820T2 no longer manufactured.

Your usb stick is not so bad, [R820T vs E4000 review](https://www.passion-radio.com/index.php?controller=attachment&id_attachment=138).
You can improve the signal reception by :
- taking care of the antenna to put as far as possible from PC / Screen / Power Supply / Electronics. 
- Add a USB cable extension between the PC and the USB stick. The E4000 is very sensitive to electronic noise as explained in the review.
- During the review, "HB9AJG" wrapped it in aluminum foil connected to the PC ground (USB ground if you have the cable extension), this will reduce the noise, the "birdies" and help the signal reception.

RodEzIO, 2025-04-26:
I have received the Nooelec, and now the reception is acceptable. I can even receive other transmitters around here. However, -Y classic is still important.

RodEzIO, 2025-04-27:
After moving the device to its final position, it doesn't work anymore. It seems that there is only one antenna position where it works — in other words, there isn't enough margin for reception.
The system reception is very unreliable.
Do you have any suggestions? Why is the pulse detector so sensitive to change?
Here is the current configuration:
```
verbose 5
frequency 868.2M
pulse_detect classic

report_meta level
report_meta noise
report_meta stats
report_meta protocol

# MQTT output settings 
# mqtt://host:port,events=TOPIC
output mqtt://192.168.8.22:1883,events=sensor/rtl_433,user=mqtt-user,pass=PBYcHevB6h
output kv

protocol 142
```

this is the last working reception

[image]

this is a sample of one working reception [g213_868.2M_1000k_cucina.cu8.zip](https://github.com/user-attachments/files/19930514/g213_868.2M_1000k_cucina.cu8.zip)

zuckschwerdt, 2025-04-27:
> RSSI : -11.8

The default required minimum RSSI is -12 dB. If you want a wider range you need set that level or use `-Y autolevel`, as config file option:
```
pulse_detect autolevel
```
(note: that's in addition to `pulse_detect classic`)

ProfBoc75, 2025-04-27:
@RodEzIO : It looks like your antenna is broken or not connected ?
Are you using the same with Nooelec stick than previous usb stick ?

Simple test:

    rtl_433 -M level -M noise -Y autolevel -Y classic -f 868M

1. Try this command, with antenna, until you have the "current noise level -xx.x dB, estimated noise -yy.y dB like you have in your screenshot. (With my R828D tuner, the noise floor is around -20 dB, receiving sensors RSSI around -0.1 dB to -7 dB)
2. Let the command running, and remove the antenna, and see if you have the same values, or any changes ? If not, this means that the antenna is badly soldered / broken / not connected (bad contact) / not working as expected. (with my stick without antenna, noise floor decreased to -35 dB, and I received very few sensor at RSSI -30 db)

RodEzIO, 2025-04-29:
This antenna is brand new, and it's the long black one included in the Nooelec kit.

Below are the test results with and without the antenna connected — can we conclude anything from this?

[image]

gdt, 2025-07-04:
As a final, comment, the level of noise and interference can also matter.   Looks like this discussion has concluded.  Feel free to open a discussion to ask for further help if still needed.