merbanan/rtl_433 #3259: Alteco WS5500

ExquisiteMetropolis, 2025-04-26:
I've been attempting to get my [Alteco WS5500](https://alectohome.com/collections/weerstations/products/alecto-ws-5500-professioneel-wifi-weerstation-met-app-zwart) to work with the RTL_433 dongle (as Home Assistant Addon). I only have the outside station, not the inside unit!

Nothing is flowing to MQTT just yet. Maybe a fresh set of eyes can help me out.
The steps I have done so far:

template config file (in /usr/share/hassio/homeassistant/rtl_433):

```
device :<s/n>

#MQTT Server for datafeed to Home Assistant
output mqtt://<IP>:1883,retain=1

#Log options
output log
report_meta time:iso:usec:tz
report_meta stats
report_meta protocol
verbose 7

#Frequency of Alteco WS-5500 is 868Mhz, Protocol t.b.d.
frequency 868M

#Tuning
gain 0
pulse_detect auto
pulse_detect level=0
pulse_detect autolevel
```

This results in the following (relevant?) info in the log:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433_868.conf...
[rtl_433_868] rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[rtl_433_868] MQTT: Publishing MQTT data to <IP> port 1883
[rtl_433_868] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433_868] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433_868] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433_868] 
[rtl_433_868] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433_868] 
[rtl_433_868] Protocols: Registered 238 out of 275 device decoding protocols
[rtl_433_868] Input: The internals of input handling changed, read about and report problems on PR #1978
[rtl_433_868] SDR: Found 2 device(s)
[rtl_433_868] SDR: trying device 1: Realtek, RTL2838UHIDIR, SN: <s/n>
[rtl_433_868] Found Fitipower FC0012 tuner
[rtl_433_868] SDR: Using device 1: Realtek, RTL2838UHIDIR, SN: <s/n>, "Generic RTL2832U OEM"
[rtl_433_868] Exact sample rate is: 1000000.026491 Hz
[rtl_433_868] SDR: Sample rate set to 1000000 S/s.
[rtl_433_868] Input: Bit detection level set to 0.0 (Auto).
[rtl_433_868] SDR: Tuner gain set to Auto.
[rtl_433_868] Input: Reading samples in async mode...
[rtl_433_868] SDR: rtlsdr_set_center_freq 868000000 = 0
[rtl_433_868] SDR: Tuned to 868.000MHz.
[rtl_433_868] acquire_thread: acquire_thread enter...
[rtl_433_868] MQTT: MQTT Connected...
[rtl_433_868] MQTT: MQTT Connection established.
[rtl_433_868] Baseband: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[rtl_433_868] calibeur_rf104_decode: DECODE_FAIL_SANITY data all 0x00 or 0xFF

#BUNCH OF PROTOCOLS BEING REGISTERED (REMOVED FOR READABILITY)

#BUNCH OF PROTOCOLS NOT BEING FOUND (REMOVED FOR READABILITY)

[rtl_433_868] fineoffset_WH25_callback: short package. Header index: 213 codes [{189}fffffffffffffffffffffffffffffffffffffffffffffff8]
[rtl_433_868] fineoffset_WH51_callback: short package. Header index: 213 codes [{189}fffffffffffffffffffffffffffffffffffffffffffffff8]
[rtl_433_868] fineoffset_wh1080_callback: short package. Header index: 213 codes [{189}fffffffffffffffffffffffffffffffffffffffffffffff8]
[rtl_433_868] honeywell_cm921_decode:  codes {422}bffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffc
[rtl_433_868] honeywell_cm921_decode: preamble_start=427 start=457 len=-30
```
The last 5 lines, show up very regularly. 
The Honeywell Thermostat is not something I own, so I will ignore this eventually in the configuration.

I assume, my WS5500 sending data as WH25, WH51 & WH1080?
Which links to protocols 78, 142 & 155 when I check the log for the protocols that were registered.
However it's not readable data for MQTT, as no entry is showing up there. 

I also have captured a .cu8 file, using "signal_grabber unknown" however that file isn't readable in Notepad++.
I'm not sure what tooling I should use to open that one. Attached for reference. (rename to .cu8 after download).

[g851_868M_1000k.cu8.txt](https://github.com/user-attachments/files/19921023/g851_868M_1000k.cu8.txt)

I did play with it before, but left the unit in storage afterwards.
Ref: https://github.com/merbanan/rtl_433/issues/3090#issuecomment-2474571683
That did result in some data, however different than from what I am seeing right now. 

I'd appreciate a helping hand in this case on what I am missing or need to (re)configure to get it working?

zuckschwerdt, 2025-04-27:
You can visualize cu8 in a spectrogram like https://triq.org/pdv/ but your sample g851 seems empty.

 I don't think the WS5500 is supported by rtl_433 yet.

The decode fails you see are false positives, they don't have any data (fff... is not meaningful data).

Try to get a good cu8 sample, then we can check timings and start to read data. https://triq.org/rtl_433/ANALYZE.html

ExquisiteMetropolis, 2025-04-28:
@zuckschwerdt thanks! :)

I've added "signal_grabber all" to my config file (same as above)
For some reason it's not generating .cu8 files after running for over an hour.
Am I missing something?

(Totally new to this). 
Weird thing is, I did pickup some data in the previous attempt: https://github.com/merbanan/rtl_433/issues/3090#issuecomment-2474571683 
As far as I can see, no one in the immediate area has a weatherstation. So where did that came frome?

zuckschwerdt, 2025-04-28:
Ah ok, WS-5500 is supported by the Fineoffset-WH24 protocol? Then no need for samples. But we should add a note to the Fineoffset-WH24 decoder.

The cu8 files will be in the starting directory of the rtl_433 process (if that path is writeable). If you run rtl_433 with some startup script like systemd then the path will likely be the running users home dir.

You can add `-F rtl_tcp` and then use some SDR UI to connect to that and watch the waterfall for signals.

ExquisiteMetropolis, 2025-04-29:
@zuckschwerdt I'm not sure anymore if it is supported already. ;) I was hoping it would be. 
Otherwise I quess I would be flooding my logs / MQTT server already with data as it's happily outside gathering environmental data.
The red led flashes as expected on the unit, indicating proper operation.

I've now installed RTL433 on a clean raspberry pi4, to be able to run commands directly to rtl_433 (instead of the HA Addon).

I ran the following command:
`sudo rtl_433 -f 868M -vv -Y autolevel -s 250k -S all -w file_868M.cu8 -T 60`

Which resulted in 1 large CU8 file and a few smaller ones. See .ZIP file.
I'm not sure if this is an expected result? 

[g002_868M_250k.zip](https://github.com/user-attachments/files/19966495/g002_868M_250k.zip)

ExquisiteMetropolis, 2025-04-30:
Thanks again for the support! :👍 
Please find the results of a 2 min run via the command you provided in the ZIP. 

[g002_868M_1000k.zip](https://github.com/user-attachments/files/19976446/g002_868M_1000k.zip)

zuckschwerdt, 2025-04-30:
There are 3 different senders in there (e.g. g002, g004, g006), two are clipping and can't be read and the third is GFSK which is difficult anyway.
Try to load samples in https://triq.org/pdv3/ -- use a Flat-Top window and lower the FFT bins, you can see the data. But rtl_433 will need a cleaner signal, no clipping (the orange and red parts in the waveform display).
Can you find a way to isolate the WS5500 signal?

[image]

[image]

[image]

ExquisiteMetropolis, 2025-05-01:
I think I found the issue. :)

As mentioned I played with it before https://github.com/merbanan/rtl_433/issues/3090#issuecomment-2474571683 and some data was obtained.
```

[rtl_433_868] time      : 2024-11-13T20:31:03.970578+0100
[rtl_433_868] model     : Fineoffset-WH24
[rtl_433_868] ID        : 164
[rtl_433_868] Battery   : 1
[rtl_433_868] Temperature: 15.7 C
[rtl_433_868] Humidity  : 67 %
[rtl_433_868] Wind direction: 59
[rtl_433_868] Wind speed: 1.1 m/s
[rtl_433_868] Gust speed: 2.2 m/s
[rtl_433_868] Rainfall  : 0.0 mm
[rtl_433_868] UV        : 1
[rtl_433_868] UVI       : 0
[rtl_433_868] Light     : 0.0 lux
[rtl_433_868] Integrity : CRC
```

Based on the WH24 identification I googled some more. 
The WS-5500 also comes up as a WH2900:
https://www.foshk.com/Wifi_Weather_Station/WH2900.html

Searching some more on that, I found this: https://groups.google.com/g/rtl_433/c/ZXslVVxNRFo
where someone references that he changed the -F to 868.2M 
So I did that, and when running this command:

`rtl_433 -f 868.2M -v -Y autolevel -F rtl_tcp`

This showed up (after a minute or 2):

[image]

So I guess it's fully supported afterall, just a minor frequency change needed.
Maybe something to add to the documentation?

For the config (of the HomeAssistant Addon) I now have (connected the dongle to that hardware again):
```
device :<s/n>

#MQTT Server for datafeed to Home Assistant
output mqtt://<IP>:1883,retain=1

#Log options
output log

#Frequency of Alteco WS-5500 is 868.2Mhz
frequency 868.2M

#Tuning
pulse_detect autolevel
pulse_detect level=0
gain 0

```
But... for some reason it's not (yet) capturing the data as was shown via the CLI command.
I'm not sure what's the difference here.

Logs show, and that's it:
```
Starting rtl_433 with rtl_433_868.conf...
[rtl_433_868] rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[rtl_433_868] 
[rtl_433_868] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433_868] 
[rtl_433_868] MQTT: Publishing MQTT data to <IP> port 1883
[rtl_433_868] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433_868] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433_868] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433_868] Found Fitipower FC0012 tuner
[rtl_433_868] Exact sample rate is: 1000000.026491 Hz
[rtl_433_868] SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: <S/N>, "Generic RTL2832U OEM"
```
Anything else you would advise adding/removing?

zuckschwerdt, 2025-05-01:
CLI and config look the same to me.

I would drop both
```
pulse_detect level=0
gain 0
```
as those are defaults and level is overridden by autolevel.

ExquisiteMetropolis, 2025-05-01:
removed them. No change.

More troubleshooting. :)
As it's clearly working on the Pi, with the dongle & antenna I have, there must be a difference somewhere.

Preferably I have it running in Home Assistant via the RTL433 addon.

I have Home Assistant running on Proxmox, in a VM with HAOS on it.
The USB stick is (ideally) mounted to the Home Assistant VM.

Trying to emulate that with a seperate VM, running debian & docker with the dongle passing through Proxmox & Docker to the container. Works fine. 

```
 docker create \
  --name rtl433 \
  --restart=unless-stopped \
  -v /etc/timezone:/etc/timezone:ro \
  -v /etc/localtime:/etc/localtime:ro \
  --device /dev/bus/usb/002/003 \
hertzg/rtl_433 "-Mlog" "-f868.2M"
```

Returns:
```
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR with TLS
New defaults active, use "-Y classic -s 250k" if you need the old defaults
Detached kernel driver
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000787, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:30:21
model     : Honeywell-CM921                        Device IDs: 967f96 94de75 unknown   : 31e0          Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:30:29
model     : Fineoffset-WH24                        ID        : 164
Battery   : 1            Temperature: 23.6 C       Humidity  : 44 %          Wind direction: 101       Wind speed: 1.5 m/s       Gust speed: 2.2 m/s       Rainfall  : 0.0 mm        UV        : 67            UVI       : 0             Light     : 8509.0 lux    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:30:45
model     : Fineoffset-WH24                        ID        : 164
Battery   : 1            Temperature: 23.6 C       Humidity  : 45 %          Wind direction: 106       Wind speed: 1.5 m/s       Gust speed: 2.2 m/s       Rainfall  : 0.0 mm        UV        : 67            UVI       : 0             Light     : 8509.0 lux    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:31:01
model     : Fineoffset-WH24                        ID        : 164
Battery   : 1            Temperature: 23.6 C       Humidity  : 45 %          Wind direction: 104       Wind speed: 2.1 m/s       Gust speed: 2.2 m/s       Rainfall  : 0.0 mm        UV        : 66            UVI       : 0             Light     : 8354.0 lux    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:32:21
model     : Fineoffset-WH24                        ID        : 164
Battery   : 1            Temperature: 23.6 C       Humidity  : 47 %          Wind direction: 174       Wind speed: 1.3 m/s       Gust speed: 2.2 m/s       Rainfall  : 0.0 mm        UV        : 61            UVI       : 0             Light     : 7626.0 lux    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-05-01 19:33:25
model     : Fineoffset-WH24                        ID        : 164
Battery   : 1            Temperature: 23.5 C       Humidity  : 45 %          Wind direction: 120       Wind speed: 3.5 m/s       Gust speed: 4.5 m/s       Rainfall  : 0.0 mm        UV        : 63            UVI       : 0             Light     : 7967.0 lux    Integrity : CRC
```

Using the same minimalistic config in the HA Addon, does not return anything:

```
frequency 868.2M
output log
```

```
Starting rtl_433 with rtl_433_868.conf...
[rtl_433_868] rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[rtl_433_868] 
[rtl_433_868] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433_868] 
[rtl_433_868] Found Fitipower FC0012 tuner
[rtl_433_868] Exact sample rate is: 1000000.026491 Hz
[rtl_433_868] SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000787, "Generic RTL2832U OEM"
```

So I guess, it must be an issue with the HA Addon. And I should open a topic there to further troubleshoot? 

One question remains, what protocol should I add eventually?
As there is also a Honeywell device showing up, that I don't own or care for to have in my logs/mqtt.

zuckschwerdt, 2025-05-01:
I see. `output log` won't show any devices, the original config also had `output mqtt...`. If you want console output you need `output kv`. Is that the missing piece?

ExquisiteMetropolis, 2025-05-02:
@zuckschwerdt you're my hero! Thank you! 🥇 

For reference, if anyone else is looking for this in the future:

Config in the HA RTL_433 addon:
```
#RTL-SDR Dongle (only if multiple are present).
device :00000787

#Frequency of Alteco WS-5500 is 868.2Mhz, protocol 78
frequency 868.2M
protocol 78

#MQTT Server for datafeed to Home Assistant
output mqtt://<IP>:1883,user=<USER>,pass=<PASSWORD>,retain=1

#Log options
output log
output kv

#Tuning
pulse_detect autolevel
```

And in the rtl_433 MQTT Auto Discovery AddOn:
```

mqtt_host: <IP>
mqtt_port: 1883
mqtt_user: <USER>
mqtt_password: <PASSWORD>
mqtt_retain: true
rtl_topic: rtl_433/+/events
device_topic_suffix: ""
discovery_prefix: homeassistant
discovery_interval: 600
force_update: false
log_level: default
```

Results in :

[image]

closed 2025-05-02 by ExquisiteMetropolis