merbanan/rtl_433 #3260: WH 1080 error with new version

kocicky, 2025-04-28:
I have issue with last version of the Addon. I use it for weather station WH1080 and it worked.
Now with the same configuration it usually stop with:
`/run.sh: line 129:   131 Killed                  rtl_433 -c "$live" > >(sed -u "s/^/[$tag] /") 2> >(sed -u "s/^/[$tag] /" 1>&2)`

My configuration is
```
sample_rate 1024000
frequency 868290000
frequency 868300000
frequency 868310000
hop_interval 360
protocol 32
report_meta time:iso:usec:tz
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
output kv
convert si
```

Hop between frequnces is because the frequency of the station is unstable and I'm trying to figure out the right hop interval. But error is even without hop_interval on one frequency only.
Sometimes, when i tried verbose output
verbose 1
Adon displayed information:
```
[wh1080] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[wh1080] 
[wh1080] time      : 2025-04-28 11:31:36
[wh1080] model     : Fineoffset-WHx080
[wh1080] Msg type  : 0
[wh1080] Station ID: 131
[wh1080] Battery   : 1
[wh1080] Temperature: 16.0 C
[wh1080] Humidity  : 42 %
[wh1080] Wind Direction: 0
[wh1080] Wind avg speed: 7.34
[wh1080] Wind gust : 11.02
[wh1080] Total rainfall: 264.9
[wh1080] Integrity : CRC
```

But didn't push it to mqtt.

zuckschwerdt, 2025-04-28:
Nothing obvious wrong with your config.

Maybe listen to the raw MQTT with `mosquitto_sub -h $host -p $port -u $username -P $password -t '#' -v` or `mqttui --broker mqtt://$host:$port --username $username --password $password`

The pipe looks fragile as the `kv` output isn't intended to be machine-readable, perhaps try `-F json` and `jq`

kocicky, 2025-04-30:
I stripped `output kv` and now it works. But I'm pretty confident that this is the configuration I used from the start, but after update this configuration didn't work.

zuckschwerdt, 2025-04-30:
We don't usually test manipulation of the console output. The recommended way to post-process is to use an output like Syslog or MQTT and then a filter script like in our examples. https://github.com/merbanan/rtl_433/tree/master/examples

zuckschwerdt, 2025-05-02:
Try to run without the pipe to see what the actual error is. Maybe add `verbose` and `output log`.

kocicky, 2025-05-04:
So, I tried this configuration:
```
frequency 868295000
analyze 42
protocol 32
verbose 1
```

And the result is still same, crashing without explanation:
```
Starting rtl_433 with wh1080.conf...
[wh1080] rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[wh1080] 
[wh1080] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[wh1080] 
[wh1080] Found Rafael Micro R820T tuner
[wh1080] Exact sample rate is: 1000000.026491 Hz
[wh1080] [R82XX] PLL not locked!
[wh1080] [R82XX] PLL not locked!
/run.sh: line 129:   132 Killed                  rtl_433 -c "$live" > >(sed -u "s/^/[$tag] /") 2> >(sed -u "s/^/[$tag] /" 1>&2)
s6-rc: info: service legacy-services: stopping

```

zuckschwerdt, 2025-05-04:
Never use `analyze`. Looks like the pipe is still there? Run just `rtl_433 -c "$live"`

gdt, 2025-07-04:
Please ask for help  from the HA addon people.   This isn't clearly an rtl_433 problem and we have been unable to get the needed info to understand what's going on.

closed 2025-07-04 by gdt