merbanan/rtl_433 #3261: Detecting 433 Schrader but not 315 GM

tpitman, 2025-04-28:
I am running rtl_433 on a Raspberry Pi. I am trying to detect TPMS sensors. I have 2 different types of TPMS sensors.

One set is the Schrader-EG53MA4. These are sending on 433MHz. I have several and they all seem to show up eventually.

The other set are supposed to be generic GM sensors at 315MHz.

I am using a U508 to trigger them sending data.

I also have an RF analyzer that I can see the signals being broadcast. I can only see them in real time and not over time, so I just see the flashing of the data as blips on this scope.

I can see both of them transmitting on the scope.

When I use this command line:

`rtl_433`

I can see the basic Schrader output like this:

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-04-28 10:16:14
model     : Schrader-EG53MA4                       type      : TPMS          flags     : 4c960080      ID        : 06C3E1
Pressure  : 357.5 kPa    Temperature: 81.0 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
```

The issue / question is about the GM sensors. In the tpms_gm.c file I see this comment:

```
rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=120,l=0,r=15600'
```
I am not sure why there isn't a -f 315000000 parameter on that command line, so I added it myself and use this full command line:

`rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=120,l=0,r=15600' -f 315000000`

I never see anything from rtl_433 even though I see blips happening on the scope after triggering with the U508. The blips continue after the U508 button is not pressed anymore, so pretty sure it is coming from the GM sensors.

What can I do to diagnose why the GM sensors won't show up?

tpitman, 2025-04-28:
I installed CubicSDR on my linux computer that I have rtl_433 on.

I ran it and verified that there is a signal at 315 MHz when I use the U508 on this TPMS sensor.

Here is the image:

[image]

The MHz in CubicSDR is set to 315 with a 200k bandwidth.

So I went to the next step in the instructions you gave and tried this command line:

`rtl_433 -w ~/file_315M_250k.cu8 -T 20 -f 315M`

That output several responses from some sensors I also have on my desk that should only be transmitting at 433MHz. I also see packets that are getting decoded from those other sensors. How can that be happening when I have the frequency set to 315MHz and not 433MHz?

So I decided to add -S unknown because the 433MHz sensors are being decoded by the Schrader decoder because that is what they are. Adding -S unknown didn't help.

How can I limit the file to only the 315MHz sensor? I want to upload it, but am not sure it will only contain the 315MHz sensor.

zuckschwerdt, 2025-04-28:
You'd usually write `rtl_433 -v -f 315M -w ~/file_315M_250k.cu8 -T 20` (first input args, then output args) but I don't think it matters. The SDR is tuned to that center frequency (adding `-v` will show `[SDR] Tuned to 315.000MHz.`) but the antenna might pick up "shadows" of other strong signals. If it's a 433/868 MHz antenna you could see those senders on your desk. Not too sure about that but I've heard that the RTL-SDR does not have good selectivity.

tpitman, 2025-04-28:
I have both a 433 and 315 antenna and currently have the 315 one installed. I will see about just moving the 433 sensors in another room or something.

Thanks for the help.

tpitman, 2025-04-28:
Big AHA moment. I was so used to see the Schrader output for the devices I have at 433MHz, that when I saw the same Schrader decoder output I didn't bother to take a look at the ID values of the packets.

Turns out these are the devices that are transmitting at 315MHz.

The problem is they are not supposed to be Schrader compatible devices. They are supposed to be GM devices. I bought them with part numbers that specifically are supposed to be GM OEM replacement TPMS sensors.

I tried adding the -R 275 so that only the GM decoder is active, but then nothing comes out.

Could you help me understand what is going on? Are the decoding protocols close enough that the Schrader works and the GM doesn't for some reason?

I can upload the .cu8 file to that other folder or attach it here. Which would you prefer?

tpitman, 2025-04-28:
Here is the zip file.

[file_315M_250k.cu8.zip](https://github.com/user-attachments/files/19948698/file_315M_250k.cu8.zip)

zuckschwerdt, 2025-04-28:
Could very well be, the Schrader EG53MA4 has the same packet length and checksum as the GM. Temp and Pressure are flipped and you'd get wrong values though. Also both use 120 µs OOK instead of the more common 50 µs FSK.
But Schrader EG53MA4 has 120 bit packets and GM has 130 bit packets -- and your sample shows 120 bit packets.

Try to pressurize the sensor and see if the values shown as EG53MA4 fit. A small Garden Sprayer works nicely to pressurize sensors on your desk. Something like this, the cap is big enough to drop a TPMS in: https://www.gardawn.co.uk/cdn/shop/files/1.5L_Garden_Sprayer_8.webp

tpitman, 2025-04-28:
I have made a pressure pipe I can use for that, but still wonder why doesn't the tpms_gm decoder do anything especially when I use the -R 275 (which is the number that comes up for the gm decoder)?

tpitman, 2025-04-29:
> Just because the part is for a GM vehicle does not necessarily mean it is a type-275 device. A big company might use various kinds of parts. Try to understand what is going on and stay open-minded about what might be. The world is often messy.

I hope I didn't come across as upset or anything like that. I am just asking questions so I can better understand.

zuckschwerdt, 2025-04-29:
Lot's of people we are trying to help, answers might be terse.

To your question: decoders can't be forced, every data stream is passed to all active decoders to see if it fits.

gdt, 2025-04-29:
Don't worry, I didn't take your comments as upset or hostile.  I just perceived that you had a strong belief that the name on the car must match the name on the decoder in rtl_433 and wanted to suggest that you treat that belief as weaker and be open to contrary evidence.    The things in rtl_433 that are figure out are because somebody figured them out, one at a time, and more help from more people is great.

gdt, 2025-07-04:
It seems like this issue is basically a question and so far we don't have any real evidence that there's a rtl_433 bug.  Just that  something is odd/unexpected about the reporter's sensors.

Feel free to open a discussion for further help.

closed 2025-07-04 by gdt