merbanan/rtl_433 #3267: Bresser Weather Center WTW 5in1 latest model different behaviour?

schunckt, 2025-05-02:
Hi all!

I own a new Bresser Weather Center WTW 5in1 which seems to behave different than implemented.
I tried -R 119  172  173 but decding doesnt find data.
Does anyone else notice something similar?
Also does anyone now the exact frequency and maybe other parameters which could affect receiving or decoding?

Raw data like so:
```
time      : 2025-05-01 13:18:26
model     : bresser      count     : 1             num_rows  : 1             rows      :
len       : 184          data      : aaaaaaaba72ba72bcdcfd41378fe12de14c5f51375a6e0
codes     : {184}aaaaaaaba72ba72bcdcfd41378fe12de14c5f51375a6e0

```
It seems already the prefix is different and the bit lenght, if I understand the code right.
I did already play with adapted prefix but then I get  "too short" or  "digest" failures.

Best
  T.

Device details:
----------------------------------------------
Indoor unit is labled
Art. No. 7002512
Weather Center WTW 5in1
Weather station w. multif. sensor
Sensor Type: 7802580/HC2

(I need to lookup the Outdoor unit ...)

ProfBoc75, 2025-05-02:
@schunckt : it should be compatible.

Check if you are using last version of rtl_433
What is your rtl stick model / tuner  ref ?

Some guidance / advice:
- Best rtl usb stick models are based on the old R820T/T2 tuner now manufactured under ref R828D / R860
- Avoid any fc0012 /fc0013 tuner, they are bad quality, bad sensitivity, the frequency range is narrow (less than 1Ghz, when above reach 1.7 or 1.8 GHz.
- Metal case is always better than the blue/black plastic case, metal reduce the external RF noise
- Take care about the antenna, put it far from electronics, and vertically the most of the time the polarity of the signal is vertical.
- Keep less than 30 meters  between the external weather station sensor and the stick, or less if you have thick walls. 
- Add a good shielded USB cable extension to put the rtl stick with its antenna as far as possible from the PC. Far from any electronic devices like screen, power supply and so on.

Bresser is working at 868 Mhz, I noticed 868.3 Mhz or 868.34 Mhz into the rtl_433 bresser decoders.

schunckt, 2025-05-03:
Hi!

Stick logs as  "trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001"
Blue plastics with external Atenna. Basically it works nice because i could receive other neighbour device data :)
Beside that I went close to the outdoor unit 5m air so I'm quite sure this is not the issue, or not only.

the rtl is compiled from the repo and shows as
"rtl_433 version 25.02-13-gcec57bc0 branch master at 202504151251 inputs file rtl_tcp RTL-SDR with TLS"

Maybe one can provide exact parameters, like center frequency, bandwidth, PCM pulse width .... ?
I've not yet found what are the exact defaults.

   T.

schunckt, 2025-05-03:
FYI it works now.

03.05.2025 WORKS
rtl_433 -f 868.3M -R 172 -X n=bresser,m=FSK_PCM,s=121,l=121,r=2500,preamble=aaaa2dd4

after that it also works with
rtl_433 -f 868.3M -R 172 
Does the -X tweaking store something?

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-05-03 16:47:48
model     : Bresser-6in1 id        : a3629444
channel   : 0            Battery   : 1             Temperature: 26.8 C       Humidity  : 42            Sensor type: 1
Wind Gust : 1.7 m/s      Wind Speed: 1.7 m/s       Direction : 281           UV        : 0.0           Startup   : 1
Flags     : 0            Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-05-03 16:48:00
model     : Bresser-6in1 id        : a3629444
channel   : 0            Sensor type: 1            Wind Gust : 3.1 m/s       Wind Speed: 2.9 m/s
Direction : 210          Rain      : 19.6 mm       Startup   : 1             Flags     : 1             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-05-03 16:48:12
model     : Bresser-6in1 id        : a3629444
channel   : 0            Battery   : 1             Temperature: 26.8 C       Humidity  : 43            Sensor type: 1
Wind Gust : 1.9 m/s      Wind Speed: 1.9 m/s       Direction : 174           UV        : 0.0           Startup   : 1
Flags     : 0            Integrity : CRC

```

868.4M also seems fine. Need to figure out the exact channel width. May help to improve signal quality. Currently it works inside the basement, Antenna behind a Window, far away from outdoor unit :)

zuckschwerdt, 2025-05-03:
No, a `-X` flex decoder does not alter any config. Use `-Y autolevel -M level -M noise` to improve reception and log signal level, freqs, and noise.

gdt, 2025-07-04:
Reading again, I think this is an answered question.