merbanan/rtl_433 #3269: Add support for ThermoPro TP827B

cmaxwe, 2025-05-06:
I could use some help adding the TP827B.

I have made some pretty good progress but wanted some eyes on it to make sure that I am not missing anything big. I haven't gotten to the checksum yet but I am also worried that I won't be able to figure that out. Also haven't written C in a fair bit so might need some help getting it built in.

Here is a capture file with the first probe in an oven: [tp827b.zip](https://filedrop.c-maxwell.net/tp827b.zip)

As with other ThermoPro units the signal is FSK PCM.

Using:
./rtl_433 -f 915M -F json -X "n=tp827b,m=FSK_PCM,s=105,l=105,r=5500,preamble=b4bb3abb" -F json | jq --unbuffered -r '.codes[0]'

I get the following codes (with observations in f + converted to c + what port the probe is in):
# 78.4f / 25.77c Probe 1
bb6 00c4 3f80 3f80 3f80 3f80 3f80 3f80 3f80 0c0240003c002 aaaaa800000000000000

# 77.2f / 25.11c Probe 2
bb6 3f80 3f80 00c1 3f80 3f80 3f80 3f80 3f80 0c0240001a402 aaaaa800000000000000

# 77.0f / 25c Probe 3
bb6 3f80 3f80 3f80 3f80 bf80 3f80 3f80 0c02400023402 aaaaa800000000000000

# 73.4f / 23c Probe 1
bb6 00b7 bf80 3f803f803f803f803f803f800c02400026402aaaaa80000000000000

# 73.0f / 22.8c Probe 1
bb6 00b6 bf803f803f803f803f803f803f800c02400032002aaaaa800000000000000

# 106.3f / 41.3c Probe 1
bb6 0109 ff803f803f803f803f803f803f800c02400037402aaaaa800000000000000

# 102.7f / 39.3c Probe 1
bb6 0100 ff803f803f803f803f803f803f800c02400002802aaaaa800000000000000

# 22.1f / -5.5 Probe 1
bb6 0037 7f803f803f803f81fc01fc0060120001780155555400000000000000

# 165.6f / 74.2c Probe 1
b6 019e 3f803f803f803f803f803f803f800c02400010c02aaaaa80000000000000

# 171f / 77.2c Probe 1
b6 01ab bf803f803f007f007f007f007f00180480007380554aaa0000000000000

# 208.4f / 98c Probe 1
b6 0209 3f803f803f803f803f803f803f800c46c00005402aaaaa80000000000000

# 232f / 111.1c Probe 1
b6 0244 3f803f803f803f803f803f803f800c46c0002b402aaaa50000000000000

I think the bitbench looks something like this (each row is an example with the probe in different port):
[bitbench](https://triq.net/bitbench#c=bb600c43f803f803f803f803f803f803f800c0240003c002&c=bb63f803f8000c13f803f803f803f803f800c0240001a402&c=bb63f803f803f803f8000c0bf803f803f800c02400023402&c=bb63f803f803f803f803f803f8002443f800c02400039802&f=hh%20ID%3Ah%20TEMP_One%3A16d%20hhhh%20TEMP_Two%3A16d%20hhhh%20TEMP_THREE%3A16d%20hhhh%20TEMP_FOUR%3A16d%20hhhh%2052h&z=1&cw=4)

Putting examples in Chatgpt with the decimal for the hex seems to guess that the formula is: 
0.4 * decimal value = the temp in f

Seems to be pretty consistent but that formula is a lot different than other ThermoPro examples I have seen which makes me wonder whether it is actually that simple. It is pretty close on all my examples if not perfect though.

Stuff I think I need to still figure out -
1. More observations up to higher temps. I mostly use this for smoking and will rarely exceed 275f - but will take it up to 350f at least.
2. What some of the other stuff is in the code - there are some alarms I am sure and maybe some stuff like low battery based on reading other Thermopro examples.
3. Checksum

Once I get the decoding solid then I guess it is jumping into the code.

Any thoughts/hints/help would be appreciated.

zuckschwerdt, 2025-05-06:
Was the sample file captured with `rtl_433 -f 915M -w tp827b.cap`? I.e. is it 915M and 1000k? A good name would be `tp827b_915M_1000k.cu8` then.

I guess your `-F json -X ... -F json` is a type, two outputs will duplicate each line.

The `b4bb3abb` isn't a sync word we have seen before and a sync word should be 16 or 32 bit, this looks 4 bit too long -- is that a type, most of your codes start with bb6?

A scale of `0.4` is indeed strange. But that factor is also `*0.1 *4` which could indicate that `0.1` is the factor and you are shifted by 2 bit, i.e. the two lowest bits are missing. Shift two bits left, or use a preamble of `aaaa d2ec eaee` with a format of `ID:hh TEMP_One:16d hhhh TEMP_Two:16h hhhh TEMP_THREE:16d hhhh TEMP_FOUR:16d hhhh 16h 16h CRC:8h 8h 80h`

Other oven temps have setpoints interleaved with the readouts, maybe that's (partly) what those 3f80's (`fe00`) are?

There are some error in your codes, one of the last columns has values of `188d` `311b` `6236` -- these are the same bit pattern but off by one bit shift. The timing has missed or inserted a bit somewhere, I guess `311b` would be the correct value.

If we assume that then the checksum is a CRC-8 poly 0x07 init 0x00.

zuckschwerdt, 2025-05-06:
[BitBench](https://triq.org/bitbench/#c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60240ff803f803f803f803f803f803f800c46c00024c02aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb602417f803f803f803f803f803f803f800c46c0001a402aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60242bf803f803f803f803f803f803f800c46c00039c02aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60243bf803f803f803f803f803f803f800c46c0002d802aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60244ff803f803f803f803f803f803f800c46c00034002aaaaa80000000000000&c=%7B348%7Daaaaaaaaaa5555697675776c048aff007f007f007f007f007f007f00188d800015005555550000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60245ff803f803f803f803f803f803f800c46c00020402aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb60248bf803f803f803f803f803f803f800c46c00030c02aaaaa80000000000000&c=%7B349%7Daaaaaaaaaaaaaab4bb3abbb602497f803f803f803f803f803f803f800c46c0003bc02aaaaa80000000000000&f=ID%3Ahh%20TEMP_One%3A16d%20hhhh%20TEMP_Two%3A16h%20hhhh%20TEMP_THREE%3A16d%20hhhh%20TEMP_FOUR%3A16d%20hhhh%2016h%2016h%20CRC%3A8h%208h%2080h&z=1&a=Preamble&m=aaaa%20d2ec%20eaee&cw=4) with shifted and only valid data (CRCs check out).

ProfBoc75, 2025-05-06:
@cmaxwe: I've already helped on the previous decoders on other ThermoPro sensors , so I can help you here, need some time to work on it.

But, I can confirm that the signal transmit is not easy to decode as the frequency is not stable along the rf signal, in result you may have 1 bit or more shifted inside the data, the reason why we put sanity controls to exclude such signal. 

May I propose you to change the preamble to aaaaad2eceaee, and reduce a little the reset value to 2500

    rtl_433 -f 915M -X "n=tp827b,m=FSK_PCM,s=105,l=105,r=2500,preamble=d2eceaee"

I guessed this [bitbench](https://triq.net/bitbench#c=%5BFrom%20Issue%5D&c=b4bb3abbbb600c43f803f803f803f803f803f803f800c0240003c002aaaaa800000000000000%20%5B78.4f%20Probe%201%2C%20wrong%2C%20shifted%5D&c=b4bb3abbbb63f803f8000c13f803f803f803f803f800c0240001a402aaaaa800000000000000%20%5B77.2f%20Probe%202%2C%20wrong%2C%20shifted%5D&c=b4bb3abbbb63f803f803f803f80bf803f803f800c02400023402aaaaa800000000000000%20%5B77.0f%20Probe%203%2C%20wrong%2C%20truncated%20and%20shifted%5D&c=b4bb3abbbb600b7bf803f803f803f803f803f803f800c02400026402aaaaa80000000000000%20%5B73.4F%20Probe%201%2C%20wrong%2C%20shifted%5D&c=b4bb3abbbb600b6bf803f803f803f803f803f803f800c02400032002aaaaa800000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbbb60109ff803f803f803f803f803f803f800c02400037402aaaaa800000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbbb60100ff803f803f803f803f803f803f800c02400002802aaaaa800000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbbb600377f803f803f803f81fc01fc0060120001780155555400000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbb6019e3f803f803f803f803f803f803f800c02400010c02aaaaa80000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbb601abbf803f803f007f007f007f007f00180480007380554aaa0000000000000%20%5Bwrong%2C%20shifted%5D&c=b4bb3abbb602093f803f803f803f803f803f803f800c46c00005402aaaaa80000000000000&c=b4bb3abbb602443f803f803f803f803f803f803f800c46c0002b402aaaa50000000000000&c=&c=%5BFrom%20BitBench%5D&c=b4bb3abbbb600c43f803f803f803f803f803f803f800c0240003c002%20%5B%2078.4F%20Probe%201%20%5D&c=b4bb3abbbb63f803f8000c13f803f803f803f803f800c0240001a402%20%5B%2077.2F%20Probe%202%20%5D&c=b4bb3abbbb63f803f803f803f8000c0bf803f803f800c02400023402%20%5B%2077.0F%20Probe%203%20%5D&c=b4bb3abbbb63f803f803f803f803f803f8002443f800c02400039802%20%5B%20232.0F%20Probe%204%5D&c=&c=%5BFrom%20Capture%20File%5D&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60240ff803f803f803f803f803f803f800c46c00024c02aaaaa8000000%20%5B%20230.7F%20Probe%201%20%5D&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb602417f803f803f803f803f803f803f800c46c0001a402aaaaa8000000%20%5B%20230.9F%20Probe%201%20%5D&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60241ff803f803f803f803f803f803f800c46c00030802aaaaa8000000%20%5B%20231.1F%20Probe%201%20%5D&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60242bf803f803f803f803f803f803f800c46c00039c02aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb602433f803f803f803f803f803f803f800c46c00007402aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60243bf803f803f803f803f803f803f800c46c0002d802aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60244bf803f803f803f803f803f803f800c46c00001802aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60244ff803f803f803f803f803f803f800c46c00034002aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb602457f803f803f803f803f803f803f800c46c0000a802aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60245ff803f803f803f803f803f803f800c46c00020402aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60246ff803f803f803f803f803f803f800c46c0001c802aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb602473f803f803f803f803f803f803f800c46c00017802aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60247bf803f803f803f803f803f803f800c46c0003d402aaaaa8000000&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb60248bf803f803f803f803f803f803f800c46c00030c02aaaaa8000000%20%5B%20233.8F%20Probe%201%20%5D&c=%7B321%7Daaaaaaaaaaaaaab4bb3abbb602497f803f803f803f803f803f803f800c46c0003bc02aaaaa8000000%20%5B%20234.1F%20Probe%201%20%5D&f=ID%20%3F%208h%20T_ONE%2016d%20T_ONE_H%2016d%20T_TWO%2016d%20T_TWO_H%2016d%20T_THREE%2016d%20T_THREE_H%2016d%20T_FOUR%2016d%20T_FOUR_H%2016d%20FIXED%20%3F%2016h%20ZERO%2016h%20CRC%208h%20ZERO%208h%20A%2024h%20TRAILING%20BIT%20100x&z=1&a=Sync&m=d8&cw=4)

Can you please, share more codes from above command. We need to confirm my guess, set the Max temp (Smart temp) on the transmitter, one probe at a time, to identified the data layout position, my attempt is based on previous decoded ThermoPro and your findings too.

In my bitbench, my attempt is Temp coded in 16 bits, scale 10, in Fahrenheit
Data layout looks: Temp Probe 1 Current, Temp Probe 1 Max Alarm, Temp Probe 2 Current, Temp Probe 2 Max Alarm ....
Without Probe and/or Without Max defined, you have 0xFE00 = 65024.

Are you able to change the unit in the transmitter from F to C, and see if any impact on the data ?

On other ThermoPro sensors, the C or F information is set at a flag level, but data values stay coded in C offset 500 scale 10, so we need to check here if the data are changed or stay in F ?

Checksum is very nice CRC-8/SMBUS from previous bytes.

    reveng -w 8 -s d80903fe00fe00fe00fe00fe00fe00fe00311b000093 d80905fe00fe00fe00fe00fe00fe00fe00311b000069 d80907fe00fe00fe00fe00fe00fe00fe00311b0000c2 d8090afe00fe00fe00fe00fe00fe00fe00311b0000e7
    width=8  poly=0x07  init=0x00  refin=false  refout=false  xorout=0x00  check=0xf4  residue=0x00  name="CRC-8/SMBUS"

ProfBoc75, 2025-05-06:
@cmaxwe :  I'm still working on it, as lot of data are shifted, I need to play with bitbench, one by one to guess the good result.

Little improvement as the pulse is between 106 and 112 µs :

    rtl_433 -f 915M -X "n=tp827b,m=FSK_PCM,s=110,l=110,r=2500,preamble=d2eceaee,bits>=319" 2>&1 | grep codes

When Celsius , the values transmit in Celsius, scale 10, and just before the CRC there are FLAGS, which changed too, because of Celsious, and also about the alarms.

Compare to other ThermoPro models, you're right, the High and Low temp are not transmitted, only alerts.

Data Layout looks like:

    Byte Position     0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28
    Sample           d8 09 03 fe 00 fe 00 fe 00 fe 00 fe 00 fe 00 fe 00 31 1b 00 00 93 00 aa aa aa 00 00 00
    Data             II 11 11 FF FF 22 22 FF FF 33 33 FF FF 44 44 FF FF XX YY 00 00 CC 00 AA AA AA 00 00 00

II:{8} Model or ID
11 11:{16} Temp Probe 1, C or F, scale 10,  0xFE00 is no probe
FF FF :{16} Fixed value, 0xFE00
22 22:{16} Temp Probe 2, C or F, scale 10,  0xFE00 is no probe
33 33:{16} Temp Probe 3, C or F, scale 10,  0xFE00 is no probe
44 44:{16} Temp Probe 4, C or F, scale 10,  0xFE00 is no probe
XX: Some Flags about normal, alert/alarm, 0x30 = Normal, 0xC0 = Alert, to be deep dive to guess other Flags/bits
YY: Some Flags, last bit = 1 if Fahrenheit, = 0 if Celsius, , to be deep dive to guess other Flags/bits
00 : Fixed Zeros
CC: CRC-8/SMBUS, poly 0x07, init 0x00, final XOR 0x00 from 21 previous bytes.
AA: Fixed 0xAA values

cmaxwe, 2025-05-06:
I will get some samples with the new command and some active alarms on each port.

cmaxwe, 2025-05-07:
I removed the bits>=319 because it was going really slow (like a few codes every 10min slow)

No Alarm + BBQ Range - Probe 1
d80321fe00ff007e00fe00fe00fc01fc01800200014402aaaaa8000000000
d80321fe00fe00fe00ff007f007f007f00600080005800aaaaaa000000
d80322ff007f007e00fc01fc01fe00fc018002000198015555540000000000
ec01917f007f007f007f803fc01fc01fe00c0010000cc00aaaaaa000000
d80326fc01f803f803f007f007f007f00600080004e80555555000000000
d80251fc01fc01fc01f803f803f007f006000400018a0155555400000

Active Low Alarm + BBQ Range - Probe 1
d802ddfc01f803f803f803f803f803f80300440001a402aaaaa80000000000000000000000000000000000000000000000
d802defc01fc01fc01fc01fe00ff007f00600880003480549554000000000
d802defe00fc01fc01f803f803f007c0180220000d2012a555000000000
d802e2fe00fe00fe00fe00fe00fe00fe00c01100005f00aaaaaa000000
d801c7f803f803f803f803f007f007f00401100008900aaaaaa0000000000000000000000000000000000000000000000

Active High Alarm + BBQ Range - Probe 1
d8035efe00fe00fe00fe00fc01fc01fc0180220002a8015555540000000
d80345fe00fe00fe00fe00fe00fe00fe00c0110001c800aaaaaa000000
d80333fe00fe00fe00fe00fe00fe00fe00c0110001fc00aaaaaa000000
d80387ff007f007f007f007e00fe00ff0060088000bf802aaaaa800000
d00713fc01f007f007f007e00fc01fe00c01100012600aaaaa400000000

Active Low Alarm + BBQ Range - Probe 2
d8fe00fc000557f803f803f007f007e00c02100016a00aaaa9400000000
d8f803f00015aff007f007e00fc01fc00c021000154014aaaa800000
d8fc01f8000ad7f803f007e00ff003f8030080000b500555555000000
d8fc01fc00056bf803f803f803f803f80300840005a802aaa950000000
d8fe00fe0002b5fe00fe00fe00fe00fe00c021000154015554a8000000000000
d8fe00fc00056bf803f803f007e00fe00c02100016a00aaaaaa000000

Active High Alarm + BBQ Range - Probe 2
d8fc01f80015fff803f007e00fc01f8030084000ed0055552a0000000000000000000
d8fc01fc000583fc01f803f007e00fc00c02100039f005555550000000000000000000
d8ff007f000163ff007f801fc00fc01fc01804200065201552a9000000
d8fe00fe0002cafe00fc01fc01fe00fe0080420007ea0155554800000
b1f803f0000ccfe00fe00fe00fe00fe00c00200068401552aa8

No Active Alarm - Meat - Probe 1
d802e3fc01fc01f803f007e00fe00fe00f00100008200aaaaaa000000
d802e3fc01f803803f803f803f803f0078008000410054aaa800000
d802e3f803f803f007f007e00fc01f0078008000410055555500000000000000000
d802e2fc01fc01f803f803f803f007e00f00100005400aaaaaa0000000000000000000000000000000000000000000000
d802c5fc01fc01fe00ff003fc01fc01fc01e0020000a80155555400000

Active Alarm - Meat - Probe 1
d80379fc01f803f803f803f803f803f00780880006b804aaaaa000000
d801c17f007f007f007f007f007f007f007011000056012aaaa800000000000
d80380fe00fe00fc01f803f007f007e00f0110000fc00aaaaa800000
d80380fe00fe00fe00fe00fc01fc01f803c0440003f002aaaaa8000000
d80380fe00fe00fe00fc01f803f007f00780880007c015555540000000

ProfBoc75, 2025-05-08:
@cmaxwe : Some updates / findings based on your last codes

Little update about the data layout, as a fixed value at 0 changed.

    Byte Position     0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28
    Sample           d8 09 03 fe 00 fe 00 fe 00 fe 00 fe 00 fe 00 fe 00 31 1b 00 00 93 00 aa aa aa 00 00 00
    Data             II 11 11 FF FF 22 22 FF FF 33 33 FF FF 44 44 FF FF MX LT 00 0H CC 00 AA AA AA 00 00 00

M:{4} Mode, Normal (no alarm set) = 3, BBQ = 0xC, Meat = 0xF
X:{1} Unknown flags, from the first sample = 1, but from last codes always = 0
L:{4} Low Temp Alarm flags: L1 = Probe 1 Low Temp reached, L2 = Probe 2 Low Temp reached ... 

     Bit   0  0  0  0
     Flag L4 L3 L2 L1

T:{4} First 3 bits unknown, last bit T, Temperature unit flag, 1 = Fahrenheit, 0 = Celsius

     Bit   0  0  0  0
     Flag ?1 ?2 ?3  T

?1 & ?3 changed in some codes, ?2 always = 0.

H:{4} High Temp Alarm flags: H1 = Probe 1 High Temp reached, H2 = Probe 2 High Temp reached ...

     Bit   0  0  0  0
     Flag H4 H3 H2 H1

L3, L4, H3, H4 to be confirmed, with more codes using the probe 3 & 4, same use cases (Normal / BBQ / Meat values) as done with probe 1 & 2.

Also, missing information regarding the nibbles X and T, how they behave, as last use cases did not behave exactly as first codes.

Here hexa values of nibbles, then binary values from different use cases observed in this issue:

| M | X | L | T | H | Comment |
| :-: | :-: | :-: | :-: | :-: | :-: |
| 3 | 1 | 1 | b | 0 | Probe 1 around 230.0F, temp F, From Sample captured file |
| 0011 | 0001 | 0001 | 1011 | 0000 | " |
| 3 | 0 | 0 | 9 | 0 | Probe 1 around 70.0F, temp F, From codes in the issue |
| 0011 | 0000 | 0000 | 1001 | 0000 | " |
| 3 | 0 | 0 | 8 | 0 | Probe 4 around 22 C, temp C, From codes in the issue |
| 0011 | 0000 | 0000 | 1000 | 0000 | " |
| C | 0 | 0 | 1 | 0 | No Alarm + BBQ Range - Probe 1, Temp F |
| 1100 | 0000 | 0000 | 0001 | 0000 | " |
| C | 0 | 1 | 1 | 0 | Active Low Alarm + BBQ Range - Probe 1, Temp F |
| 1100 | 0000 | 0001 | 0001 | 0000 | " |
| C | 0 | 1 | 1 | 1 | Active High Alarm + BBQ Range - Probe 1, Temp F |
| 1100 | 0000 | 0001 | 0001 | 0001 | " |
| C | 0 | 2 | 1 | 1 | Active Low Alarm + BBQ Range - Probe 2, Temp F |
| 1100 | 0000 | 0010 | 0001 | 0001 | " |
| C | 0 | 2 | 1 | 3 | Active High Alarm + BBQ Range - Probe 2, Temp F |
| 1100 | 0000 | 0010 | 0001 | 0011 | H = 3, expected = 2, may be probe1 alarm kept ? |
| F | 0 | 0 | 1 | 0 | No Active Alarm - Meat - Probe 1, Temp F |
| 1111 | 0000 | 0000 | 0001 | 0000 | " |
| F | 0 | 1 | 1 | 0 | Active Alarm - Meat - Probe 1, Temp F |
| 1111 | 0000 | 0001 | 0001 | 0000 | " |

ProfBoc75, 2025-05-09:
@cmaxwe : In the meantime I created a new decoder based on my guessed and your sample codes. 

I create a PR with this new decoder for this sensor. I let you test it, and let me know what need to be corrected.

As the signal of ThermoPro sensor is difficult to decode, the frequency change/variation during the transmit (in result the bits are shifted/truncated), you may have less success to decode the signal.
Notice also, that the CRC is checked and in case of error, the data are not presented.

cmaxwe, 2025-05-09:
Thanks - I will post new samples tomorrow and will definitely test it once it is merged.

cmaxwe, 2025-05-09:
I compiled it and tried running it but not seeing anything come up.

Running with "rtl_433 -f 915M"

First line is:
`rtl_433 version 25.02-16-ga816bc70 branch feat-thermopro_tp827b at 202505081327 inputs file rtl_tcp RTL-SDR with TLS`

So I am running your branch but I don't see the TP827B device in the `rtl_433 -R` list.

cmaxwe, 2025-05-09:
Played with the decoder a bit. Looking really good overall!

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-05-08 23:45:27
model     : ThermoPro-TB827B                       id        : d8
Display Unit: Fahrenheit Temperature 1: 72.1 F     Temperature 2: 71.4 F     Temperature 3: 71.8 F     Temperature 4: 71.6 F     Flags1    : 0             Flags2    : 1             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
`

Few things I noticed...
1. Each probe can be set to Meat or BBQ. If I set the first one to Meat it shows "Mode: Meat" but if I set it as BBQ then there is no mode shown. Other probes don't seem to show a mode regardless of whether they are Meat or BBQ.
2. The first few results had `Alarm Max 1: 1` but there was no active alarm
3. Most of the updates were every 4 seconds or so but once in a while it missed a bunch and was around 45 seconds. I think this got better as I moved the unit to a different room.
 
I will get you more examples with alarms and different modes on the different probes.

I didn't test Celsius and I didn't test alarms as it makes noise and the rest of my family is sleeping... :-)

Really working well though - for my usage I don't really care much about the alarms or modes. So I can actually start using this right away.

ProfBoc75, 2025-05-09:
@cmaxwe: about the mode, you're right, we have alarm flags for each probe, but I was not able to guess  mode flags by probe. I need more codes, please, can you run the decoder with verbose option to get message values like flex. 

    rtl_433 -f 915M -R 277:vv

cmaxwe, 2025-05-09:
@ProfBoc75 - some more samples for you

[Directory with Samples](https://filedrop.c-maxwell.net/rtl_outputs/05092025/)

I separated them into files like this:

all_probes_c_b_m_m_m_n_n_n_n.txt

c = Celsius or f = Fahrenheit
b = bbq
m = meat
n = no alarm
a = meat alarm
ha  = bbq high alarm
la = bbq low alarm

So the one above is:
Celcius
Probe 1 = BBQ
Probe 2 = Meat
Probe 3 = Meat
Probe 4 = Meat
Probe 1 = No Alarm
Probe 2 = No Alarm
Probe 3 = No Alarm
Probe 4 = No Alarm

I have the raw log then I have a separate file in the directory called "only_codes.txt" that has just each file name and the codes that were returned without anything else.

ProfBoc75, 2025-05-10:
@cmaxwe : Thanks to your last codes, we have some findings.

Mode is like low or high alarms, one bit for each prob.

M: {4}  0 = BBQ / range Lo / Hi , 1 = Meat / Max target temp

    bit 0  0  0  0
        M4 M3 M2 M1

all_probes_f_m_m_m_m_n_n_n_ha.txt , ha not possible with meat, it's probably a typo, and must be all_probes_f_b_b_b_b_n_n_n_ha.txt
confirmed by the values all = 0 at mode level.

About the alarms, the behavior is not the same as previous codes, may be because your tried here with all the probes, and previous codes were done with 1 or 2 probes plugged ?

X:{4} Contains alarm flags also.

Byte position 19 is also going to 0x02 or 0x03 when alarm turns ON, but what ever the probes, the value is mostly 0x03, and one time 0x02 , and stay 0x00 when no alarm triggered.

I just put the probe's number and keep your naming convention.

```
byte:
   17          18          19          20
1110 0000   0000 0001   0000 0000   0000 0000 m4 m3 m2 b1 N4  N3  N2  N1
1101 0000   0000 0001   0000 0000   0000 0000 m4 m3 b2 m1 N4  N3  N2  N1
1011 0000   0000 0001   0000 0000   0000 0000 m4 b3 m2 m1 N4  N3  N2  N1
0111 0000   0000 0001   0000 0000   0000 0000 b4 m3 m2 m1 N4  N3  N2  N1

1111 1111   1111 1111   0000 0011   0000 0000 m4 m3 m2 m1 a4  a3  a2  a1  F
1111 1110   1110 1111   0000 0011   0000 0000 m4 m3 m2 m1 a4  a3  a2  N1  F
1111 1100   1100 1111   0000 0011   0000 0000 m4 m3 m2 m1 a4  a3  N2  N1  F
1111 1000   1000 1111   0000 0011   0000 0000 m4 m3 m2 m1 a4  N3  N2  N1  F
0000 1000   1000 1111   0000 0011   0000 1000 m4 m3 m2 m1 ha4 N3  N2  N1  F
1110 0000   0000 0000   0000 0000   0000 0000 m4 m3 m2 b1 N4  N3  N2  N1  Celsius
0000 1111   1111 1111   0000 0000   0000 0110 b4 b3 b2 b1 la4 ha3 ha2 la1 F
0000 1100   1100 1111   0000 0011   0000 0100 b4 b3 b2 b1 la4 ha3 N2  N1
0000 1100   1100 1111   0000 0010   0000 0000 b4 b3 b2 b1 la4 la3 N2  N1
0000 1000   1000 1111   0000 0011   0000 0000 b4 b3 b2 b1 la4 N3  N2  N1

0011 0001   0001 1011   0000 0000   0000 0000 N4 N3 N2 m1 N4  N3  N2  L1, from the Capture file

4321 4321   4321 4321          xx        4321
MEAT        LOW  ???U          xx        High
BBQ?                n        Alarm
                    i
                    t
```

Going to sleep and will continued this week end to correct the decoder accordingly.

I noticed also that temp range is from 14F to 572F or -10C to 300C, so I will correct also the decoder based on that information I found into the user guide.

ProfBoc75, 2025-05-10:
@cmaxwe : about the byte position 19, with values 0x0, 0x02 or 0x03, could be related to the Alarm mode. I saw in the user guide that the alarm can be on/off globally or set on/off by probe at the transmitter level.

I did lot of updates into the decoder. Please test the last version and let me know. Some controls were wrong, so you may have less success but better accurate results.

cmaxwe, 2025-05-11:
Looks really good so far.

Modes confirmed working as expected on each sensor.

Tested alarm on Meat 3 and it worked - going to test the other alarms tomorrow but looking good.

I am not sure if the alarm setting on the transmitter gets sent - I don't think it has any impact on the receiver. It just changes whether the transmitter device makes a sound/vibrates I think.

JSON for all set to meat with an alarm on 3:
{
    "time": "2025-05-10 22:20:55",
    "model": "ThermoPro-TB827B",
    "id": 216,
    "temp_unit": "Fahrenheit",
    "mode_p1": "Meat",
    "mode_p2": "Meat",
    "mode_p3": "Meat",
    "mode_p4": "Meat",
    "temperature_1_F": 73.9,
    "temperature_2_F": 73.2,
    "temperature_3_F": 85.1,
    "temperature_4_F": 73.4,
    "alarm_meat_3": 1,
    "alarm_on": 1,
    "mic": "CRC",
    "mod": "FSK",
    "freq1": 914.91078,
    "freq2": 914.99693,
    "rssi": -0.118771,
    "snr": 25.21301,
    "noise": -25.3318
}

Working great though!

cmaxwe, 2025-05-11:
Two suggestions / extra features:
1. Can you include the response attributes even if they are not active? (e.g. "alarm_meat_3" was set to 1 but once it went back to 0 it just dropped off the response)
2. Can you include a general "active alarm" response attribute that is 1 if any alarm is actively going and is 0 if there are no alarms?

I know the spirit of this is that it should only return things that are in the signal but #2 would be pretty useful. If you don't agree then no problem - I get that point of view.

ProfBoc75, 2025-05-11:
> Two suggestions / extra features:
>
> 1. Can you include the response attributes even if they are not active? (e.g. "alarm_meat_3" was set to 1 but once it went back to 0 it just dropped off the [...]

@cmaxwe Yes, easy, I just have to remove DATA_COND.
But I will present the alarm_xxx_1/2/3/4 status only for plugged probes, like it's done for the temp, and alarm name aligned with the cooking mode.

ProfBoc75, 2025-05-11:
@cmaxwe : Done, I moved back unknown flags within the second nibble of byte position 18.

Now you have always any alarm status 0 or 1 aligned with the cooking mode (now one criteria of data condition to show the good metric instead of 4)

cmaxwe, 2025-05-20:
Still haven't really tested the alarms yet 100% but my batteries on the transmitter died and it seems to have stopped working. I think maybe the preamble chanaged?

Codes with new batteries:
aaaaaaaaaaaaaab4b675766c01657f007e00fc01fc01fe00fe00fe00300900008b00aa555400000
aaaaaaaa955555697675766c014afe007f007f007e00fe00fe00fe00300900008b00aaaaaa000000
aaaaaaaaaaaaaab4bb3abb3600a57e007e00fe00fe00fe00fe00fe00300900008b00aaaaaa0000000
aaaaaaa95555556969d5d9b00595fc01f803f803f803f803fc01f800c02400022c02aaaaa800000
aaaaaaaaaaaaaab4bb3abb3600b2bf803f803f803f803f803f803f00180480004580555555000000
aaaaaaaaaaaaaab4bb3abb3600b2bf803f803f803f803f007f007f001804800022c02aaaaa800000
aaaaaaaaaaaaaab4bb3abb3600b2bf803f803f803f007f007f007f00180400008b00aaaaa400000
aaaaaaaaaaaaaab4bb3abb3600b17f007f007e00fe00fe00fe00fe00300900008b00aaaaaa000000
aaaaaaaaaaaaa969766aecd802c9fc01fe00fc01f803f803f803f00100900008b00aaaaaa000000
aaaaaaaaaaaaaab4bb3abb3600a57f007f007f803f803f803f803f800c02400022c02aaaa5000000
aaaaaaaaaaaaaab4bb3abb3600b2bf007e00fc01fc01fc01f803f800c0080004580555555000000
aaaaaaaa55555552ecd5d9b00595f803f803f803f007f007f007f00180480004580554aaa000000
aaa95555555555696ceaecd802cafe00fe00fe00fe00f803f803f800c02400022c02aaaaa800000
aaaaaaa55555556974eaecd802cafc01fc01f803f803f803f007f00180480004580552aaa000000
aaaaaaaaaaaaaab4b66aecd802c9fc01fc01f803f007e00fe00fc00402400022c02aaaaa800000
aaaaaaaaaaaaa5697675766c014afe00fe00fe00fe00fe00fe00fc0060120001160155555400000
aaaaaaaaaaaaaab4bb3abb3600a57e00fe00fe00fe00fe00fc01fe00300900008b00aaaaaa000000
aaaaaaaaaaaaaab4bb3ab66c01647e00fe00fe00fe00fe00fe00fe00300900002000aaaaaa000000
aaaaaaaa55555552e9d5d9b00591fc01f803f803f803f803f801fc0060120000400155555400000
aaaaaa95555555697675766c01647f007f007e00fe00fe00fe00fc0060120000400155555400000
aaaaaaaaaaaaaab4bb3abb3600b23f007f007f007f007e00fe00fe0030090000200055555400000
aaaaaaaaaaaaaab4bb3ab66c0163fe00fc01fc01fc01fc01fc01fe0030090000ac0155555400000
aaaaaaaaaaaaaab47675766c0163ff007f007e00fe00fe00fe00fe0030090000ae00aaaaaa000000
aaaaaaaaaaaaaaa9767abb3600b1ff007f007f007e00fe00fe00ff001804800057004aaaaa00000000
aaaaaa95555555697675766c0163ff007f007f803f803f803f803f800c0200002bc0155555400000

cmaxwe, 2025-05-20:
preamble seems to have changed from 'eceaee' to 'eceaec'.

Wonder if the preamble is really 'd2ecea' and that last bit 'ee'/'ec' is like a channel or something.

ProfBoc75, 2025-05-20:
@cmaxwe : strange as these byte/bit are not part of the CRC and the reason why is a preamble.
This sensor is using GFSK modulation, so, it's not easy to get the good rf settings, try other gain value, like -g 15
Another point, if your previous batteries were not very good, may be the frequency was also impacted during  the test and the capture of codes. Try to get the pulse information with -A , see if pulse is still around 110 µs (it was between 106 to 112 µs)

Because of GFSK, I put a tolerance of 5 µs into the decoder, depends on the -A result above, adjust the pulse short_width = long_width = 110 to 105 or 115 , could help (last lines into the decoder), or increase the tolerance to 10 µs, but not sure how it works / behaves here.

cmaxwe, 2025-05-21:
In think you are right. I had strong batteries in it and it was ec then it went back to ee after a bit as they weakened.  I put other known m batteries in and it was ee.  Going to buy some fresh ones tomorrow and confirm that it goes to ec. 

I played around with the short, long and threshold but I didn’t find a setting that was noticeably better than what you had.

Maybe it would be best to set that bit to be ignored and just use ‘d2ecea’ as the preamble?

cmaxwe, 2025-05-24:
-Y classic seems to make a difference on the range.

107 us 106;112 pretty much every time

Sample rate of 1000k works the best too.

250k gives me an error: "pulse_slicer_pcm] sample rate too low for protocol 277 "ThermoPro TP827B BBQ Meat Thermometers 4 probes with Temp, Meat and BBQ Target LO and HI alarms"

All my batteries are reading ee. I don't have any super super fresh ones though so will have to try that this weekend.

cmaxwe, 2025-05-24:
@ProfBoc75 - I don't think the frequency not being stable is related to batteries. I have tried all kinds of batteries (including the old ones) today and it hasn't switched back. Something else must be going on here.

I was planning on using this as a sensor for a PID temperature controller so even if it is the batteries I can't really use this for that application as I will always be worried about whether the batteries have drained from brand new to medium new.

Will likely just put a temp probe on an ESP32 or look at a BLE option. I am suspecting that it might be related to my usb dongle or antenna but replacing those on a hunch will cost as much as going BLE.

ProfBoc75, 2025-05-25:
@cmaxwe : understand, the other point is the GFSK modulation which adding a difficulty to decode the signal. We were facing same problem on other ThermoPro meat / bbq thermometers, anyway, I will take into account your last findings and change the decoder accordingly to increase success.

cmaxwe, 2025-05-25:
> [@cmaxwe](https://github.com/cmaxwe) : understand, the other point is the GFSK modulation which adding a difficulty to decode the signal. We were facing same problem on other ThermoPro meat / bbq [...]

Thanks @ProfBoc75 - I know this was work for you and appreciate the time that you put into this. If you want me to try other things than let me know happy to continue to support this even if I am not going to use it. I am sure someone will stumble upon this that has the same thermometer.

Why can the receiving unit decode the signal so reliably when rtl_433 struggles? Is it just that it is different hardware?

gdt, 2025-07-04:
Where are we on this issue?   It seems like the device is GFSK, and thus it isn't going to work, but it might be nice to capture the wisdom (as doc, not as an open issue!).