merbanan/rtl_433 #3273: "Protocol number specified (276) is larger than number of protocols" when using newest config example with newest stable release.

streamofstars, 2025-05-08:
Current github readme and https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf contain 276 supported protocols whereas the newest official release binary appears to support only 275 protocols, at least when running this binary https://github.com/merbanan/rtl_433/releases/download/25.02/rtl_433-rtlsdr-openssl3-Linux-amd64-25.02.zip and trying to use it with above linked config I get the result as below:
```
rtl_433 -d :00000001 -c rtl_433.example.conf
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR with TLS
Protocol number specified (276) is larger than number of protocols

= Supported device protocols =
<here list of all protocols that stops at 275>
```

Upon manually removing the line with protocol no. 276 from the config file the program starts correctly, so this is clearly the source of the problem.

It is understandable that the main branch can contain newer files that the latest release, however in that case there should be some kind of an indication in the readme and/or config file. Perhaps in the config newer than latest official release such newer options could be disabled by default. Another way of dealing with this may be including example of a config file valid for a given version in the release zip file. Thanks.

zuckschwerdt, 2025-05-08:
Yes, something to think about. We also have nightly binaries and the RainPoint sensor (276) is available there.

ProfBoc75, 2025-05-08:
[Nightly binaries here](https://github.com/merbanan/rtl_433/releases/tag/nightly)

gdt, 2025-07-04:
I don't see problems resulting from mixing the tip of master and a release as a bug.

If there are files in git that shoudl be in releases, and aren't, please feel free to open a specific bug about that -- but github tagging normally just includes all (which is another kind of bug....).

closed 2025-07-04 by gdt