merbanan/rtl_433 #3281: Improve file output resilience

HaxtonFale, 2025-05-22:
I'm now successfully using `rtl_433` to log the output of my weather sensor(s) to JSONL but I would like to periodically take that file away and compress it so I have a (sorted) backup. Unfortunately, if I take away the file when the program is running, it seems like the program gets a bit confused and no longer writes any entries. I hadn't (yet) configured it as a systemd service (I probably should, to make restarting it easier) but I'd like to propose adopting one of the following approaches:

1. Provide filename templating with an ISO 8601 date placeholder, e.g. for daily rollover. That way any files older than "today" can be safely removed.
2. Create a new output file when the existing one has been removed.

zuckschwerdt, 2025-05-22:
Good idea. Note that moving/removing a file will only move/remove the file entry. The file is still open by inode and should still be written to.
We already process SIGHUP and should be able to extend that to reopen files.

zuckschwerdt, 2025-05-22:
You can test #3282 -- unfinished but working.

zuckschwerdt, 2025-05-23:
An example would be
```sh
mv log.jsonl log_old.jsonl
# note that log_old.jsonl is still being written to
killall -HUP rtl_433
# now it is safe to process the old log file
gzip log_old.jsonl
```

gdt, 2025-07-04:
There's an open PR, so this issue is done.

closed 2025-07-04 by gdt