merbanan/rtl_433 #3284: playback of an iq file not parsing inputs correctly

mhassell, 2025-05-25:
I believe there is a bug when parsing inputs for playing back an iq file.   To reproduce:

1. Record an iq file through gnuradio or gqrx with a sample rate of 250kHz through an rtlsdr v3 dongle.
2. run the command `rtl_433 -r CU64:/tmp/iq_dump.txt -f 433958500 -v ` 
3. rtl_433 will run briefly and output ` [Input] Test mode file issued 92 packets` even though there was a single burst of data from the sensor
4. To demonstrate functionality of the rtl_433 with live iq data from the same sensor, run `rtl_433  -f 433958500 -v ` and see the successful output from the sensor.

I have some detail on what is going on.  The inputs are not being parsed/initialized correctly and the demod->load_info.format field is garbage data.  This is noticed in rtl_433.c:1882 in that if statement:
`
            if (demod->load_info.format == CU8_IQ
                    || demod->load_info.format == CS8_IQ
                    || demod->load_info.format == S16_AM
                    || demod->load_info.format == S16_FM) {
                demod->sample_size = sizeof(uint8_t) * 2; // CU8, AM, FM
            } else if (demod->load_info.format == CS16_IQ
                    || demod->load_info.format == CF32_IQ) {....
`
We can fix the load_info.format field by:
1. `gdb --args rtl_433 -r CU64:/tmp/iq_dump.txt -f 433958500 -v`
2. `b rtl_433.c:1881`
3. hit `r`
4. at the breakpoint, `set demod->load_info.format=S16_FM` then `c`
5. the screen will print the correct data from the sensor.

zuckschwerdt, 2025-05-25:
Run with `-v` to see the used input format, e.g.
`[Input] Input format "CU8 IQ (2ch uint8)"`
rtl_433 does not support .txt or cu64 as I/Q input and will fallback to .cu8

mhassell, 2025-05-25:
Thanks.  That was my bad, I was messing with the parameters.  I was able to kick this off now, so that helped.  Thanks!

closed 2025-05-25 by mhassell