merbanan/rtl_433 #3287: Decoding Maverick XR-50

Luca7993, 2025-06-02:
Hi,

I'm working on decoding signals from the Maverick XR-50 BBQ Thermometer, and I'd appreciate some help.

I capture the signal using these parameters:
-f 868000000 -g 40 -S 

Here is the configuration I'm using in my decoder:
`r_device const maverick_xr50 = {
    .name        = "Maverick XR-50 BBQ Sensor",  // Nome del nuovo modello
    .modulation  = FSK_PULSE_PCM,
    .short_width = 107,
    .long_width  = 107,
    .reset_limit = 109568,
    .decode_fn   = &maverick_xr50_callback,  // Collegamento alla nuova funzione di decodifica
    .fields      = output_fields,
};`

I’ve written a decoder function, and it usually works (though I still need to fine-tune the low/high temperature conversion — that's not the issue right now).

I’d like to ask:

- Am I on the right track? I have a solid background in programming, but I’m still learning about RF and modulation.
- What is the purpose of the digest function in this context?
- Is there a checksum in each message that I should validate?

I’ve attached:
Some .cu8 sample files
My decoder code

All files decode correctly, except for old18.cu8. I’m not sure why that one fails. When I load it in triq.org/pdv, it auto-detects:

Guessing modulation: Pulse Code Modulation (Not Return to Zero)
modulation: PCM short: 107300000.0 long: 107300000.0 sync: - gap: - reset: 109875200000.0

Which clearly isn't correct — so I suspect there’s something wrong in how I’m parsing or preprocessing the data.

Any suggestions would be greatly appreciated!

Thanks in advance!

[maverick.zip](https://github.com/user-attachments/files/20548202/maverick.zip)

Luca7993, 2025-06-03:
Thanks, I'll let you know when I have time to work on it.

zuckschwerdt, 2025-06-07:
Usually groups would be 2 or 3 hex characters, i.e. 8 or 12 bit (12 bits is common for a .1 degree resolution temperature value).
Try to table 20-40 distinct codes, preferably with only very slight changes in values, e.g. slowly warming up or cooling down.

Luca7993, 2025-06-08:
All temperature values are 3 hex characters (After i convert them in decimali, subtract 500 and divide 10 and i get all correct temperature values)
My problem now Is to understand how to get the checksum to understand if the signal Is completely ok.
Anyway, as soon as i have time i Will give you distinct codes, hoping you help me go understand how is build the checksum

Thank you

ProfBoc75, 2025-06-08:
The end of the message after the CRC8 , ie 0x00aaaa000000, it's trailing bit, could be used at rf decoding level to synchronize by the receiver, but useless with rtl_433, and not part of the decoding.

Luca7993, 2025-06-08:
have an Excel file that shows the decoded values, but I need some time to organize and share it (I have a child and now I have to go out 🙂).
I also managed to calculate the CRC correctly.

So far, this is what we have understood:

- 4-digit prefix

- Then 4 groups of 10 characters (one for each probe):
 
-     Each group has:     
-       1 character (still unknown purpose)     
-       3 characters for the temperature      
-       3 characters for the high threshold     
-       3 characters for the low threshold

After these 40 characters, there are 2 characters for the CRC-8 checksum (as you suggested)

Then the remaining part: a00aaaa000000, which seems to be constant and likely not part of the decoding

I attach the working decoder file, even I have to refine it

[maverick_xr50.zip](https://github.com/user-attachments/files/20643969/maverick_xr50.zip)

Luca7993, 2025-06-09:
thank you for your help :)

 I have only changed high and low values to int (even if are float in the message, on devices are only seen as int)

ProfBoc75, 2025-06-09:
@Luca7993 I understand for your needs, up to you, but for overall usage, keep any temperature value as float even if the source is int, this is a standard to be respected. rtl_433 is used by other programs like HA or Domoticz, where Temperature is expected to be float value.

zuckschwerdt, 2025-06-09:
Are the high/low outputs actual temperature readings or settings? Other decoders report such as `setpoint_C`.

Luca7993, 2025-06-09:
On the device, you can set only a High alarm (for example, to alert you when the meat reaches a target temperature), or both Low and High alarms (usually for ambient temperature) to ensure it doesn’t get too hot or too cold during cooking.

zuckschwerdt, 2025-06-10:
Also think about chaning to `setpoint_high_1_C` etc.

Luca7993, 2025-06-18:
Hi, I've made the changes a few days ago and updated the pull request. Just wanted to check if there's anything else I should do, or if you're just busy (no worries if that's the case!).

zuckschwerdt, 2025-06-18:
Just a fewer minor code style changes which can be cleaned up after merge. But if you've got the time I'll review #3293 just now.

Luca7993, 2025-06-18:
Thanks! I should have applied those minor style fixes. Let me know if you spot anything else.

closed 2025-07-04 by gdt