merbanan/rtl_433 #3290: Capturing available signals

andreaskoelsch, 2025-06-04:
I just wanted to see what signals are available in my neighborhood, so I ran `rtl_433 -f 868M -s 1024k`. If I understand the output correctly, it did capture some FSK signals. How can I see their content?
[image]

I was expecting something like the screenshot in the README
[image]
What command was used to create that screenshot?

zuckschwerdt, 2025-06-04:
The stats you see are about undecodeable signals.

Try this for more verbose output: `rtl_433 -v -Y autolevel -M level -M noise -f 868M`

andreaskoelsch, 2025-06-04:
Tried that. I guess, it's just very quiet around here or the devices are not known for decoding.

```
{"time" : "2025-06-04 09:34:38.032698", "enabled" : 216, "since" : "2025-06-04T09:24:38", "frames" : {"count" : 3, "fsk" : 2, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 09:44:38.001303", "enabled" : 216, "since" : "2025-06-04T09:34:38", "frames" : {"count" : 4, "fsk" : 3, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 09:54:38.097544", "enabled" : 216, "since" : "2025-06-04T09:44:38", "frames" : {"count" : 8, "fsk" : 2, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 10:04:38.065850", "enabled" : 216, "since" : "2025-06-04T09:54:38", "frames" : {"count" : 11, "fsk" : 1, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 10:14:38.034134", "enabled" : 216, "since" : "2025-06-04T10:04:38", "frames" : {"count" : 13, "fsk" : 3, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 10:24:38.002295", "enabled" : 216, "since" : "2025-06-04T10:14:38", "frames" : {"count" : 21, "fsk" : 1, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 10:34:38.098448", "enabled" : 216, "since" : "2025-06-04T10:24:38", "frames" : {"count" : 15, "fsk" : 8, "events" : 0}, "stats" : []}
{"time" : "2025-06-04 10:44:38.066580", "enabled" : 216, "since" : "2025-06-04T10:34:38", "frames" : {"count" : 16, "fsk" : 2, "events" : 0}, "stats" : []}
```

Thanks anyways 🙂

zuckschwerdt, 2025-06-04:
If you enable JSONL then the colored pretty output is disabled. Add `-F kv` (or `-F log`) to see more info.

andreaskoelsch, 2025-06-04:
This is the same information, just more logs about noise level adjustments and more color. But no devices found. It's okay though. Guess, there's just nothing there in my neighborhood

zuckschwerdt, 2025-06-04:
The noise and level information can show you if the band is clear or maybe jammed / bad antenna.
If you add `-A` you will see all possible signals -- but that may spam the output at a high rate.

AkaBoing, 2025-06-15:
for the picture:
with Debian and XFCE
use xfce-screenshooter.
and select "active window"

closed 2025-06-15 by zuckschwerdt