merbanan/rtl_433 #3304: Support for EMOS doorbell (similar to EV1527)

vodickado, 2025-06-20:
I have purchased a EMOS doorbell which transmits on 433mhz using chip **cmt2150L**.
From what I read it should be very similar to EV1527.
I'm able to receive somewhat consistently device ID using following slicer arguments:
 -X "n=EMOS_doorbell,m=OOK_PWM,s=60,l=272,y=672,g=132,r=1500,bits>=4,bits<=4,get=@0:{24}:id,unique"

But I'm not really sure if thats really id, and the repeatability is questionable.
I'm attaching few received signals.

[emos-doorbell_signals.zip](https://github.com/user-attachments/files/20834634/emos-doorbell_signals.zip)

zuckschwerdt, 2025-06-20:
It looks like each instance is 3 packets: a single pulse then a wider timing (60/260 µs pulses + 140 µs gap) then 1.5 ms pause and a narrower timing (50/150 µs pulses + 150/50 µs gaps), then ~200 ms pause and maybe the second packet again.

Hard to make out as the signal is directly on 433.92M. Try to record again on 433.85M and maybe try more distance to prevent some clipping (signal is too "loud").

vodickado, 2025-12-07:
Sorry, forgot to leave a feedback:
After using the suggested parameter "-s 1024k", the signal was reliably received every single time :)